autowarefoundation/autoware_universe #2272: `obstacle_stop_planner` does not revert its avoidance path after the avoided obstacle is gone.

VRichardJP, 2022-11-12:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I have deployed several vehicles near a train station. It is very common for other vehicles to come, stop for a minute to drop off someone, then go away.

Although the avoidance works nicely while the vehicle is parked on the side of the road, autoware still wants to follow the avoidance path after the vehicle is gone. This is very akward for the safety driver, as autoware moves away from its lane for no apparent reason

This can be reproduced with the planning simulator easily:

1. No obstacle -> no avoidance
[image]

2. A vehicle comes and park -> avoidance path is generated after a few seconds
[image]

3. The vehicle is gone, but autoware does not revert the path to 1 (even if you wait for a very long time)
[image]

### Expected behavior

If there is no obstacle, there should be no avoidance

### Actual behavior

generated avoidance path stays for ever.

### Steps to reproduce

See the pictures. You can use the fake bus to emulate a parked vehicle, and delete all obstacles after avoidance path is generated.

satoshi-ota, 2022-11-12:
@VRichardJP There is currently no function to cancel/revert avoidance path when it is no longer necessary, so this is how it would behave in the scene described above. I am currently working on developing a function that takes into account vehicles in adjacent lanes when avoiding, and if it is not safe to follow avoidance path right now, the vehicle stops behind the avoidance target and lets the adjacent lane's vehicle pass over, so I will consider this function together.

Of course, if you would like to implement revert function in avoidance module, your contributions are welcome :smile:

VRichardJP, 2022-11-12:
@satoshi-ota I am not well versed in the planning logic, but if I find the time I'll try to implement it.

ismetatabay, 2022-11-17:
@satoshi-ota @VRichardJP This issue had caught my attention before, and I did a little research on it. I think the problem is that the ```avoidance``` module in the Behavior path planner tree, after detecting the object once and running, stays in the running state until the vehicle reach its goal. This is how I solved this problem:

https://github.com/autowarefoundation/autoware.universe/blob/29e8b435a4416e5db0eb6c7da0c4b0457e6c529b/planning/behavior_path_planner/src/scene_module/avoidance/avoidance_module.cpp#L97-L114

I'm checking the object data from the ```planner_data_``` by adding an if statement to above method. If there is no object, I make the module complete the task by making the avoidance module ```current_state_``` = SUCCESS. If you think this solution is a suitable solution, I can create a PR on this issue by adding this check option as a parameter. Here is the video:

https://user-images.githubusercontent.com/56237550/202425100-9ceca809-f4f6-4ee3-8586-d70c03f4eaf3.mp4

VRichardJP, 2022-11-17:
That looks perfect!

PR #2314 (2022-12-01): feat(behavior_path_planner): update path when object is gone

## Description

At avoidance module in behavior_path_planner, if the detected objects occur in path,  avoidance module shift path points to around the vehicle. Moreover, if the detected objects are gone the shifted path is not updated. The reason of this, avoidance module' current state stays in the "RUNNING" state until the vehicle reach its goal. I added a feature to change this state, and this feature allows the path to be updated by completing the task of the avoidance module if the detected objects are gone.

To control this, I [... cut]
closed 2022-12-01 by satoshi-ota