autowarefoundation/autoware_universe #2277: component_state_monitor does not start properly

satoshi-ota, 2022-11-13:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

show following error

```
Task exception was never retrieved                                                                                                                                                                             
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py:226> exception=AttributeError("'PathJoinS
ubstitution' object has no attribute 'get_launch_description'")>                                                                                                                                               
Traceback (most recent call last):                                                                                                                                                                             
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event                                                                                               
    await self.__process_event(next_event)                                                                                                                                                                     
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event                                                                                                  
    visit_all_entities_and_collect_futures(entity, self.__context))                                                                                                                                            
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures                                     
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)                                                                                                                           
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures                                     
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)                                                                                                                           
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures                                     
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)                                                                                                                           
  [Previous line repeated 12 more times]                                                                                                                                                                       
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures                                     
    sub_entities = entity.visit(context)                                                                                                                                                                       
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/action.py", line 108, in visit                                                                                                                    
    return self.execute(context)                                                                                                                                                                               
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/actions/include_launch_description.py", line 127, in execute                                                                                      
    launch_description = self.__launch_description_source.get_launch_description(context)                                                                                                                      
AttributeError: 'PathJoinSubstitution' object has no attribute 'get_launch_description' 
```

I commented out the following line, the error messages were shown.

https://github.com/autowarefoundation/autoware.universe/blob/89d137a68bca15c9a4faead6fbc13778093553cb/launch/tier4_system_launch/launch/system.launch.xml#L40-L45

### Expected behavior

launch autoware without error messages

### Actual behavior

show above error messages

### Steps to reproduce

1. clone [autoware](https://github.com/autowarefoundation/autoware)
2. vcs import src < autoware.repos
3. rosdep install --from-paths src --ignore-src -y
4. colcon build
5. ros2 launch autoware_launch planning_simulator.launch.xml map_path:=/$HOME/map/kashiwanoha sensor_model:=sample_sensor_kit vehicle_model:=sample_vehicle

### Versions

- OS: 20.04
- galactic
- use [autoware](https://github.com/autowarefoundation/autoware)

isamu-takagi, 2022-11-15:
@satoshi-ota Could you try updating `ros-galactic-launch`?

satoshi-ota, 2022-11-15:
@isamu-takagi OK. I'll try it.

satoshi-ota, 2022-11-18:
it works well with `ros-galactic-launch` update.
Thanks :+1:

closed 2022-11-18 by satoshi-ota