autowarefoundation/autoware_universe #2291: Global planner fails when the goal point is on the same lanelet, behind the vehicle

VRichardJP, 2022-11-15:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

When the lanelet2 map has a loop, it is possible to have Autoware driving over the loop by setting the goal point behind the vehicle. It is very convenient for testing.

However, if the goal points and the ego vehicle happen to be on the same lanelet, the global planner fails to generate a valid path. For example:
[image]

If the goal point is behind the vehicle, but on a different lanelet, then there is no problem.

### Expected behavior

As long as there is a possible path from start to goal on the lanelet2 map, Autoware should be able to generate it. 

### Actual behavior

When the goal point is behind the ego vehicle, on the same lanelet, no valid global plan is generated.

### Steps to reproduce

Can be reproduced with the simulator, using any map with a loop.

shulanbushangshu, 2022-11-15:
Yes,this is a known limitation in the current planning stack.You also can  see  the discussion in https://github.com/autowarefoundation/autoware.universe/issues/1715

VRichardJP, 2022-12-16:
duplicate of #1715

closed 2022-12-16 by VRichardJP