autowarefoundation/autoware_universe #2306: Distortion correction would not work using point_cloud_msg_wrapper

miursh, 2022-11-16:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Distortion correction node does not compensate ego vehicle motion.

### Expected behavior

The xy position of `/sensing/lidar/top/rectified/pointcloud_ex` must have some offset based on its vehicle velocity like this image below.
(Yellow: rectified/pointcloud_ex, Dark gray: and pointcloud_raw)

[image]

### Actual behavior

Distortion correction node outputs warning on the terminal.
```
[component_container-8] [WARN 1668606722.543281569] [sensing.lidar.rear.distortion_corrector_node]: Required field time stamp doesn't exist in the point cloud.
[component_container-6] [WARN 1668606722.947966127] [sensing.lidar.left.distortion_corrector_node]: Required field time stamp doesn't exist in the point cloud.
[component_container-7] [WARN 1668606722.950508348] [sensing.lidar.right.distortion_corrector_node]: Required field time stamp doesn't exist in the point cloud.
[component_container_mt-5] [WARN 1668606723.075554512] [sensing.lidar.top.distortion_corrector_node]: Required field time stamp doesn't exist in the point cloud.
```

The xy position of `/sensing/lidar/top/rectified/pointcloud_ex` is exact same as `/sensing/lidar/top/pointcloud_raw_ex`. 
This image shows rectified/pointcloud_ex in yellow, and pointcloud_raw in dark gray.

[image]

### Steps to reproduce

1. run logging_simulator.launch.xml with certain args.
2. play sample rosbag

miursh, 2022-11-16:
I think because of [this PR](https://github.com/autowarefoundation/autoware.universe/pull/1276), some pointcloud preprocessor does not work as expected.
1. Distortion corrector could not parse timestamp field.
2. If concatenate_data subscribes pointcloud topic other than XYZI type, the crashes with runtime error.

@esteve Have you try one of those?

yukkysaito, 2022-11-18:
@miursh cc @esteve 
It may be better to revert once if it seems to take a long time to respond.

miursh, 2022-11-22:
https://github.com/autowarefoundation/autoware.universe/pull/2317 has resolved this issue.