autowarefoundation/autoware_universe #2308: TVM copy overhead

angry-crab, 2022-11-17:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

In a heterogeneous context, multi-state inference introduces copy overhead. 
https://github.com/liuzf1988/autoware.universe/blob/0fcaf94d7836f42eaf4a256fca97e5b63d2487d0/perception/lidar_centerpoint_tvm/lib/centerpoint_tvm.cpp#L225-L228
For example, the inference above first copies data from cpu to gpu, then gpu to cpu for `scatter`, and gpu back to cpu for the final stage inference. Data tranfer between devices should be avoided. 

### Purpose

Avoid data copy between devices and improve performance. 

### Possible approaches

Implement `scatter` with TVM script. 

### Definition of done

- [ ] modify ci pipleline to include pre/postprocessing. 
- [ ] implement `scatter`

angry-crab, 2022-12-01:
@ambroise-arm 
Regarding the `scatter` compilation, it is something like this
https://github.com/angry-crab/tvm_example/blob/d4d146bdc657915553ebf510e044e45743a6a32b/scatter_tir.py#L8-L21
To push it to `modelzoo`, I think we have two options:
1. Compile it into one of the onnx model with a predefined name such as `util`
2. Consider it as another "network" and compiled into a different runtime lib. 

I guess the second seems better because there might be multiple such operators. 

As for `modelzoo` pipeline, do you have any suggestion where I should put it?

ambroise-arm, 2022-12-01:
> I guess the second seems better because there might be multiple such operators.
> As for modelzoo pipeline, do you have any suggestion where I should put it?

I like option 1. better, unless you have examples of the same `scatter` function being used in other processing pipelines. 

Instead of the `util` name, I would go for `preprocess` (if we consider it is the preprocessing for the backbone).
And then this could be made a general design rule that any network could benefit from `preprocess` and `postprocess` utilities like this. So the layout of a network archive would become:
```
deploy_graph.json
deploy_lib.so
deploy_param.params
inference_engine_tvm_config.hpp
preprocess.so (optional)
postprocess.so (optional)
```
And the `definition.yaml` of the networks could be modified to add the information on whether those optional libraries are to be generated and point to the python script (the kind you have written as example) to generate it. And finally the `tvm_cli` script would be modified to execute those when building the network.

It sounds like more modifications to `modelzoo` than option 2, but I think it will be lighter on the `modelzoo`/`Autoware` interface changes. It also keeps the backend of `scatter` the same as the one of the main inference, so the Autoware developer doesn't have to worry about that. What do you think?

angry-crab, 2022-12-07:
`NetworkNode` of `tvm_utility` needs to be modified according to [Template config.hpp input and output type](https://github.com/autowarefoundation/autoware.universe/issues/2309)

closed 2023-01-10 by angry-crab