autowarefoundation/autoware_universe #2309: Implement yolo_TVM

angry-crab, 2022-11-17:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Implement yolo with TVM

### Purpose

Implement yolo with TVM

### Possible approaches

Implement cuda kernels using cpp/tvm

### Definition of done

- [ ] verify accuracy of compiled model
- [ ] implement cuda kernel using cpp/tvm
- [ ] port yolo peripheral functions

raminterpl, 2022-12-07:
@angry-crab : Could you pls provide more technical details 
documents for this task @Deepak3994 : will look into it and we will plan any discussion if required

angry-crab, 2022-12-07:
@raminterpl @Deepak3994 
Thank you so much for your interest.
For information about [tvm](https://tvm.apache.org/#about), please see the link above. 
AS for the current tvm pipeline of Universe and steps to port TensorRT packages to tvm, please check the following link:
[tvm_overview](https://docs.google.com/presentation/d/12KPSEyX7Gce-PtYlZhgw0Iqt4XPk1VqUNXbdHaUrc-w/edit?usp=share_link)

To put simply, 
1. we need to first verify the accuracy of compiled model for llvm and opencl backend. Please see
[tvm_test](https://github.com/angry-crab/tvm_example/blob/main/tvm_test.py) for an example. 
2. Implement cuda kernel functions with cpp/tvmscript
3. Port other functions to the tvm package. 

Please let me know if there is any question/problem.

raminterpl, 2023-01-10:
@Deepak3994 : Deepak is looking into it.

Deepak3994, 2023-02-02:
@angry-crab 

I see the initial step for verifying accuracy of the compiled model is to run the test from "https://github.com/angry-crab/tvm_example/blob/main/python/tvm_test.py". But in that script i see that we will be importing the pre-compiled models, backbone etc. From where can we find these models for the current yolo part?

angry-crab, 2023-02-03:
> From where can we find these models for the current yolo part?

Hi, maybe you can check the one currently being used by tensorrt. 
https://github.com/autowarefoundation/autoware.universe/blob/main/perception/tensorrt_yolox/CMakeLists.txt

Deepak3994, 2023-02-03:
Hi, I went through the camke file. It mentioned that models will be downloaded. For now i tried "https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation-devel/"
this link for setting up the enviornment. But after the last step of running "colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release", i am not able to see any model files getting downloaded.

Is it the ryt way or there are some other commands/ methods to get the model files downloaded?

angry-crab, 2023-02-06:
> Is it the ryt way or there are some other commands/ methods to get the model files downloaded?

Hi, sry for the confusion. We need to do `colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release -DDOWNLOAD_ARTIFACTS=ON` to download the package. 
Further information could be found here. 
https://github.com/autowarefoundation/autoware.universe/blob/main/common/tvm_utility/design/tvm-utility-design.md

Deepak3994, 2023-02-14:
Hi @angry-crab , I have downloaded the models for yolo-tinyx. Currently I am working on the test script which you have suggested to look into for getting accuracy of compiled model for llvm and opencl backend. 

**Script**: `https://github.com/angry-crab/tvm_example/blob/main/python/tvm_test.py`

I cloned the code and ryt now working on running this script for the "yolo-tinyx.onnx" model file. Few libraries its requesting to be installed, currently doing the same. I will post the test results here once its generated.

angry-crab, 2023-02-15:
@Deepak3994 It seems that the operation is available. Which version of tvm are you using?

Deepak3994, 2023-02-15:
Actually, import tvm itself is throwing error. Do you happen to have any doc which has installations for tvm?

[image]

angry-crab, 2023-02-15:
> Do you happen to have any doc which has installations for tvm?

I'm simply following the instructions from official tvm doc. 
https://tvm.apache.org/docs/install/index.html

btw, maybe you can use the docker image from model_zoo. but make sure to change the entry point and rebuild the image. 
https://github.com/autowarefoundation/modelzoo

sujithvemi, 2023-02-21:
Hi,

The issue we are facing is registered as a bug on TVM Github issues page here.

https://github.com/apache/tvm/issues/13507

Please let us know what call to take because I am facing this issue even after a thorough source installation and not just through package installation.

Regards,
Sujith.

angry-crab, 2023-02-22:
@sujithvemi 
Thanks for the info. I've never met this issue before. Have you tried to use the docker image of modelzoo? 
https://github.com/autowarefoundation/modelzoo

raminterpl, 2023-06-27:
@xmfcx @mitsudome-r : Deepak is leaving Interplai this week.  could you please assign this task to any other team who can take up this

amadeuszsz, 2024-07-31:
@kminoda 
I see we slowly move towards TensoRT based inference only. Since we keep removing TVM based package, should we close this issue?

xmfcx, 2024-07-31:
Yeah, this task is long overdue.

closed 2024-07-31 by xmfcx