autowarefoundation/autoware_universe #2310: The  behavior_path_planner can not plan the right path in some positions as the target

shulanbushangshu, 2022-11-17:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

When I set the different positions of target in the lane,the behavior_path_planner can not plan the right path in some positions.
case 1:

[image]

case 2:

[image]

case 3:
[image]

case 4:

[image]

### Expected behavior

The behavior_path_planner shall plan the right path in all positions of the lanes.

### Actual behavior

The behavior_path_planner can not plan the right path when I set some different positions  as the target in the lanes.

### Steps to reproduce

1.Start autoware planning simulator;
2. Set initial position 
3. Set target in the different positions in the lane(far from the initial position)

### Versions

1.OS: Ubuntun 20.04
2.ROS2: Galactic
3.Autoware: Autoware.universe:main

ismetatabay, 2022-11-17:
Hi, can you please share map (pcd and lanelet) files? The lanelets which highlighted as I have shown below picture show that the route has been successfully generated by the mission_planner, I didn't see any problem in this case. If you're worried about not seeing the green path in rviz, the path will move forward as the vehicle move.  The parameter that is ```forward_path_length``` limits the path . You can also increase the generated path by changing this parameter.

https://github.com/autowarefoundation/autoware.universe/blob/29e8b435a4416e5db0eb6c7da0c4b0457e6c529b/launch/tier4_planning_launch/config/scenario_planning/lane_driving/behavior_planning/behavior_path_planner/behavior_path_planner.param.yaml#L4

[image]

shulanbushangshu, 2022-11-18:
@ismetatabay  As you say, the route has been successfully generated by the mission_planner.Yes,I get what you mean.The distance from the target is more than 300m.But  the problem is that final path of the behavior_path_planner is short(I print the size of the path and it shows that there are just  two points in the path)

shulanbushangshu, 2022-11-18:
@ismetatabay  I  test the code and show the information about the path of behavior_path_planner as below:

case1:

[image]

case 2:

[image]

ismetatabay, 2022-11-18:
@shulanbushangshu This is weird thing that it shouldn't happen. At this case, if ```forward_path_length``` is 300m and ```backward_path_length``` is 5m, the number of path points created should be approximately 305/2 ≅152. To see if this situation is related to distance goal distance , I modified the sample .osm file of autoware as in the image below. 

I've tried many different scenarios but haven't run into a problem. As expected, if the distance to the goal was greater than 300 meters, approximately 152 path points were generated on ```bpp```.

Is it possible for you to share the lanelet2 .osm file?

[image]

closed 2023-02-01 by shulanbushangshu