autowarefoundation/autoware_universe #2364: Lane change fails in this situation

VRichardJP, 2022-11-24:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Found a situation where Autoware fails to generate a lane change:
[lanelet2_map.txt](https://github.com/autowarefoundation/autoware.universe/files/10081588/lanelet2_map.txt)
[image]

All the lanes are changeable up to the traffic merge area, and the road looks long enough for the lane change. Yet, the vehicle seems not to be able to change lane:
[image]

Why is that?

### Expected behavior

Vehicle can lane change

### Actual behavior

Autoware fails to generate a lane change and stops halfway 

### Steps to reproduce

see lanelet file

zulfaqar-azmi-t4, 2022-11-24:
Hi @VRichardJP 

For lane change lane, it is recommeded to add `lane_change` tag to ensure it works properly.
I attached the edited version here.

[lanelet2_map.txt](https://github.com/autowarefoundation/autoware.universe/files/10082923/lanelet2_map.txt)

[image]

VRichardJP, 2022-11-24:
@zulfaqar-azmi-t4 Thank you!

How come lane change is possible in most cases with only dashed lines (no `lane_change` tag)? Is it normal for Autoware to have multiple behaviors, depending on whether the line is dashed and whether the `lane_change` tag is set to true?

zulfaqar-azmi-t4, 2022-11-24:
The behavior differences might not came from the lane change module. From the general observation, for no `lane change` tag version, as lane is not registered in the preferred lane (related to either mission planner or route handler), the lane change module is not activated. As I am not well versed in there part, let me take some time to dig into the code.

xmfcx, 2022-12-13:
Hello @zulfaqar-azmi-t4, did you have time to work on this issue? If you've created a draft PR, can you link it here please?

zulfaqar-azmi-t4, 2022-12-19:
@xmfcx @VRichardJP 
I deeply apologize for the delay. At the moment I am a bit occupied with the internal related task. Therefore I still haven't been able to find out the root cause yet.

xmfcx, 2023-01-10:
@zulfaqar-azmi-t4 I've assigned to @piotr-zyskowski-rai they will look into it. If you have any tips, you can share them here to start with the task.

zulfaqar-azmi-t4, 2023-01-11:
@xmfcx @piotr-zyskowski-rai 

Thank you very much. I deeply apologize for all the inconveniences.

The possible reason is due to the intersection at the merging points. 

Due to the internal safety requirement, it was decided that lane change should not be performed nearby any intersections.

It is not related to lane change module, but instead it might be due to `route_handler` or `mission planner`

zulfaqar-azmi-t4, 2023-01-30:
@VRichardJP @piotr-zyskowski-rai @xmfcx 

Hi everyone, sorry for the delay.

Recently there are several lane change commit made, so among them is this PR #2720 

I have tested @VRichardJP's map, and it seems to work in my case. 

https://user-images.githubusercontent.com/93502286/215445908-847cf5dd-be57-44c7-a249-78ad0c7d994b.mp4

Therefore, I very much appreciate if you could retest and let me know whether it is working.

VRichardJP, 2023-02-14:
It works!
Thank you for your fix!

closed 2023-02-14 by VRichardJP