autowarefoundation/autoware_universe #2374: Out-of-range access in `obstacle_stop_planner::insertSlowDownSection`

VRichardJP, 2022-11-25:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

While using the slow down planner to increase ACC detection range (see https://github.com/autowarefoundation/autoware.universe/issues/2262), we experienced multiple crash due to out-of-range accesses in `obstacle_stop_planner::insertSlowDownSection`.

We added the following snippet to detect the issue, and the node stopped crashing:
```diff
diff --git a/planning/obstacle_stop_planner/src/node.cpp b/planning/obstacle_stop_planner/src/node.cpp
index 2d26ce7b3..d993c0b32 100644
--- a/planning/obstacle_stop_planner/src/node.cpp
+++ b/planning/obstacle_stop_planner/src/node.cpp
@@ -756,6 +756,33 @@ void ObstacleStopPlannerNode::insertSlowDownSection(
   const auto & start_idx = slow_down_section.slow_down_start_idx;
   const auto & end_idx = slow_down_section.slow_down_end_idx;
 
+  // for debug
+  if(start_idx >= output.size())
+  {
+    RCLCPP_ERROR(get_logger(), "start_idx >= output.size() (%lu, %lu)",
+      start_idx, output.size());
+    return;
+  }
+  if(std::min(start_idx + 1, traj_end_idx) >= output.size())
+  {
+    RCLCPP_ERROR(get_logger(), "std::min(start_idx + 1, traj_end_idx) >= output.size() (%lu, %lu)",
+      std::min(start_idx + 1, traj_end_idx), output.size());
+    return;
+  }
+  if(end_idx >= output.size())
+  {
+    RCLCPP_ERROR(get_logger(), "end_idx >= output.size() (%lu, %lu)",
+      end_idx, output.size());
+    return;
+  }
+  if(std::min(end_idx + 1, traj_end_idx) >= output.size())
+  {
+    RCLCPP_ERROR(get_logger(), "std::min(end_idx + 1, traj_end_idx) >= output.size() (%lu, %lu)",
+      std::min(end_idx + 1, traj_end_idx), output.size());
+    return;
+  }
+
+
   const auto p_base_start = output.at(start_idx);
   const auto p_next_start = output.at(std::min(start_idx + 1, traj_end_idx));
   const auto & p_insert_start = slow_down_section.start_point;
```

However, we now see several of the added error messages appear from time to time in the logs.

Example of backtrace when the crash occurs (before the changes):
```
[INFO] [1668673327.955292305] [planning.scenario_planning.lane_driving.motion_planning.obstacle_avoidance_planner]: Replan with resetting optimization since path shape was changed.
[INFO] [1668673329.745063671] [planning.scenario_planning.lane_driving.motion_planning.obstacle_avoidance_planner]: Replan with resetting optimization since path shape was changed.
[INFO] [1668673330.634366514] [planning.scenario_planning.lane_driving.motion_planning.obstacle_avoidance_planner]: Replan with resetting optimization since path shape was changed.
[INFO] [1668673364.590127236] [planning.scenario_planning.lane_driving.motion_planning.obstacle_avoidance_planner]: [Avoidance] Not extend trajectory
[INFO] [1668673374.641078530] [planning.scenario_planning.lane_driving.motion_planning.obstacle_avoidance_planner]: [Avoidance] Not extend trajectory
[INFO] [1668673384.671440235] [planning.scenario_planning.lane_driving.motion_planning.obstacle_avoidance_planner]: [Avoidance] Not extend trajectory
[INFO] [1668673394.756116096] [planning.scenario_planning.lane_driving.motion_planning.obstacle_avoidance_planner]: [Avoidance] Not extend trajectory
[WARN] [1668673411.718066140] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: size of vel que is 0. Something has wrong.
[WARN] [1668673411.817246888] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: size of vel que is 0. Something has wrong.
[WARN] [1668673434.463485101] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: size of vel que is 0. Something has wrong.
[WARN] [1668673434.647350688] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: size of vel que is 0. Something has wrong.
[WARN] [1668673434.687731847] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: size of vel que is 0. Something has wrong.
[WARN] [1668673435.664530239] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: size of vel que is 0. Something has wrong.
terminate called after throwing an instance of 'std::out_of_range'
[... 68 lines cut ...]
    at /usr/include/c++/9/variant:989
#23 0x00007fffd2613942 in std::__detail::__variant::__gen_vtable_impl<true, std::__detail::__variant::_Multi_array<void (*)(rclcpp::AnySubscriptionCallback<autoware_auto_planning_msgs::msg::Trajectory_<std::allocator<void> >, std::allocator<void> >::dispatch(std::shared_ptr<autoware_auto_planning_msgs::msg::Trajectory_--Type <RET> for more, q to quit, c to continue without paging--
<std::allocator<void> > >, rclcpp::MessageInfo const&)::{lambda(auto:1&&)#1}&&, std::variant<std::function<void (autoware_auto_planning_msgs::msg::Trajectory_<std::allocator<void> > const&)>, std::variant<void (autoware_auto_planning_msgs::msg::Trajectory_<std::allocator<void> > const, rclcpp::MessageInfo const&)>, std::variant<void (std::unique_ptr<autoware_auto_planning_msgs::msg::Trajectory_<std::allocator<void> >, std::default_delete<autoware_auto_planning_msgs::msg::Trajectory_<std::allocator<void> > > >)>, std::variant<void (std::default_delete<autoware_auto_planning_msgs::msg::Trajectory_<std::allocator<void> > >, rclcpp::MessageInfo const&)>, std::variant<void (std::shared_ptr<std::function>)>, std::variant<void (std::variant<void (std::default_delete<autoware_auto_planning_msgs::msg::Trajectory_<std::allocator<void> > >, rclcpp::MessageInfo const&)>, rclcpp::MessageInfo const&)>, std::variant<void (std::variant<void (std::default_delete<autoware_auto_planning_msgs::msg::Trajectory_<std::allocator<void> > >, rclcpp::MessageInfo const&)> const&)>, std::variant<void (std::variant<void (std::default_delete<autoware_auto_planning_msgs::msg::Trajectory_<std::allocator<void> > >, rclcpp::MessageInfo const&)> const, rclcpp::MessageInfo const&)>, std::variant<void (std::shared_ptr<autoware_auto_planning_msgs::msg::Trajectory_<std::allocator<void> > >)>, std::variant<void (std::shared_ptr<autoware_auto_planning_msgs::msg::Trajectory_<std::allocator<void> > >, rclcpp::MessageInfo const&)> >&)>, std::tuple<std::variant<std::function<void (autoware_auto_planning_msgs::msg::Trajectory_<std::allocator<void> > const&)>, std::variant<void (autoware_auto_planning_msgs::msg::Trajectory_<std::allocator<void> > const, rclcpp::MessageInfo const&)>, std::variant<void (std::unique_ptr<autoware_auto_planning_msgs::msg::Traje--Type <RET> for more, q to quit, c to continue without paging--
ctory_<std::allocator<void> >, std::default_delete<autoware_auto_planning_msgs::msg::Trajectory_<std::allocator<void> > > >)>, std::variant<void (std::default_delete<autoware_auto_planning_msgs::msg::Trajectory_<std::allocator<void> > >, rclcpp::MessageInfo const&)>, std::variant<void (std::shared_ptr<std::function>)>, std::variant<void (std::variant<void (std::default_delete<autoware_auto_planning_msgs::msg::Trajectory_<std::allocator<void> > >, rclcpp::MessageInfo const&)>, rclcpp::MessageInfo const&)>, std::variant<void (std::variant<void (std::default_delete<autoware_auto_planning_msgs::msg::Trajectory_<std::allocator<void> > >, rclcpp::MessageInfo const&)> const&)>, std::variant<void (std::variant<void (std::default_delete<autoware_auto_planning_msgs::msg::Trajectory_<std::allocator<void> > >, rclcpp::MessageInfo const&)> const, rclcpp::MessageInfo const&)>, std::variant<void (std::shared_ptr<autoware_auto_planning_msgs::msg::Trajectory_<std::allocator<void> > >)>, std::variant<void (std::shared_ptr<autoware_auto_planning_msgs::msg::Trajectory_<std::allocator<void> > >, rclcpp::MessageInfo const&)> > >, std::integer_sequence<unsigned long, 4ul> >::__visit_invoke(rclcpp::AnySubscriptionCallback<autoware_auto_planning_msgs::msg::Trajectory_<std::allocator<void> >, std::allocator<void> >::dispatch(std::shared_ptr<autoware_auto_planning_msgs::msg::Trajectory_<std::allocator<void> > >, rclcpp::MessageInfo const&)::{lambda(auto:1&&)#1}, std::variant<std::function<void (autoware_auto_planning_msgs::msg::Trajectory_<std::allocator<void> > const&)>, std::variant<void (autoware_auto_planning_msgs::msg::Trajectory_<std::allocator<void> > const, rclcpp::MessageInfo const&)>, std::variant<void (std::unique_ptr<autoware_auto_planning [... cut ...] auto_planning_msgs::msg::Trajectory_<s--Type <R--Type <RET> for m--Type <RET> for m--Type <RET> for more, q to quit, c to contin--Type <R--Type <RET> for more--Type <RET> for more, q to quit, c--Type <RET> for more--Type <RET> for more, q to --Type <RET> f--Type --Type <RET> for more, q to quit, c to continue without paging--Quit
(gdb) 
```

### Expected behavior

No crash

### Actual behavior

crash

### Steps to reproduce

see https://github.com/autowarefoundation/autoware.universe/issues/2262

xmfcx, 2022-12-13:
@satoshi-ota Did you have time to work on this? If you've created a PR, can you link it here?

satoshi-ota, 2022-12-15:
@xmfcx @VRichardJP Sorry late reply :pray: 
Unfortunately, I did not check it. I'll try to fix in next week.