autowarefoundation/autoware_universe #2375: rtc_replayer can't replay rosbag rtc command for logsim

taikitanaka3, 2022-11-25:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

This uuid is unique for ll2 in behavior velocity but in behavior path planner uuid is completely random so it's impossible to send command using uuid in rosbag. https://github.com/autowarefoundation/autoware.universe/blob/4f6c3d4c5ae134985b2bc2ec0033e86a41530a0b/planning/rtc_replayer/src/rtc_replayer_node.cpp#L113

### Expected behavior

replay behavior path rtc

### Actual behavior

cant replay behavior path rtc

### Steps to reproduce

ros2 bag play $BAG --remap rosbag api/external/get/rtc_status:=/debug/rtc_status
ros2 launch rtc_replayer rtc_replyer_node

taikitanaka3, 2022-11-30:
@mitsudome-r 
This is a part of [reminder](https://star4.slack.com/archives/C03QW0GU6P7/p1669345676191219?thread_ts=1669253894.277569&cid=C03QW0GU6P7) for me, 
I mean logsim for logging simulator launch, so it's already open source (sorry for confusion)

What I am trying to do is to replay rtc from rosbag data using rtc status topic.
This PR description describes purpose.

xmfcx, 2023-01-10:
@taikitanaka3 Has https://github.com/autowarefoundation/autoware.universe/pull/1993 resolved this issue?

taikitanaka3, 2023-01-11:
@xmfcx 
This issue needs this change to solve uuid problem but I didn't test this change
https://github.com/tier4/autoware.universe/commits/fix/rtc_replayer_use_palnning_uuid

closed 2023-03-12 by taikitanaka3