autowarefoundation/autoware_universe #2390: Planning simulator stops for non-existant virtual traffic light

xmfcx, 2022-11-28:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Following the https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/ the vehicle stops at a right turn and stays there forever.

### Expected behavior

Vehicle to move if there is no traffic light.

### Actual behavior

Vehicle stops.

### Steps to reproduce

- Follow https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/
- Give following path:
  - [image]
- Vehicle will stop here:
  - [image]

### Versions

- OS: Ubuntu 22.04
- ROS2: humble
- Autoware Universe

### Additional context

There is not even a traffic light nearby, I don't know why it stops.

[image]

mitsudome-r, 2022-11-29:
Virtual traffic light is meant to be used for V2X communication. The vehicle will wait until it receives approval from the external software (infrastructure) to pass.
https://autowarefoundation.github.io/autoware.universe/main/planning/behavior_velocity_planner/virtual-traffic-light-design/.

The reason why virtual traffic light being defined in the sample map is because the sample map is originally created by TIER IV to test all features in behavior modules. Maybe we should add tutorial for sending approval in the instruction to avoid confusion.

kosuke55, 2022-12-02:
thank you for your report. sorry, I did not check the actual behavior but the following command may approve it and the vehicle starts to move.
```
sec=$(($(date +%s) + 30)); ros2 topic pub /awapi/tmp/virtual_traffic_light_states tier4_v2x_msgs/msg/VirtualTrafficLightStateArray "{stamp: {sec: $sec, nanosec: 999999999}, states: [{stamp: {sec: $sec, nanosec: 999999999}, type: 'dummy_infrastructure', id: '$1', approval: true, is_finalized: true}]}" -1
```

xmfcx, 2023-01-10:
Thanks, I will add this to the https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/ tutorial steps as a note for newcomers.

soblin, 2023-01-25:
@xmfcx Recently we updated the sample map to contain all kinds of map primitives including virtual_trafic_light, so this is an expected behavior. So the workaround is to
- (1) disable vtl module by changing the flag defined [here](https://github.com/autowarefoundation/autoware.universe/blob/c1a98f7f8822d7766a65c9fedf4ef805861d320e/planning/behavior_velocity_planner/config/behavior_velocity_planner.param.yaml#L9) in autoware_launch OR
- (2) follow @kosuke55 -san's  above instruction
. But this behavior maybe confusing for newcomers, so the tutorial page should be updated. Are you already working on it ? If not, I'll handle that.

mitsudome-r, 2023-01-31:
@soblin It might be difficult for beginners to understand the difference between virtual traffic light(V2I command) and normal traffic light. Unless there is a good documentation for the virtual traffic light use-cases, it might be better to just turn the flag off.

mitsudome-r, 2023-01-31:
@soblin I have talked with Fatih, and it would be nice if you can work on this issue.

soblin, 2023-01-31:
@mitsudome-r Okay! I think we can just disable this flag in awf autoware_launch only.

closed 2023-03-14 by mitsudome-r