autowarefoundation/autoware_universe #2422: `behavior_path_planner` needs to lock `planner_data_` mutex before calling `getTurnSignal`

VRichardJP, 2022-12-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

This line is outside the `mutex_pd_` lock guard:
https://github.com/autowarefoundation/autoware.universe/blob/17f1b956a397b1661471ad820cb53219aaeaaa54/planning/behavior_path_planner/src/behavior_path_planner_node.cpp#L645

This means the `planner_data_` content may be modified when the `getTurnSignal` function is trying to access it.

For example I have modified the `route_handler::getClosestLaneletWithinRoute` to crash if the route handler is not ready:

```
bool RouteHandler::getClosestLaneletWithinRoute(const Pose & search_pose, lanelet::ConstLanelet * closest_lanelet) const
{
  if (!is_handler_ready_) {
    // TODO(vrichard) clean
    std::cerr << "Trying to call getClosestLaneletWithinRoute() but handler is not ready" << std::endl; 
    throw std::runtime_error("debug");
    return false;
  }
  std::optional<lanelet::ConstLanelet> optional_lanelet = lanelet_route_.getClosestLaneletWithinRoute(search_pose);
  if (!optional_lanelet) {
    return false;
  }
  *closest_lanelet = *optional_lanelet;
  return true;
}
```

Then I run the planning simulator and set the goal pose with Rviz, many times. After many tries (it's all about timing), the exception is thrown:

```
[New Thread 0x7fff808ab700 (LWP 114166)]
Trying to call getClosestLaneletWithinRoute() but handler is not ready
terminate called after throwing an instance of 'std::runtime_error'
  what():  debug
[Thread 0x7fff808ab700 (LWP 114166) exited]

Thread 17 "component_conta" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffce7fc700 (LWP 108608)]
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff7947859 in __GI_abort () at abort.c:79
#2  0x00007ffff7bd0911 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff7bdc38c in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff7bdc3f7 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff7bdc6a9 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007fffec0f8d68 in route_handler::RouteHandler::getClosestLaneletWithinRoute(geometry_msgs::msg::Pose_<std::allocator<void> > const&, lanelet::ConstLanelet*) const [clone .cold] () at /home/sig/autoware/install/route_handler/lib/libroute_handler.so
#7  0x00007fffec5acc49 in behavior_path_planner::util::calcLaneletPathAroundPose(std::shared_ptr<route_handler::RouteHandler>, geometry_msgs::msg::Pose_<std::allocator<void> > const&, double, double) () at /home/sig/autoware/install/behavior_path_planner/lib/libbehavior_path_planner_node.so
#8  0x00007fffec6542ff in behavior_path_planner::TurnSignalDecider::getTurnSignal(std::shared_ptr<behavior_path_planner::PlannerData const> const&, autoware_auto_planning_msgs::msg::PathWithLaneId_<std::allocator<void> > const&, behavior_path_planner::TurnSignalInfo const&) () at /home/sig/autoware/install/behavior_path_planner/lib/libbehavior_path_planner_node.so
#9  0x00007fffec4fdca2 in behavior_path_planner::BehaviorPathPlannerNode::run() () at /home/sig/autoware/install/behavior_path_planner/lib/libbehavior_path_planner_node.so
#10 0x00007fffec50ab77 in rclcpp::GenericTimer<std::_Bind<void (behavior_path_planner::BehaviorPathPlannerNode::*(behavior_path_planner::BehaviorPathPlannerNode*))()>, (void*)0>::execute_callback() () at /home/sig/autoware/install/behavior_path_planner/lib/libbehavior_path_planner_node.so
#11 0x00007ffff7e6ef7d in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&) () at /opt/ros/galactic/lib/librclcpp.so
#12 0x00007ffff7e754bb in rclcpp::executors::MultiThreadedExecutor::run(unsigned long) () at /opt/ros/galactic/lib/librclcpp.so
#13 0x00007ffff7c08de4 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#14 0x00007ffff7883609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#15 0x00007ffff7a44133 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
(gdb) 
```

By the time I print `is_handler_ready_` in gdb it is already back to true.

### Expected behavior

planner data requires a mutex lock.

### Actual behavior

route_handler may be read/written simulataneously.

### Steps to reproduce

add a printf or an exception in the `route_handler` function above, and set the goal pose many times in the plannig simulator.

satoshi-ota, 2022-12-02:
@purewater0901 Could you check this issue?

TakaHoribe, 2022-12-07:
@purewater0901 is assigned for this issue

purewater0901, 2022-12-13:
Submit [PR](https://github.com/autowarefoundation/autoware.universe/pull/2493).

@VRichardJP Thank you for your reporting.

VRichardJP, 2023-01-20:
same problem than https://github.com/autowarefoundation/autoware.universe/issues/2495

closed 2023-01-20 by VRichardJP