autowarefoundation/autoware_universe #2473: The lane_change can not plan the right path in some lane conditions

shulanbushangshu, 2022-12-08:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

In the lane_change of behavior_path_planner,the path can plan the right path in some conditions.
The simulation is  shown as below:

[image]

The direct phenomenon is that there is no enough distance after changing lane once.
But there are also lanes exactly that can be used in the map.In the condition of  this map,the route does not include the additional lanes which can be used in the lane_change. And in the lane_change of behavior_path_planner,there are no additional measures to deal with the lane conditions in this case.

### Expected behavior

The planning shall  deal with the lane conditions in this case when changing lane.

### Actual behavior

The planning can not  deal with the lane conditions in this case when changing lane.

### Steps to reproduce

1.Start autoware planning simulator;
2. Set initial position
3. Set target in the lane conditions similar to above

### Versions

1.OS: Ubuntun 20.04
2.ROS2: Galactic
3.Autoware: Autoware.universe:main

### Possible causes

In the condition of this map,the route does not include the additional lanes which can be used in the lane_change

takayuki5168, 2022-12-08:
@zulfaqar-azmi-t4 @rej55

zulfaqar-azmi-t4, 2022-12-08:
Hi @shulanbushangshu 
Thank you very much for the report.
My apologies, I couldn't be able to understand the issue correctly.
Can you include videos or any illustrations that can further help me to be able to get better idea of the scenario?

shulanbushangshu, 2022-12-08:
The lanes in simulation are  shown as below:
[image]
In the simulation,the car shall change lane twice.There is no enough distance after changing lane once.So the car can not change lane twice to arrive the target.But there are also lanes exactly that can be used in the map
And in the lane_change of behavior_path_planner,there are no additional measures to deal with the lane conditions in this case.
 The video simulation are  shown as below:

https://user-images.githubusercontent.com/102840938/206384493-d3d8b1be-56df-46ad-a14d-2bfdfcae6c2d.mp4

zulfaqar-azmi-t4, 2022-12-09:
I see. Thank you very much for your explanation. 

Based on your images, the computation of the lane change will also take care of the additional lane. But i might be wrong so it is okay if you can send me the sample of your lanelet so that I can have a better look at the issue?

Based on look at the video, the drivable area seems weird so that might also be one of the possibilities.

zulfaqar-azmi-t4, 2022-12-19:
@shulanbushangshu 
sorry for taking a bit of time. I think I will have the solution ready by the end of this week.

zulfaqar-azmi-t4, 2023-01-06:
@shulanbushangshu 
The PR is still not yet made due to I am checking other things, but if you have some time, can you test this branch 

```
https://github.com/zulfaqar-azmi-t4/autoware.universe/tree/refactor-lane-change
```

The following parameters in `lane_change.param.yaml` need to be modified
(actually as long as `backward_length_buffer_for_end_of_lane > lane_change_finish_judge_buffer + 1`, then it would be sufficient)

```
      backward_length_buffer_for_end_of_lane: 3.0
      lane_change_finish_judge_buffer: 2.0      # [m]
```

I tried to get the scenario similar to the one you've drawn.

https://user-images.githubusercontent.com/93502286/210921758-cd402fc5-d789-4b6a-9e2d-248c63dc557e.mp4

closed 2023-03-29 by shulanbushangshu