autowarefoundation/autoware_universe #2497: Dataset License & Usage for Autoware

armaganarsln, 2022-12-13:


samet-kutuk, 2023-01-22:
Here's the KITTI dataset owner's comment: "The dataset can be used as long as the results of its usage are made public (e.g., via an openly accessible publication) and as long as the dataset does not serve for product development."

I'm assuming, Autoware wouldn't be categorized as a product. But I'd like to get some comments from @mitsudome-r 

I will continue with other datasets, but who's maintaining the list of datasets? Are there only four of them? @armaganarsln

samet-kutuk, 2023-04-11:
I have some updates regarding the dataset use topic.

- I talked to the KITTI dataset owner, and because the Autoware project can be used to power customer solutions, **it's deemed inappropriate to be used.**
- I talked to Prof. Scaramuzza about the Stereo Event Camera Dataset for Driving Scenarios. We can **freely use** this dataset (their paper should be cited in the code)
- I talked to Li Qingqing from the University of Turku about Multi-Modal Lidars Dataset for Benchmarking General-Purpose Localization and Mapping Algorithms, **we can freely use** this dataset
- Next up, I'll talk to KAIST about their dataset, and keep you posted.

armaganarsln, 2023-04-12:
Thank you Samet, lets discuss the details at TSC. I think we should make Kitti part at least official because we are currently using it for development

mitsudome-r, 2026-03-24:
I'm closing the issue that was inactive for a long time. Feel free to reopen if the issue is still relevant

closed 2026-03-24 by mitsudome-r