autowarefoundation/autoware_universe #2567: Change traffic signal interface to support V2X/V2I

isamu-takagi, 2022-12-22:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Hi, I'm thinking about V2X support in traffic signal interface. Currently, the planning component (traffic_light module in behavior_velocity_planner) has two traffic signal inputs, `perception` and `external`. It looks like we can input V2X information into `external`, but it was actually created for the purpose of overriding vehicle behavior by the operator. So this interface means the operator's decision such as pass and stop rather than the traffic signal color such as green or red, and the pass throuh logic described [here](https://autowarefoundation.github.io/autoware.universe/main/planning/behavior_velocity_planner/traffic-light-design/#limitations) is not performed.

[image]

### Purpose

For the planning component, it doesn't matter if the input is image recognition results or V2X. Therefore, I'll unify the input so that V2X can be handled in the same way as image recognition results.

### Possible approaches

The new architecture is as follows. Merge the image recognition results and V2X before inputting into the planning component. Typically the merge node will use V2X if it is available, otherwise it will use the most confident image recognition result. Also, as described later, light bulbs and traffic signals are treated separately.

[image]

### Definition of done

- [ ] Change the perception output interface.
- [ ] Change the planning input interface.
- [ ] Create the merge node from perception to planning.
- [ ] Create the V2X output interface.

TakaHoribe, 2022-12-23:
>For the planning component, it doesn't matter if the input is image recognition results or V2X. Therefore, I'll unify the input so that V2X can be handled in the same way as image recognition results.

I totally agree. The approach looks good to me as well.

isamu-takagi, 2022-12-26:
@TakaHoribe Thank you. I would like to proceed with this proposal.

As a first step, create the autoware_perception_msgs package in [autoware_msgs](https://github.com/autowarefoundation/autoware_msgs) and define a message type as follows. Note that I changed the definition of traffic signal and traffic light. If there are other good words to distinguish between these, please let me know.

| Type | Current implementation | Related lanelet ID |
| ---- | ---- | ---- |
| TrafficSignal.msg | none | Relation ID (type: regulatory_element) |
| TrafficLight.msg | TrafficSignal.idl | Way ID (type: traffic_light)  |
| TrafficLightElement.msg | TrafficLight.idl | Node ID if possible |

isamu-takagi, 2023-01-12:
I've created a PR to add messages.
https://github.com/autowarefoundation/autoware_msgs/pull/48

yukkysaito, 2023-01-14:
cc @Mingyu1991 @miursh

isamu-takagi, 2023-03-09:
The migration steps are as follows.

[image]

yukkysaito, 2023-03-09:
@isamu-takagi The diagram above is a bit old, so the actual roles and such will be different.
@Mingyu1991 @miursh Could you share the updated figure?

Mingyu1991, 2023-03-13:
@yukkysaito 
Do you mean this figure?
[image]

@isamu-takagi 
There would be two merge nodes within perception.
The "traffic_light_merger" would merge the results of the same traffic light from different cameras.
The "traffic_signal_arbitrator" would merge the results of traffic lights from the same group.

isamu-takagi, 2023-03-14:
Thank you. Updated the diagram.

isamu-takagi, 2023-03-15:
Step 1 is done. https://github.com/autowarefoundation/autoware.universe/pull/2903

KhalilSelyan, 2023-05-16:
@isamu-takagi Any updates ?

isamu-takagi, 2023-05-16:
@KhalilSelyan 
I asked the maintainers of each package to implement it. I hear they are working on it now.
- @TomohitoAndo  for planning (step 2)
- @TomohitoAndo @Mingyu1991 for perception (step 3, 4)

isamu-takagi, 2023-06-22:
Update the step 4 diagram to split merge step by targets. An implementation can perform multiple merge steps on the same node.
[image]

isamu-takagi, 2023-06-28:
I noticed that `TrafficLight.msg` is converted to `TrafficSignal.msg` in perception, and only `TrafficSignal.msg` is used for interface between components. So I will remove `TrafficLight.msg`.

miursh, 2023-07-08:
Merged perception update https://github.com/autowarefoundation/autoware.universe/pull/4050 https://github.com/autowarefoundation/autoware.universe/pull/4125 https://github.com/autowarefoundation/autoware.universe/pull/4176

This is the latest perception node diagaram.

[image]

miursh, 2023-07-26:
@TakaHoribe @TomohitoAndo @isamu-takagi 
Let me just check one thing. Are all works in this issue already done?

isamu-takagi, 2023-07-26:
@miursh I think so.
@TakaHoribe @TomohitoAndo @Mingyu1991 Are there any remaining tasks?

TomohitoAndo, 2023-07-26:
@isamu-takagi @miursh 
The changes for the planning modules haven't been merged yet. ([This PR](https://github.com/autowarefoundation/autoware.universe/pull/4133))
The review for this PR is already done, but scenario simulator doesn't support the new interface yet, so we are waiting for it.
It's scheduled to be done by next week.

TomohitoAndo, 2023-08-23:
Support for V2X in the scenario simulator has been delayed until early September. Thus, the PR for the planning modules is on hold until then.

miursh, 2023-08-29:
Changing the node order of crosswalk_estimator
https://github.com/autowarefoundation/autoware.universe/pull/4734

TomohitoAndo, 2023-09-09:
@isamu-takagi @miursh @TakaHoribe 
The changes for the planning modules have been merged, so I close this issue.
https://github.com/autowarefoundation/autoware.universe/pull/4133

closed 2023-09-09 by TomohitoAndo