autowarefoundation/autoware_universe #2597: The positioning module is unstable for unknown reasons (rosbag2 has been provided for testing)

mikechan0731, 2022-12-28:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi I'm Mike from ITRI

I'm currently using autoware.universe's locator program for locating large buses.

However, the positioning is always abnormal. After many tests and adjustments, there is still no improvement. 

The details are as follows:

### Expected behavior

As mentioned in regular meetings, we hope that large buses can output positioning information smoothly and smoothly during positioning, so that the process of automatic driving is comfortable.

### Actual behavior

But now our large bus will have positioning problems during travel, for example, the positioning is not smooth but shaking, as shown in the following video link:

Video1: [https://1drv.ms/v/s!AnJ4ubRnmXsujepAtTg_SjH237xLqg?e=FxwLMU](https://1drv.ms/v/s!AnJ4ubRnmXsujepAtTg_SjH237xLqg?e=FxwLMU)

Video2: [https://1drv.ms/v/s!AnJ4ubRnmXsujepBy2bgmlmdUxPQUQ?e=6qhyTx](https://1drv.ms/v/s!AnJ4ubRnmXsujepBy2bgmlmdUxPQUQ?e=6qhyTx)

It will also happen that halfway through the positioning error:

Video3: [https://1drv.ms/v/s!AnJ4ubRnmXsujepE9Rijr1uZjKBKtg?e=ttVkaK](https://1drv.ms/v/s!AnJ4ubRnmXsujepE9Rijr1uZjKBKtg?e=ttVkaK)

Video4: [https://1drv.ms/v/s!AnJ4ubRnmXsujepCgb9so2WKTomtaQ?e=c6HDt6](https://1drv.ms/v/s!AnJ4ubRnmXsujepCgb9so2WKTomtaQ?e=c6HDt6)

### Steps to reproduce

The localization parameters used are provided here as a reference:
- voxel size: 2.0, 2.0, 2.0
- downsample pointclouds: 10,000
- ndt_scan_matcher step_size: 2.1

The ros2bag link in the corresponding positioning process is provided here. We use the topic /LidarFrontTop as the positioning input point cloud.

ROSBAG2 Link: [https://1drv.ms/u/s!AnJ4ubRnmXsujeoj9Au-pDLAfEl2mw?e=wujOP1](https://1drv.ms/u/s!AnJ4ubRnmXsujeoj9Au-pDLAfEl2mw?e=wujOP1)

### Versions

OS: Ubuntu 20.04
ROS2 Galactic
Autoware Universe

### Possible causes

We speculate that these problems occur because the frequency of the point cloud information provided for positioning is too low. If you use `ros2 topic hz /LidarFrontTop` (point cloud input for positioning), you will find that the frequency is only about 3~5Hz.

This part extends the second problem: We found that on the host, if two subscribers register the same topic at the same time, it will cause serious transmission slowdown.

For example, /LidarFrontTop If there is only one subscriber, the transmission speed is 20Hz. But if another subscriber is turned on, the transmission speed will immediately drop to 3~5Hz, even if I just start two `ros2 topic hz /LidarFrontTop` at the same time, this will happen.

I would like to ask for advice on the above two problems: 

1. The positioning is not smooth but shaking,
2. The problem of transmission slowdown, 

Any suggestion is welcomed…Thank you!

YamatoAndo, 2023-01-11:
@mikechan0731 Hi!
There are several things that need to be improved, but first, the TF for base_link to LiDAR should be set.

In the rosbag you provided, the TF is not set properly, so there is a large error in the estimated position of EKF on curves, as shown in the video below.

https://user-images.githubusercontent.com/13819566/211896698-885d5aad-97c5-4198-80cf-eaff92f36b11.mp4

blue line: estimated positioin of EKF
green line: estimated positioin of NDT
※Normally, the estimated position of the NDT and the estimated position of the EKF should be almost the same

For example, if the LiDAR point cloud is estimated with an offset of 4.3m in the x-axis direction, it will be as shown in the movie below.
※The 4.3m is a value I have roughly estimated, so please make sure you get the correct value.

https://user-images.githubusercontent.com/13819566/211896927-7ce0974a-b0ac-4bb0-a899-9bb623c13aad.mp4

xmfcx, 2023-02-07:
Can you follow the suggestion here [Error accumulation in EKF localizer](https://github.com/autowarefoundation/autoware.universe/issues/1426#issuecomment-1195070643) for adjusting the lidar to base-link calibration to see if it would help with the localization performance?

mitsudome-r, 2023-02-07:
@mikechan0731 
I'm afraid we don't have any calibration tool that can calculate the value automatically.
You first have to measure the relative position from your base_link (center of the rear wheels) to lidar position. You can use the design value for the initial position, then use ekf_localizer to adjust yaw angle error as mentioned in Fatih's comment.

To set base_link->lidar tf, you can run the following command:
```
ros2 run tf2_ros static_transform_publisher --x 0.0 --y 0.0 --z -.0 --roll 0.0 --pitch 0.0 --yaw 0.0 --frame-id base_link --child-frame-id lidar
```

mikechan0731, 2023-02-14:
Hello,

As mentioned in the meeting, the phenomenon of "Dropping FPS when two subscriber subscribe to same 1 topic" will occur at present,

At present, I think it may be due to the DDS setting problem under different modules, so here is a brief description of the DDS situation I set here.

1. I use a separate terminal to launch the mod
2. When the following terminal is opened, if two subscribers request to read the same topic at the same time, the situation of 20Hz -> 4 Hz will occur. For example, /LidarFrontTop =20Hz, but if you open two terminals at the same time and start `ros2 topic hz /LidarFrontTop` at the same time, it will be reduced to 4Hz (rviz will also have the same situation).

3. ~/.bashrc used by each terminal:
```
source /opt/ros/galactic/setup.bash
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
export PATH=/usr/local/cuda/bin${PATH:+:${PATH}}
export LD_LIBRARY_PATH=/usr/local/cuda/lib64${LD_LIBRARY_PATH:+:${LD_LIBRARY_PATH}}
```

4. I have opened a total of 10 terminals, the list is as follows, and cpu usage will be 80~90%:
* terminal 1:
ros2 run tf2_ros static_transform_publisher 4.3 0.0 2.0 0.0 0.0 0.0 base_link lidar

* terminal2.
ros2 launch ros2_ouster driver_launch_left.py

* terminal3.
ros2 launch ros2_ouster driver_launch_right.py

* terminal4.
ros2 launch ros2_ouster driver_launch_top.py

* terminal5.
ros2 run trimble_grabber trimble_grabber

* terminal6.
source ~/shukang_doc/docker_test/export_cyclonedds_camera_1.sh
ros2 launch lidar_grabber lidar_grabber.launch.xml

* terminal7.
source ~/shukang_doc/docker_test/export_cyclonedds_camera_1.sh
ros2 launch autoware_launch autoware.launch.xml map_path:=/home/lidar/Downloads/itri-map vehicle_model:=cbus_vehicle sensor_model:=sample_sensor_kit

* terminal8.
source ~/shukang_doc/docker_test/export_cyclonedds_camera_1.sh
ros2 launch edge_detection edge_detection.launch.xml

* terminal9.
source ~/shukang_doc/docker_test/export_cyclonedds_camera_3.sh
ros2 run objmsg_converter objmsg_converter

* terminal10.
source ~/shukang_doc/docker_test/export_cyclonedds_camera_3.sh
ros2 launch lidar_centerpoint lidar_centerpoint.launch.xml model_name:=centerpoint

The corresponding DDS settings are provided as follows:
### export_cyclonedds_camera_1.sh ###
```
export CYCLONEDDS_URI=file:///home/lidar/shukang_doc/docker_test/cyclonedds_camera_1.xml
```

### cyclonedds_camera_1.xml ###
```
<?xml version="1.0" encoding="UTF-8" ?>
   <CycloneDDS xmlns="https://cdds.io/config" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="https://cdds.io/config https://raw.githubusercontent.com/eclipse-cyclonedds/cyclonedds/master/etc/cyclonedds.xsd">
     <Domain id="any">
         <General>
             <NetworkInterfaceAddress>enp0s31f6</NetworkInterfaceAddress>
         </General>
         <Internal>
             <MinimumSocketReceiveBufferSize>900MB</MinimumSocketReceiveBufferSize>
         </Internal>
     </Domain>
</CycloneDDS>
```

### export_cyclonedds_camera_3.sh ###
```
export CYCLONEDDS_URI=file:///home/lidar/shukang_doc/docker_test/cycloneds_camera_3.xml
```

### cyclonedds_camera_3.xml ###
```
<?xml version="1.0" encoding="UTF-8" ?>
   <CycloneDDS xmlns="https://cdds.io/config" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="https://cdds.io/config https://raw.githubusercontent.com/eclipse-cyclonedds/cyclonedds/master/etc/cyclonedds.xsd">
     <Domain id="any">
         <General>
             <NetworkInterfaceAddress>enp1s0f0</NetworkInterfaceAddress>
         </General>
         <Internal>
             <MinimumSocketReceiveBufferSize>900MB</MinimumSocketReceiveBufferSize>
         </Internal>
     </Domain>
</CycloneDDS>
```

xmfcx, 2023-02-21:
This is being discussed in https://discord.com/channels/953808765935816715/1074965040307916850

mikechan0731, 2023-03-17:
Hi here, first of all, I would like to thank @xmfcx for his great help and tireless testing on this issue. The rigorous attitude towards solving bugs is what I need to learn. Thank you very much!

Based on the above responses, I have adjusted the following items focusing on DDS corrections:

```
net.core.rmem_max=2147483647
net.ipv4.ipfrag_time=3
net.ipv4.ipfrag_high_thresh=134217728
```

Then either restart the pc or run:

```
sudo sysctl -w net.core.rmem_max=2147483647
sudo sysctl net.ipv4.ipfrag_time=3
sudo sysctl net.ipv4.ipfrag_high_thresh=134217728
```

Make sure .bashrc has the following:

```
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
export ROS_LOCALHOST_ONLY=1
export CYCLONEDDS_URI=/home/lidar/cyclonedds.xml
```

And cyclonedds.xml is created by your suggestion.

Then I try to turn on autoware using this command
`ros2 launch autoware_launch autoware.launch.xml map_path:=/home/lidar/Download/itri-map vehicle_model:=cbus_vehicle sensor_model:=sample_sensor_kit use_sim_time:=true`

But the following error occurs, and Rviz2 does not run automatically, so I can't continue testing.

```
[INFO] [launch]: All log files can be found below /home/lidar/.ros/log/2023-03-17-18-20-17-636546-lidar-4589
[INFO] [launch]: Default logging verbosity is set to INFO
1679048418.199332 [0]       ros2: selected interface "lo" is not multicast-capable: disabling multicast
[INFO] [component_container_mt-1]: process started with pid [4607]
[INFO] [robot_state_publisher-2]: process started with pid [4609]
[INFO] [component_container-3]: process started with pid [4611]
[INFO] [map_hash_generator-4]: process started with pid [4613]
[INFO] [map_height_fitter-5]: process started with pid [4615]
[INFO] [pose_initializer_node-6]: process started with pid [4617]
[INFO] [automatic_pose_initializer-7]: process started with pid [4619]
[INFO] [component_container_mt-8]: process started with pid [4621]
[INFO] [ndt_scan_matcher-9]: process started with pid [4623]
[ERROR] [ndt_scan_matcher-9]: process has died [pid 4623, exit code -6, cmd '/home/lidar/autoware_ws3/install/ndt_scan_matcher/lib/ndt_scan_matcher/ndt_scan_matcher --ros-args -r __node:=ndt_scan_matcher -r __ns:=/localization/pose_estimator --params-file /tmp/launch_params_fxy3orc5 --params-file /home/lidar/autoware_ws3/install/tier4_localization_launch/share/tier4_localization_launch/config/ndt_scan_matcher.param.yaml -r points_raw:=/localization/util/downsample/pointcloud -r ekf_pose_with_covariance:=/localization/pose_twist_fusion_filter/biased_pose_with_covariance -r pointcloud_map:=/map/pointcloud_map -r ndt_pose:=/localization/pose_estimator/pose -r ndt_pose_with_covariance:=/localization/pose_estimator/pose_with_covariance -r regularization_pose_with_covariance:=/sensing/gnss/pose_with_covariance'].
[INFO] [gyro_odometer-10]: process started with pid [4625]
[INFO] [ekf_localizer-11]: process started with pid [4627]
[INFO] [stop_filter-12]: process started with pid [4629]
[INFO] [twist2accel-13]: process started with pid [4631]
[INFO] [localization_error_monitor-14]: process started with pid [4633]
[INFO] [component_container_mt-15]: process started with pid [4635]
[INFO] [web_server.py-16]: process started with pid [4653]
[INFO] [initial_pose_adaptor-17]: process started with pid [4718]
[INFO] [routing_adaptor-18]: process started with pid [4720]
[INFO] [awapi_awiv_adapter-19]: process started with pid [4735]
[INFO] [component_container-20]: process started with pid [4744]
[INFO] [component_container_mt-21]: process started with pid [4767]
[ERROR] [component_container_mt-21]: process has died [pid 4767, exit code -6, cmd '/opt/ros/galactic/lib/rclcpp_components/component_container_mt --ros-args -r __node:=autoware_iv_adaptor -r __ns:=/autoware_api/external --params-file /tmp/launch_params_yvyvkmeb'].
[INFO] [component_container_mt-22]: process started with pid [4770]
[ERROR] [component_container_mt-22]: process has died [pid 4770, exit code -6, cmd '/opt/ros/galactic/lib/rclcpp_components/component_container_mt --ros-args -r __node:=autoware_iv_adaptor -r __ns:=/autoware_api/internal --params-file /tmp/launch_params_buq83pa7'].
[INFO] [relay-23]: process started with pid [4789]
[ERROR] [relay-23]: process has died [pid 4789, exit code -6, cmd '/opt/ros/galactic/lib/topic_tools/relay /api/autoware/set/traffic_signals /external/traffic_light_recognition/traffic_signals --ros-args -r __node:=traffic_signals -r __ns:=/autoware_api/internal --params-file /tmp/launch_params_t7a46y5k'].
[INFO] [relay-24]: process started with pid [4795]
[INFO] [relay-25]: process started with pid [4799]
[INFO] [rosbridge_websocket-26]: process started with pid [4802]
[INFO] [rosapi_node-27]: process started with pid [4807]
[INFO] [rviz2-28]: process started with pid [4809]
[ERROR] [relay-24]: process has died [pid 4795, exit code -6, cmd '/opt/ros/galactic/lib/topic_tools/relay /api/autoware/set/intersection_state /planning/scenario_planning/lane_driving/behavior_planning/behavior_velocity_planner/input/external_intersection_states --ros-args -r __node:=intersection_states -r __ns:=/autoware_api/internal --params-file /tmp/launch_params_2qem_76t'].
[... 481 lines cut ...]
[rosapi_node-27] rclpy._rclpy_pybind11.RCLError: error creating node: rcl node's rmw handle is invalid, at /tmp/binarydeb/ros-galactic-rcl-3.1.3/src/rcl/node.c:413
[... cut ...]
[component_container-3] [INFO] [1679048420.808673804] [map.map_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<VectorMapTFGeneratorNode>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/map/vector_map_tf_generator' in container '/map/map_container'
[component_container-3] [INFO] [1679048421.135998095] [map.vector_map_tf_generator]: broadcast static tf. map_frame:map, viewer_frame:viewer, x:2282.18, y:41438.2, z:-1.9301

```

Want to know what is the reason for this phenomenon? Thank you very much!!

xmfcx, 2023-03-18:
You see it says: `ros2: selected interface "lo" is not multicast-capable: `

To resolve it, you need to run `sudo ip link set lo multicast on` every time the you restart the pc. (source: https://autowarefoundation.github.io/autoware-documentation/main/installation/additional-settings-for-developers/#enabling-localhost-only-communication )

You could also create a service that runs that when ubuntu boots, we will probably add it to the documentation in the future.

mikechan0731, 2023-03-20:
Hi @xmfcx ,
I have test the code with `sudo ip link set lo multicast on` , it's now ok for localizaiton, but topic: /downsample still 5Hz.

[image]

Wondering if this is actually normal, or is there something else I haven't noticed?

Thank you so much!

xmfcx, 2023-03-20:
Can you check `ros2 topic hz /LidarFrontTop/Raw` when autoware is **NOT** running?

And then re-run  `ros2 topic hz /LidarFrontTop/Raw` when autoware is running?

And also make sure you are playing exactly the same bag file with same parameters:
```
ros2 bag play rosbag2_2022_09_22-11_56_33/ --remap /LidarAll:=/LidarFrontTop  --clock 100 -l
```

mikechan0731, 2023-03-21:
Hi @xmfcx, Here is two video I just test on our machine.

https://youtu.be/cimgqqQ27o8

https://youtu.be/hvnnnntFyaw

These video are almost the same but I just want to show the real case when I boot autoware localization.

liuXinGangChina, 2023-03-21:
hi mike , it's lucas from autocore.ai
It will be better if you can provide a rosbag of the video you upload today, so @cyn-liu can help u with it
by the way, is the localization module work well, especiall in the t-junction that you mentioned before

xmfcx, 2023-03-21:
> hi mike , it's lucas from autocore.ai It will be better if you can provide a rosbag of the video you upload today, so @cyn-liu can help u with it by the way, is the localization module work well, especiall in the t-junction that you mentioned before

Hi @liuXinGangChina All the steps to reproduce and the bag files are included in my post: https://github.com/autowarefoundation/autoware.universe/issues/2597#issuecomment-1469398009

On my machine, everything seems to be working just fine. Their machine has some networking issues probably. Since after launching Autoware, the `ros2 topic list` command doesn't show all the topics.

Right now this is not a localization issue, rather ros2 performance and configuration issue in general.

xmfcx, 2023-03-31:
Hello,
This evening I had a chance to access ITRI's lidar pc with help of @mikechan0731 with a remote desktop control software.

They had reinstalled Ubuntu 20.04 and installed everything from scratch.
When I got the pc, following were missing:
- cyclonedds configuration
- dds configuration (done with `sudo gedit /etc/sysctl.d/10-cyclone-max.conf`, [explained in my post](https://github.com/autowarefoundation/autoware.universe/issues/2597#issuecomment-1469398009))
- [ccache configuration](https://autowarefoundation.github.io/autoware-documentation/main/how-to-guides/advanced-usage-of-colcon/#using-ccache) (not relevant for this issue but really helps with the compilation times, i find it almost mandatory)
- [colorized ros2 terminal output and custom format for the logger output](https://autowarefoundation.github.io/autoware-documentation/main/installation/additional-settings-for-developers/#colorizing-logger-output) (again not mandatory but i need my colorful output)

I added them but still surprised that the downsampled point cloud output was 4-5Hz.

```bash
lidar@lidar:~$ screenfetch -n
 OS: Ubuntu 20.04 focal
 Kernel: x86_64 Linux 5.15.0-69-generic
 Resolution: 1920x1080
 DE: GNOME 3.36.5
 WM: Mutter
 Disk: 114G / 912G (14%)
 CPU: Intel Xeon E-2278G @ 16x 5GHz [48.0°C]
 GPU: NVIDIA GeForce RTX 2080 Ti
 RAM: 3302MiB / 64158MiB
```

System specs are OK and `gnome-system-monitor` didn't show much cpu activity when running things.

Then I got suspicious that maybe the downsampling node was not compiled with `Release` or `RelWithDebInfo` tags. When I checked the `~/.bash_history` my suspicions turned out to be correct.

```bash
lidar@lidar:~$ cat ~/.bash_history | grep "colcon build"
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
colcon build --symlink-install --continue-on-error
```

So it was started to be built with the release tag but in the end, the tag was dropped.

Then I've built it with `colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_EXPORT_COMPILE_COMMANDS=1 --continue-on-error` and then the issue was fixed.

[image]

ajay1606, 2023-03-31:
@xmfcx Excellent, Indeed it was very useful as i was also following this issues since quite long time and happy to see this got fixed.

mikechan0731, 2023-04-07:
Hi, thanks to @xmfcx  for the continued testing, I'm doing a rebuild test today,

First, I used the original build command `colcon build` to compile the program, resulting in the low FPS phenomenon of the original downsample module:

https://youtu.be/wWB4-_YJbow

Later, according to @xmfcx  's suggestion, I re-used the build command

 `colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_EXPORT_COMPILE_COMMANDS=1 --continue-on-error` 

to compile the program, and the original downsample module.

The phenomenon of low FPS (5Hz) is immediately increased to 30Hz, and the output of /pose also conforms to the output frequency of point cloud original file output frequency of 15Hz:

https://youtu.be/wWB4-_YJbow

I think this problem can be considered solved, thanks to everyone who helped, especially @xmfcx, thank you!

closed 2023-04-07 by mikechan0731