autowarefoundation/autoware_universe #2605: Add a validation node for the planning trajectory

TakaHoribe, 2023-01-02:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Sometimes, a faulty path is published due to the poor quality of the planning. This can cause the vehicle to behave abnormally in terms of steering, acceleration, and braking. It is important to continually improve the quality of the planning, but we also need to implement a validation system to ensure the quality of the paths being published.

### Purpose

To avoid publishing abnormal paths.

### Possible approaches

A planning validator node is introduced at the final stage of the planning process. The node validates the trajectory and stops publishing if it is invalid.

The planning node diagram would be like below:
`planning_modules` -> `planning_validator` -> `control`.

### Definition of done

A validation mechanism of the planning is in the autoware.

TakaHoribe, 2023-01-02:
Related:

- https://github.com/autowarefoundation/autoware.universe/pull/1947
- https://github.com/autowarefoundation/autoware.universe/pull/2604
- https://github.com/tier4/autoware_launch/pull/675

TakaHoribe, 2023-01-02:
cc. @yukkysaito

yukkysaito, 2023-01-10:
I think it's a good idea to add one.
However, I would like to separate what is verified online and what is verified offline (like CI)
Otherwise, it would be inefficient to run a huge verification online.

TakaHoribe, 2023-01-18:
> to separate what is verified online and what is verified offline (like CI)

Ok. In the first phase, the validator node focus only on the online checks (curvature, acceleration, etc). The offline validation item could be something regarding the map, kinematics parameters, etc. We should discuss this further.

mehmetdogru, 2023-03-29:
@TakaHoribe I guess this issue can be closed?

TakaHoribe, 2023-05-11:
This can be closed with these PRs.
https://github.com/autowarefoundation/autoware.universe/issues/2605#issuecomment-1368926141

closed 2023-05-11 by TakaHoribe