autowarefoundation/autoware_universe #2607: The node of behavior_path_planner dies in some conditions.

shulanbushangshu, 2023-01-03:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The node of behavior_path_planner dies  when I set target in change lane.The simulation is shown as below:
[image]

[image]

[image]

There is no enough distance for the second solution of lane change.But the node shall not die after setting target.

### Expected behavior

The node of behavior_path_planner shall not die  in any conditions of lanes and target.

### Actual behavior

The node of behavior_path_planner dies  in some conditions of lanes and target.

### Steps to reproduce

1.Start autoware planning simulator;
2. Set initial position
3. Set target in the change lane(there are more than two lanes and the distance of lane change shall is not enough)

### Versions

1.OS: Ubuntun 20.04
2.ROS2: Galactic
3.Autoware: Autoware.universe:main

zulfaqar-azmi-t4, 2023-01-04:
Hi @shulanbushangshu 
Thank you very much for the report.
The bug probably is related to the https://github.com/autowarefoundation/autoware.universe/issues/2473

There is a bit delay due to some circumstances, but I'll try fix it as soon as possible.

zulfaqar-azmi-t4, 2023-01-10:
@shulanbushangshu 
Can you help to confirm whether #2615 solve this issue ?
thank you very much in advance.

shulanbushangshu, 2023-01-10:
@zulfaqar-azmi-t4 
I  change the codes with https://github.com/autowarefoundation/autoware.universe/pull/2615 and finish the simulation.

case1:
[image]

case2:
I set the target in the third lane and the distance of lane change shall is not enough.
[image]

The  node of behavior_path_planner  still died.

zulfaqar-azmi-t4, 2023-01-11:
@shulanbushangshu 
Is this the latest commit in the upstream? I seems to me the lane change path is a bit weird.
I couldn't reproduce the error, therefore, could you share to me the lanelet map file?

zulfaqar-azmi-t4, 2023-03-14:
@shulanbushangsh
is there any update regarding this issue?
If not, I might close the thread.

shulanbushangshu, 2023-03-24:
@zulfaqar-azmi-t4 
The lanelet2_map:
[lanelet2_map.zip](https://github.com/autowarefoundation/autoware.universe/files/11059967/lanelet2_map.zip)

The position of testing lane:
[image]

shulanbushangshu, 2023-04-07:
I test the issue in the autoware of humble.There is still problem and the test is as follows:
[image]
[image]
I have updated the pr

shulanbushangshu, 2023-04-13:
It is fixed in the newest code. I close the issue

closed 2023-04-13 by shulanbushangshu