autowarefoundation/autoware_universe #2633: The lane_following can not operate normally in some conditions

shulanbushangshu, 2023-01-11:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

When I use the map to do simulation about lane_following, there are three cases.

case1:
[image]

case2:
[image]

case3:
[image]

In the case1 and case3,the mode can work normally.But in the case2,the mode is unable to operate normally.

### Expected behavior

In the conditions of case2,the lane_following is unable to operate normally.

### Actual behavior

In the conditions of case2,the lane_following is unable to operate normally.

### Steps to reproduce

1.Use the map of demo and start autoware planning simulator;
2. Set initial position in the lane of 34976
3. Set target in the lane of 35016

### Versions

1.OS: Ubuntun 20.04
2.ROS2: Galactic
3.Autoware: Autoware.universe:main

### Possible causes

In the path  based on sampling distance by parameters,there is the lane which  connects  to  the start lane.

zulfaqar-azmi-t4, 2023-01-13:
@shulanbushangshu 
This PR https://github.com/autowarefoundation/autoware.universe/pull/2648 might interest you.

mitsudome-r, 2023-01-17:
@shulanbushangshu Could you explain what you mean by "not operating normally"? Is it not engaging?

shulanbushangshu, 2023-01-18:
@mitsudome-r No,it is engaged.I test the new code after the pr https://github.com/autowarefoundation/autoware.universe/pull/2648 and the result is that the path can not get the target lane as below shown:
[image]

purewater0901, 2023-01-18:
@shulanbushangshu Thank you for your reporting. I also encountered the same issue, but this problem is not related to [this PR](https://github.com/autowarefoundation/autoware.universe/pull/2648). I think the root cause of the problem comes from either Lanelet library or the route handler. The route handler(or Lanelet library) gives incorrect Lanelet sequences.

purewater0901, 2023-01-18:
It seems [this funciton](https://github.com/autowarefoundation/autoware.universe/blob/main/planning/route_handler/src/route_handler.cpp#L715-L729) chooses wrong lanelet in this case

shulanbushangshu, 2023-01-18:
@purewater0901 Yes,this problem is not related to https://github.com/autowarefoundation/autoware.universe/pull/2648. The function of getNextLaneletWithinRoute() has no judgement when chose the next lane.
[image]

purewater0901, 2023-01-18:
@shulanbushangshu Yes, when you make a "half-looped" route, it won't work correctly. In order to fix this problem, we have to re-design the route handler.