autowarefoundation/autoware_universe #2636: DevOps Dojo: ROS Node Configuration (Phase 2)

kasperornmeck, 2023-01-11:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

DevOps: Ros Node Configuration will address the topic of how ROS nodes are configured. Guidelines, documentation and changes to configuration files will be made, based on best-practices from cloud-native and software defined development methodologies.

Phase 1 - Motivation
- https://github.com/autowarefoundation/autoware.universe/issues/2631

Phase 2 - Definition
- [Todo: Add link]

Phase 3 - Execution
- [Todo: Add link]

Phase 4 - Integration
- [Todo: Add link]

### Purpose

Phase 2 - Definition
- define structure and milestone(s)
- identify gaps in current software and developers’ skills
- produce actionable items
- consensus on solution

### Possible approaches

From https://12factor.net/config:
- “An app’s config is everything that is likely to vary between deploys (staging, production, developer environments, etc).”

We want to be able to configure any parameter so that the ROS node can be used in different deploys.

Use following ROS nodes to use for evaluation of changes to be made:
- lidar apollo segmentation tvm
- localization

Parameters are declared using
  - [declare_parameter(...)](https://docs.ros2.org/dashing/api/rclcpp/classrclcpp_1_1Node.html#a26508ac4c6ca671ed25802a6a1ca6d23)
All parameters must be listed in the config file
 

### Definition of done

Once consensus on how to structure ROS node configurations has been achieved and actionable items produces.

kasperornmeck, 2023-01-11:
We want to avoid using the default values in the [declare_parameter(...)](https://docs.ros2.org/dashing/api/rclcpp/classrclcpp_1_1Node.html#a64df64d004b6263c3c04f42179fc8f52) function. This is to ensure that the ROS node configuration file has all the necessary parameters. Is there a way to throw an exception if [declare_parameter(...)](https://docs.ros2.org/dashing/api/rclcpp/classrclcpp_1_1Node.html#a64df64d004b6263c3c04f42179fc8f52) expects a parameter not existing in the ROS node configuration file?

kasperornmeck, 2023-01-18:
Two options:
1. Rule that any configurable parameter exists in the yaml file
2. Throw an exception if a parameter is not in the yaml file

We don't want to create unnecessary work.

kasperornmeck, 2023-01-20:
@mitsudome-r, after discussion with @ambroise-arm, we propose that we go with option number 2. That is:
- Throw an exception if a parameter is not in the parameter configuration yaml file

This is because [declare_parameter(...)](https://docs.ros.org/en/ros2_packages/humble/api/rclcpp/generated/classrclcpp_1_1Node.html#_CPPv4N6rclcpp4Node17declare_parameterERKNSt6stringEN6rclcpp13ParameterTypeERKN14rcl_interfaces3msg19ParameterDescriptorEb) states the following:
[image]

That is, if _default_value_ is not set and the parameter configuration file does not provide the parameter or the right format for the parameter, an exception will be thrown. This then solves our issue of matching the expected parameters provided and the actual parameters provided.

This gives us a single source of truth for parameter configuration. Can we agree on
- not assigning _default_value_ a value
- parameter configuration yaml file is comprehensive
  - I.e., contains all _declare_parameter()_ values

?

kasperornmeck, 2023-01-20:
If we agree upon having a singe source of truth in the parameter configuration yaml file, we should then decide on a certain format for the yaml file so that it can be parsed and interpreted to an .md file to be used with the documentation.

For example, in [ndt_scan_matcher](https://autowarefoundation.github.io/autoware.universe/main/localization/ndt_scan_matcher/#core-parameters) there is a table with _Core Parameters_. Let's assume the parameter configuration yaml file [ndt_scan_matcher.param.yaml](https://github.com/autowarefoundation/autoware.universe/blob/main/localization/ndt_scan_matcher/config/ndt_scan_matcher.param.yaml) was parsed and interpreted as an .md file. We would then have a single source for both ROS node configuration and documentation.

Can we agree upon
- using parameter configuration yaml file for both
  - configuring the ROS node
  - used in ROS node documentation

?

kasperornmeck, 2023-01-20:
To give an idea of the yaml to .md, see [yaml-to-table/](https://github.com/ishswar/yaml-to-table/blob/master/README.md), (I'm not suggesting we use this).

doganulus, 2023-01-31:
I like the idea of having unique and hierarchical names for each and every parameter in the system. For example, I think the dot notation `autoware.module.node.parent.variable` can be adopted as other similar applications. Further, it is a good idea to specify certain other attributes next to the implementation. 

- autoware.module.node.parent.variable.value // Current value of the variable
- autoware.module.node.parent.variable.default_value // Default value of the variable
- autoware.module.node.parent.variable.type  // Type of the variable -- e.g. bool, number, string, enum, custom
- autoware.module.node.parent.variable.range // List of allowed values (if the type is enum)
- autoware.module.node.parent.variable.range.min // Minimum allowed value (if the type is number)
- autoware.module.node.parent.variable.range.max // Maximum allowed value (if the type is number)

Not sure if ROS gives that flexibility to define parameters in such a hierarchical way but it would be good to explore possibilities under this Dojo. Definitely, it would improve automated documentation generation using any templating engine and speed up overall development.

kasperornmeck, 2023-02-01:
Autoware.Universe architecture w. nodes: https://autowarefoundation.github.io/autoware-documentation/main/design/autoware-architecture/node-diagram/overall-node-diagram-autoware-universe.drawio.svg

kasperornmeck, 2023-02-02:
In the documentation, list all possible parameter values as defined by _Class Parameter Value_. See [Enum ParameterType](Enum ParameterType[](https://docs.ros.org/en/ros2_packages/humble/api/rclcpp/generated/enum_namespacerclcpp_1a1b307a4b7368b68f325812cd2aeb8784.html#enum-parametertype).

Question: is this list of parameter extensive?

kasperornmeck, 2023-02-07:
An alternative idea is to use https://github.com/PickNikRobotics/generate_parameter_library which 
- Pro: good solution which also checks that 
- Cons: we create a dependency on a 3rd party repo

Action-point: create an example for e.g., https://github.com/autowarefoundation/autoware.universe/blob/main/localization/stop_filter/src/stop_filter.cpp to show how PickNikRobotic's generate parameter library could be used.

Create discussion here: https://github.com/orgs/autowarefoundation/discussions/categories/design
See: https://github.com/orgs/autowarefoundation/discussions/3171#discussioncomment-4597112

mitsudome-r, 2023-02-08:
@kasperornmeck 
It looks like the contribution guide was updated for parameter in https://github.com/autowarefoundation/autoware-documentation/pull/271/files. 

It now includes the statement: 
```
It is recommended not to set default values when declaring ROS parameters to avoid unintended behaviors due to accidental use of default values.
Instead, set parameters from configuration files named `*.param.yaml`.
```

For the future packages that will be added to autoware, the above rule should be applied.
For packages that is currently using default values for parameters, we can fix them through this dojo activity.

kasperornmeck, 2023-02-10:
@mitsudome-r 
This is a good start, we're currently evaluating the PickNikRobotics generate parameter library, see https://github.com/orgs/autowarefoundation/discussions/3260.

This would remove the use of _declare_parameter(...)_ completely.

kasperornmeck, 2023-02-13:
Update: https://github.com/orgs/autowarefoundation/discussions/3260#discussioncomment-4964594

kasperornmeck, 2023-02-24:
Find final proposal here: https://github.com/orgs/autowarefoundation/discussions/3288

kasperornmeck, 2023-03-09:
Find final2 proposal here: https://github.com/orgs/autowarefoundation/discussions/3325

kasperornmeck, 2023-03-27:
Find final3 proposal here: https://github.com/orgs/autowarefoundation/discussions/3371

closed 2023-04-14 by kaspermeck-arm