autowarefoundation/autoware_universe #2652: DevOps Dojo: ROS Node Configuration (Phase 4)

kasperornmeck, 2023-01-13:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

DevOps: Ros Node Configuration will address the topic of how ROS nodes are configured. Guidelines, documentation and changes to configuration files will be made, based on best-practices from cloud-native and software defined development methodologies.

Phase 1 - Motivation
- https://github.com/autowarefoundation/autoware.universe/issues/2631

Phase 2 - Definition
- https://github.com/autowarefoundation/autoware.universe/issues/2636

Phase 3 - Execution
- https://github.com/autowarefoundation/autoware.universe/issues/2651

Phase 4 - Integration
- https://github.com/autowarefoundation/autoware.universe/issues/2652

### Purpose

Phase 4 - Integration:
- scale out solution into system
- system wide test and verification
- education and documentation

### Possible approaches

TBD.

### Definition of done

TBD.

kaspermeck-arm, 2023-06-29:
https://github.com/autowarefoundation/autoware.universe/pull/3992

kaspermeck-arm, 2023-07-19:
Want to participate?
- Call for volunteers https://github.com/orgs/autowarefoundation/discussions/3636
- Discord https://discord.com/channels/953808765935816715/1131298526501027972

ahuazuipiaoliang, 2023-07-27:
@kaspermeck-arm I think the Phase 4 is updating the parameter configuration of ROS nodes? Am I correct?

kaspermeck-arm, 2023-07-27:
@ahuazuipiaoliang - yes, that's correct! Phase 4 (integration) includes the following:
- scale out solution into system
- system wide test and verification
- education and documentation

Are you interested in working with the node config?

kaspermeck-arm, 2023-08-14:
PR to update coding guidelines to correct schema template:
* https://github.com/autowarefoundation/autoware-documentation/pull/434

kaspermeck-arm, 2023-09-21:
@mitsudome-r to seek alignment in the AWF with regards to copyright
- configuration files don't have copyright notices
- schema files don't have copyright notices
- should we change copyright to only Autoware Foundation?

mitsudome-r, 2023-09-21:
For now we would like to stick with the following:
1. We don't add copyright notice to parameter files (.param.yaml) or schema files (.schema.json) 
2. We keep the current copyright notice for the existing files

TODO:
1. @mitsudome-r to raise the copyright notice discussion in TSC
2. We can consider applying the proposal mentioned in this [thread](https://github.com/autowarefoundation/autoware/issues/2420#issuecomment-1166428358) to use `Autoware Contributors` for copyright notice.
3. If approved, we can change all copyright notice in all the existing files.