autowarefoundation/autoware_universe #2653: DevOps Dojo: Developer (Phase 1)

kasperornmeck, 2023-01-14:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

DevOps Dojo: Developer will introduce Docker Compose, remote development and VS Code into the AWF community. These modern tools and methodologies will improve the efficiency and availability for our developers.

Phase 1 - Motivation
- (Add link to Phase 1)

Phase 2 - Definition
- (Add link to Phase 2)

Phase 3 - Execution
- (Add link to Phase 3)

Phase 4 - Integration
- (Add link to Phase 4)

### Purpose

Phase 1 - Motivation
- clear purpose
  - ...
- outline proposed work item
  - ...
- identify beneficiaries, stakeholders and shareholders
  - ...

### Possible approaches

TBD.

### Definition of done

TBD.

VRichardJP, 2023-01-20:
I don't know if it is of any help, but here is my dev environment setup for autoware using vscode devcontainer:

- .devcontainer/devcontainer.json
```json
{
  "name": "autoware_dev",

  "dockerComposeFile": "docker-compose.yaml",
  "service": "autoware_dev",

  "extensions": [
    "ms-azuretools.vscode-docker",
    "ms-vscode.cpptools",
    "ms-vscode.cpptools-extension-pack",
    "llvm-vs-code-extensions.vscode-clangd",
    "ms-vscode.cmake-tools",
    "ms-iot.vscode-ros",
    "dotjoshjohnson.xml",
    "davidanson.vscode-markdownlint",
    "donjayamanne.python-extension-pack",
    "wmaurer.change-case",
    "fabiospampinato.vscode-diff"
  ],

  "initializeCommand": "xhost +local:docker",

  "settings": {
    "terminal.integrated.profile.linux": {
      "bash": {
        "source": "/bin/bash"
      }
    },
    "terminal.integrated.defaultProfile.linux": "bash",
    "python.pythonPath": "/usr/bin/python3",
    "files.watcherExclude": {
      "**/.git/objects/**": true,
      "**/.git/subtree-cache/**": true,
      "**/node_modules/*/**": true
    }  
  },

  "workspaceFolder": "/home/dev/autoware"
}
```

- .devcontainer/docker-compose.yaml
```yaml
version: "3"
services:
  autoware_dev:
    build: 
      context: .
      dockerfile: Dockerfile
      args:
        ROS_DISTRO: humble
        TAG: humble-latest-cuda
    environment:
      - DISPLAY
    volumes: 
      - ..:/home/dev/autoware
      - ~/data:/home/dev/data
      - ~/.Xauthority:/home/sig/.Xauthority
      - /tmp/.X11-unix:/tmp/.X11-unix
      - /lib/modules:/lib/modules
      - /dev:/dev
    stdin_open: true
    tty: true
    security_opt:
      - seccomp:unconfined
    privileged: true
    cap_add: 
      - ALL
    network_mode: host
    runtime: nvidia
```

- .devcontainer/Dockerfile:
```Dockerfile
ARG TAG
FROM ghcr.io/autowarefoundation/autoware-universe:${TAG}
ARG ROS_DISTRO

# apt-add-repository
RUN apt update && apt install -y software-properties-common

# kvaser interface deps
RUN apt-add-repository ppa:astuff/kvaser-linux && \
 apt update && apt install -y kvaser-canlib-dev

# extra rosdep
RUN apt update && apt install -y \
  ros-${ROS_DISTRO}-nmea-msgs \
  ros-${ROS_DISTRO}-can-msgs \
  ros-${ROS_DISTRO}-radar-msgs \
  ros-${ROS_DISTRO}-rmw-fastrtps-cpp \
  ros-${ROS_DISTRO}-rmw-fastrtps-dynamic-cpp \
  ros-${ROS_DISTRO}-rmw-fastrtps-shared-cpp \
  ros-${ROS_DISTRO}-nav2-msgs \
  ros-${ROS_DISTRO}-octomap-msgs \
  ros-${ROS_DISTRO}-nav2-costmap-2d \
  ros-${ROS_DISTRO}-mavros-msgs \
  ros-${ROS_DISTRO}-apex-test-tools \
  ros-${ROS_DISTRO}-ros2-socketcan \
  libtar-dev \
  libapr1-dev \
  libaprutil1-dev \
  bluez \
  librange-v3-dev

# devtools
RUN apt update && apt install -y \
  vim \
  gdb \
  gnome-terminal \
  dbus-x11 \
  ninja-build \
  ccache \
  less \
  ripgrep \
  fd-find

# create dev user with sudo rights
RUN adduser --disabled-password --gecos '' dev && \
  adduser dev sudo && \
  echo '%sudo ALL=(ALL) NOPASSWD:ALL' >> /etc/sudoers

# locales
RUN apt update && apt install -y locales locales-all
ENV LC_ALL en_US.UTF-8
ENV LANG en_US.UTF-8
ENV LANGUAGE en_US.UTF-8

# default entrypoint
USER dev
WORKDIR /home/dev/autoware

# no more "--Type <RET> for more" in GDB
RUN echo "set pagination off" > /home/dev/.gdbinit

# ROS2 config
RUN echo "source /opt/ros/$ROS_DISTRO/setup.bash" >> /home/dev/.bashrc && \
    echo "export RCUTILS_COLORIZED_OUTPUT=1" >> /home/dev/.bashrc && \
    echo "export RCUTILS_CONSOLE_OUTPUT_FORMAT=\"[{severity} {time}] [{name}]: {message} ({function_name}() at {file_name}:{line_number})\"" >> /home/dev/.bashrc && \
    echo "export ROS_LOCALHOST_ONLY=1" >> /home/dev/.bashrc

CMD ["/bin/bash"]
```

kasperornmeck, 2023-01-20:
@VRichardJP 
Thanks for your contribution!

At this stage we still need approval of Phase 1, once that's been achieved, we can move to Phase 2 where we'll define the dojo. The files you've shared are perfect for a discussion in Phase 2. Once we get there, I'll copy over your comment which we can use as the foundation for discussion.

oguzkaganozt, 2024-04-04:
Closing this issue as we have implemented this vs-code devcontainer with the latest containers

closed 2024-04-04 by oguzkaganozt