autowarefoundation/autoware_universe #2661: Vehicle stops in front of speed bump

lchojnack, 2023-01-16:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

During testing the current Autoware in AWSIM for ISUZU project, I've noticed that the vehicle stops in front of a speed bump, even when the speed bum is defined in lanelet2 as follows:
```
  <way id="4568">
    <nd ref="4564"/>
    <nd ref="4565"/>
    <nd ref="4566"/>
    <nd ref="4567"/>
    <tag k="type" v="speed_bump"/>
    <tag k="area" v="yes"/>
    <tag k="height" v="0.15"/>
  </way>
```
In the picture below, I've marked the situation when the vehicle stops in front of the speed bump. 
[image]

I've noticed that in autoware is a new feature that should detect the speed bump and the vehicle should slow down and pass the speed bump (https://github.com/autowarefoundation/autoware.universe/pull/647). I've tested the current main branch (2eafda6d4af8bd5c044d94cef68781fafe34968d)  with ` launch_speed_bump: true` param in [behavior_velocity_planner.param.yam](https://github.com/autowarefoundation/autoware_launch/blob/main/autoware_launch/config/planning/scenario_planning/lane_driving/behavior_planning/behavior_velocity_planner/behavior_velocity_planner.param.yaml) and the result is that `behavior_velocity_planner` node correctly detect the speed bump but the `obstacle_stop_planner` node treat the speed bump as obstacle and stop the vehicle as in the video below

https://user-images.githubusercontent.com/8416175/212693498-c85e854c-8065-4bdc-9af5-0c0e4df1084b.mp4

### Expected behavior

The vehicle shouldn't stop in front of the speed bump defined in the lanelet2

### Actual behavior

The vehicle stops in front of the speed bump defined in the lanelet2

### Steps to reproduce

1. Run AWSIM

piotr-zyskowski-rai, 2023-01-17:
@piotr-zyskowski-rai 
1. Share the speed bump model - is it the same as the real one?
2. Check the parameters in the forked repository: https://github.com/vautonomous/autoware

djargot, 2023-01-18:
@lchojnack @piotr-zyskowski-rai 
There are two types of speedbumps in GTU Campus environment. The approximate sizes are:
```
Big Bump:
    width: around 7 m
    height: around 0.1 - 0.15 m
    length: around 3.5 m

Small Bump:
    width: around 7 - 7.5 m
    height: around 0.1 - 0.2 m
    length: around 0.35 - 0.5 m
```
[image]
![Screenshot from 2023-01-18 11-01-33](https://user-images.githubusercontent.com/103112671/213144078-f3ae0cac-1975-4fe3-866d-7f274c8f48ec
[image]
.png)

xmfcx, 2023-01-24:
[image]

This is the scariest speed-bump we have with a near rectangular profile with 6-7cm height.

With the ground segmentation parameters we are using, these shouldn't be detected as obstacles.

mitsudome-r, 2023-01-31:
Robotec will modify the model to lower the speed and test on simulator again.

xmfcx, 2023-03-07:
This week the binary will be provided and it will be tested again.

WJaworskiRobotec, 2023-03-09:
This problem does not exist anymore. Ego properly interacts with the speed bump as shown on the attached video.

https://user-images.githubusercontent.com/85557834/224006562-d4672aa3-1f3a-41f6-988a-2056225467ed.mp4

closed 2023-03-14 by mitsudome-r