autowarefoundation/autoware_universe #2663: obstacle_avoidance_planner/mpt considers that ego vehicle is symmetrical around x axis

mehmetdogru, 2023-01-16:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

In mpt optimizer it is considered that vehicle is symmetrical around x axis. However if the vehicle is not symmetrical around x axis like isuzu bus where `right_overhang > left overhang` this assumption causes some incorrect calculations. Currently for the calculations `vehicle_width/2` is used:

```cpp
vehicle_width = wheel_tread + left_overhang + right_overhang
```

Instead `wheel_tread / 2.0 + left_overhang` and  `wheel_tread / 2.0 + rigth_overhang` should be used for the bound and circle calculations. For circles lateral offsets should be defined on y axis:

```cpp
lateral_offset =  ( right_overhang - left_overhang ) / 2.0
```

[image]

### Expected behavior

Consider the vehicle does not have to be symmetrical around x axis.

### Actual behavior

It is not considered that the vehicle does not have to be symmetrical around x axis.

### Steps to reproduce

1. Get isuzu [vehicle model](https://github.com/vautonomous/isuzu_vehicle_launch.git) & [sensor kit](https://github.com/vautonomous/isuzu_sensor_launch.git)
2. Launch planning_sim

takayuki5168, 2023-02-08:
@mehmetdogru Was this issue already solved?

beyzanurkaya, 2023-02-10:
@takayuki5168 It still some packages use `vehicle_width/2`. Not completely resolved.

beyzanurkaya, 2023-02-10:
I'll work this issue next week.

closed 2023-02-22 by beyzanurkaya