autowarefoundation/autoware_universe #2684: `trajectory_follower_node` smoke test failure

ambroise-arm, 2023-01-18:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Running `colcon test` on `trajectory_follower_node` can result in a failure depending on the way the package is built and on the machine it runs.

### Expected behavior

The test passes consistently.

### Actual behavior

The test can fail with different scenarios depending on the timing.

"could not create publisher" (same can happen with subscription depending on the timing):
```
2: [INFO] [controller_node_exe-1]: process started with pid [621346]
2: ok
2: 
2: ----------------------------------------------------------------------
2: Ran 1 test in 0.066s
2: 
2: OK
2: [INFO] [controller_node_exe-1]: sending signal 'SIGINT' to process[controller_node_exe-1]
2: [controller_node_exe-1] [INFO] [1674040064.730967355] [rclcpp]: signal_handler(signum=2)
2: [controller_node_exe-1] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
2: [controller_node_exe-1]   what():  could not create publisher: rcl node's context is invalid, at ./src/rcl/node.c:428
2: [ERROR] [controller_node_exe-1]: process has died [pid 621346, exit code -6, cmd '.../autoware.universe/install/trajectory_follower_node/lib/trajectory_follower_node/controller_node_exe --ros-args -r __ns:=/test --params-file .../autoware.universe/install/trajectory_follower_node/share/trajectory_follower_node/param/lateral_controller_defaults.param.yaml --params-file .../autoware.universe/install/trajectory_follower_node/share/trajectory_follower_node/param/longitudinal_controller_defaults.param.yaml --params-file .../autoware.universe/install/trajectory_follower_node/share/trajectory_follower_node/param/test_vehicle_info.param.yaml --params-file .../autoware.universe/install/trajectory_follower_node/share/trajectory_follower_node/param/test_nearest_search.param.yaml --params-file .../autoware.universe/install/trajectory_follower_node/share/trajectory_follower_node/param/trajectory_follower_node.param.yaml'].
2: test_exit_code (trajectory_follower_node.TestProcessOutput) ... FAIL
2: 
2: ======================================================================
2: FAIL: test_exit_code (trajectory_follower_node.TestProcessOutput)
2: ----------------------------------------------------------------------
2: Traceback (most recent call last):
2:   File ".../autoware.universe/install/autoware_testing/share/autoware_testing/autoware_testing/smoke_test.py", line 96, in test_exit_code
2:     launch_testing.asserts.assertExitCodes(proc_info)
2:   File "/opt/ros/humble/lib/python3.10/site-packages/launch_testing/asserts/assert_exit_codes.py", line 62, in assertExitCodes
2:     assert info.returncode in allowable_exit_codes, 'Proc {} exited with code {}'.format(
2: AssertionError: Proc controller_node_exe-1 exited with code -6
```

"escalating signal":
```
[INFO] [controller_node_exe-1]: process started with pid [21860]
ok

----------------------------------------------------------------------
Ran 1 test in 0.042s

OK
[INFO] [controller_node_exe-1]: sending signal 'SIGINT' to process[controller_node_exe-1]
[controller_node_exe-1] [INFO] [1673915693.854546118] [rclcpp]: signal_handler(signum=2)
[ERROR] [controller_node_exe-1]: process[controller_node_exe-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [controller_node_exe-1]: sending signal 'SIGTERM' to process[controller_node_exe-1]
[controller_node_exe-1] [INFO] [1673915698.860760916] [rclcpp]: signal_handler(signum=15)
[ERROR] [controller_node_exe-1]: process[controller_node_exe-1] failed to terminate '10.0' seconds after receiving 'SIGTERM', escalating to 'SIGKILL'
[INFO] [controller_node_exe-1]: sending signal 'SIGKILL' to process[controller_node_exe-1]
[ERROR] [controller_node_exe-1]: process has died [pid 21860, exit code -9, cmd '.../autoware.universe/install/trajectory_follower_node/lib/trajectory_follower_node/controller_node_exe --ros-args -r __ns:=/test --params-file .../autoware.universe/install/trajectory_follower_node/share/trajectory_follower_node/param/lateral_controller_defaults.param.yaml --params-file .../autoware.universe/install/trajectory_follower_node/share/trajectory_follower_node/param/longitudinal_controller_defaults.param.yaml --params-file .../autoware.universe/install/trajectory_follower_node/share/trajectory_follower_node/param/test_vehicle_info.param.yaml --params-file .../autoware.universe/install/trajectory_follower_node/share/trajectory_follower_node/param/test_nearest_search.param.yaml --params-file .../autoware.universe/install/trajectory_follower_node/share/trajectory_follower_node/param/trajectory_follower_node.param.yaml'].
test_exit_code (trajectory_follower_node.TestProcessOutput) ... FAIL

======================================================================
FAIL: test_exit_code (trajectory_follower_node.TestProcessOutput)
----------------------------------------------------------------------
Traceback (most recent call last):
  File ".../autoware.universe/install/autoware_testing/share/autoware_testing/autoware_testing/smoke_test.py", line 96, in test_exit_code
    launch_testing.asserts.assertExitCodes(proc_info)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch_testing/asserts/assert_exit_codes.py", line 62, in assertExitCodes
    assert info.returncode in allowable_exit_codes, 'Proc {} exited with code {}'.format(
AssertionError: Proc controller_node_exe-1 exited with code -9
```

Or it can also pass.

### Steps to reproduce

Build the package with `--cmake-args -DCMAKE_BUILD_TYPE=Release`, but not with the code coverage options (as is done in the CI).

### Possible causes

I think when the smoke test shuts down the node here: https://github.com/autowarefoundation/autoware.universe/blob/5f9d8478c97d8991cd5dd1ec85c9b5a50d582654/common/autoware_testing/autoware_testing/smoke_test.py#L89 if the node is in the process of creating a subscription or publisher, then either an RCLError exception can be thrown, or SIGINT can be escalated to SIGTERM and eventually SIGKILL, thus failing the test.

I didn't look into it, but I believe the code coverage options in the CI slow down the code execution enough that the node doesn't reach the point of creating pubs and subs when the shutdown is initiated, hence always passing.

I don't know why I only see the issue with `trajectory_follower_node` even though there are other packages that make use of the same smoke test.

### Additional context

Possible solution:
Adding a delay would give time to the node to finish creating subscriptions and publishers before shutting it down.
The problem is that there is no way of knowing what delay is correct.

takayuki5168, 2023-01-18:
In my local environment, this error sometimes occurred.
The error message is different from yours, but it seems the cause is similar.

This may be related. https://answers.ros.org/question/314295/ros2-crystal-rclcppexecutors-interrupt-guard-condition/
According to this, 
> This can happen if the executor is created after rclcpp::shutdown is called or SIGINT is received, either due to ctrl-c or something else signaling.

```
test 2
    Start 2: controller_node_exe_smoke_test

2: Test command: /usr/bin/python3.10 "-u" "/opt/ros/humble/share/ament_cmake_test/cmake/run_test.py" "/home/takayuki/pilot-auto/build/trajectory_follower_node/test_results/trajectory_follower_node/controller_node_exe_smoke_test.xunit.xml" "--package-name" "trajectory_follower_node" "--output-file" "/home/takayuki/pilot-auto/build/trajectory_follower_node/ros_test/controller_node_exe_smoke_test.txt" "--command" "ros2" "test" "/home/takayuki/pilot-auto/install/autoware_testing/share/autoware_testing/cmake/../autoware_testing/smoke_test.py" "arg_package:=trajectory_follower_node" "arg_package_exe:=controller_node_exe" "arg_param_filenames:=lateral_controller_defaults.param.yaml longitudinal_controller_defaults.param.yaml
test_vehicle_info.param.yaml test_nearest_search.param.yaml trajectory_follower_node.param.yaml" "--junit-xml=/home/takayuki/pilot-auto/build/trajectory_follower_node/test_results/trajectory_follower_node/controller_node_exe_smoke_test.xunit.xml" "--package-name=trajectory_follower_node"
2: Test timeout computed to be: 30
2: -- run_test.py: invoking following command in '/home/takayuki/pilot-auto/build/trajectory_follower_node':
2:  - ros2 test /home/takayuki/pilot-auto/install/autoware_testing/share/autoware_testing/cmake/../autoware_testing/smoke_test.py arg_package:=trajectory_follower_node arg_package_exe:=controller_node_exe arg_param_filenames:=lateral_controller_defaults.param.yaml longitudinal_controller_defaults.param.yaml
2: test_vehicle_info.param.yaml test_nearest_search.param.yaml trajectory_follower_node.param.yaml --junit-xml=/home/takayuki/pilot-auto/build/trajectory_follower_node/test_results/trajectory_follower_node/controller_node_exe_smoke_test.xunit.xml --package-name=trajectory_follower_node
2: Running with ROS_DOMAIN_ID 1
2: ROS_DOMAIN_ID 1
2: [INFO] [launch]: All log files can be found below /home/takayuki/.ros/log/2023-01-18-23-47-36-820574-npc2109028-humble-3459524
2: [INFO] [launch]: Default logging verbosity is set to INFO
2: test_wait_for_node_ready (trajectory_follower_node.DummyTest)
2: Waiting for the node is ready. ...
2: [WARN 1674053256.845195588] [rcl]: Found remap rule 'arg_package:=trajectory_follower_node'. This syntax is deprecated. Use '--ros-args --remap arg_package:=trajectory_follower_node' instead. (rcl_parse_arguments():576)
2: [WARN 1674053256.845203921] [rcl]: Found remap rule 'arg_package_exe:=controller_node_exe'. This syntax is deprecated. Use '--ros-args --remap arg_package_exe:=controller_node_exe' instead. (rcl_parse_arguments():576)
2: [INFO] [controller_node_exe-1]: process started with pid [3459526]
2: ok
2: 
2: ----------------------------------------------------------------------
2: Ran 1 test in 0.030s
2: 
2: OK
2: [INFO] [controller_node_exe-1]: sending signal 'SIGINT' to process[controller_node_exe-1]
2: [controller_node_exe-1] [INFO 1674053256.876278436] [rclcpp]: signal_handler(signum=2) (signal_handler():71)
2: [controller_node_exe-1] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
2: [controller_node_exe-1]   what():  failed to create guard condition: the given context is not valid, either rcl_init() was not called or rcl_shutdown() was called., at ./src/rcl/guard_condition.c:67
2: [ERROR] [controller_node_exe-1]: process has died [pid 3459526, exit code -6, cmd '/home/takayuki/pilot-auto/install/trajectory_follower_node/lib/trajectory_follower_node/controller_node_exe --ros-args -r __ns:=/test --params-file /home/takayuki/pilot-auto/install/trajectory_follower_node/share/trajectory_follower_node/param/lateral_controller_defaults.param.yaml --params-file /home/takayuki/pilot-auto/install/trajectory_follower_node/share/trajectory_follower_node/param/longitudinal_controller_defaults.param.yaml --params-file /home/takayuki/pilot-auto/install/trajectory_follower_node/share/trajectory_follower_node/param/test_vehicle_info.param.yaml --params-file /home/takayuki/pilot-auto/install/trajectory_follower_node/share/trajectory_follower_node/param/test_nearest_search.param.yaml --params-file /home/takayuki/pilot-auto/install/trajectory_follower_node/share/trajectory_follower_node/param/trajectory_follower_node.param.yaml'].
2: test_exit_code (trajectory_follower_node.TestProcessOutput) ...
2: FAIL
2: 
2: ======================================================================
2: FAIL: test_exit_code (trajectory_follower_node.TestProcessOutput)
2: ----------------------------------------------------------------------
2: Traceback (most recent call last):
2:   File "/home/takayuki/pilot-auto/install/autoware_testing/share/autoware_testing/autoware_testing/smoke_test.py", line 99, in test_exit_code
2:     launch_testing.asserts.assertExitCodes(proc_info)
2:   File "/opt/ros/humble/lib/python3.10/site-packages/launch_testing/asserts/assert_exit_codes.py", line 62, in assertExitCodes
2:     assert info.returncode in allowable_exit_codes, 'Proc {} exited with code {}'.format(
2: AssertionError: Proc controller_node_exe-1 exited with code -6
2: 
2: ----------------------------------------------------------------------
2: Ran 1 test in 0.000s
2: 
2: FAILED (failures=1)
2: -- run_test.py: return code 1
2: -- run_test.py: verify result file '/home/takayuki/pilot-auto/build/trajectory_follower_node/test_results/trajectory_follower_node/controller_node_exe_smoke_test.xunit.xml'
2/6 Test #2: controller_node_exe_smoke_test ...***Failed    0.65 sec
```

takayuki5168, 2023-01-18:
Our issue seems very similar to the following issue.
https://github.com/ros2/launch/issues/534

According to the issue, using `launch_pytest` is suggested rather than using `launch_testing` which will not be supported soon.
https://github.com/ros2/launch/pull/528

takayuki5168, 2023-01-18:
@adamdbrw I found you are the maintainer of the `autoware_testing` package.
Do you have any opinion on this issue.

TauTheLepton, 2023-04-04:
On my machine (commit `70cf872517cad0e92cdbd92510c8ac28333efeb6`) I was getting errors like

```log
2/6 Testing: controller_node_exe_smoke_test
2/6 Test: controller_node_exe_smoke_test
Command: "/usr/bin/python3.10" "-u" "/opt/ros/humble/share/ament_cmake_test/cmake/run_test.py" "/home/aorusadmin/autoware_issue2755/build/trajectory_follower_node/test_results/trajectory_follower_node/controller_node_exe_smoke_test.xunit.xml" "--package-name" "trajectory_follower_node" "--output-file" "/home/aorusadmin/autoware_issue2755/build/trajectory_follower_node/ros_test/controller_node_exe_smoke_test.txt" "--command" "ros2" "test" "/home/aorusadmin/autoware_issue2755/install/autoware_testing/share/autoware_testing/cmake/../autoware_testing/smoke_test.py" "arg_package:=trajectory_follower_node" "arg_package_exe:=controller_node_exe" "arg_param_filenames:=lateral_controller_defaults.param.yaml longitudinal_controller_defaults.param.yaml
test_vehicle_info.param.yaml test_nearest_search.param.yaml trajectory_follower_node.param.yaml
test.param.yaml" "--junit-xml=/home/aorusadmin/autoware_issue2755/build/trajectory_follower_node/test_results/trajectory_follower_node/controller_node_exe_smoke_test.xunit.xml" "--package-name=trajectory_follower_node"
Directory: /home/aorusadmin/autoware_issue2755/build/trajectory_follower_node
"controller_node_exe_smoke_test" start time: Apr 04 13:11 CEST
Output:
----------------------------------------------------------
-- run_test.py: invoking following command in '/home/aorusadmin/autoware_issue2755/build/trajectory_follower_node':
 - ros2 test /home/aorusadmin/autoware_issue2755/install/autoware_testing/share/autoware_testing/cmake/../autoware_testing/smoke_test.py arg_package:=trajectory_follower_node arg_package_exe:=controller_node_exe arg_param_filenames:=lateral_controller_defaults.param.yaml longitudinal_controller_defaults.param.yaml
test_vehicle_info.param.yaml test_nearest_search.param.yaml trajectory_follower_node.param.yaml
test.param.yaml --junit-xml=/home/aorusadmin/autoware_issue2755/build/trajectory_follower_node/test_results/trajectory_follower_node/controller_node_exe_smoke_test.xunit.xml --package-name=trajectory_follower_node
ROS_DOMAIN_ID 97
[INFO] [launch]: All log files can be found below /home/aorusadmin/.ros/log/2023-04-04-13-11-20-991559-Aorus15-530235
[INFO] [launch]: Default logging verbosity is set to INFO
test_wait_for_node_ready (trajectory_follower_node.DummyTest)
Waiting for the node is ready. ... [WARN] [1680606681.002752028] [rcl]: Found remap rule 'arg_package:=trajectory_follower_node'. This syntax is deprecated. Use '--ros-args --remap arg_package:=trajectory_follower_node' instead.
[WARN] [1680606681.002758684] [rcl]: Found remap rule 'arg_package_exe:=controller_node_exe'. This syntax is deprecated. Use '--ros-args --remap arg_package_exe:=controller_node_exe' instead.
[INFO] [controller_node_exe-1]: process started with pid [530247]
[controller_node_exe-1] terminate called after throwing an instance of 'rclcpp::exceptions::UninitializedStaticallyTypedParameterException'
[controller_node_exe-1]   what():  Statically typed parameter 'lateral_controller_mode' must be initialized.
[ERROR] [controller_node_exe-1]: process has died [pid 530247, exit code -6, cmd '/home/aorusadmin/autoware_issue2755/install/trajectory_follower_node/lib/trajectory_follower_node/controller_node_exe --ros-args -r __ns:=/test --params-file /home/aorusadmin/autoware_issue2755/install/trajectory_follower_node/share/trajectory_follower_node/param/lateral_controller_defaults.param.yaml --params-file /home/aorusadmin/autoware_issue2755/install/trajectory_follower_node/share/trajectory_follower_node/param/longitudinal_controller_defaults.param.yaml --params-file /home/aorusadmin/autoware_issue2755/install/trajectory_follower_node/share/trajectory_follower_node/param/test_vehicle_info.param.yaml --params-file /home/aorusadmin/autoware_issue2755/install/trajectory_follower_node/share/trajectory_follower_node/param/test_nearest_search.param.yaml --params-file /home/aorusadmin/autoware_issue2755/install/trajectory_follower_node/share/trajectory_follower_node/param/trajectory_follower_node.param.yaml --params-file /home/aorusadmin/autoware_issue2755/install/trajectory_follower_node/share/trajectory_follower_node/param/test.param.yaml'].
Processes under test stopped before tests completed
Process 'controller_node_exe-1' exited with -6
##### 'controller_node_exe-1' output #####
terminate called after throwing an instance of 'rclcpp::exceptions::UninitializedStaticallyTypedParameterException'
  what():  Statically typed parameter 'lateral_controller_mode' must be initialized.

##########################################
-- run_test.py: return code 1
-- run_test.py: verify result file '/home/aorusadmin/autoware_issue2755/build/trajectory_follower_node/test_results/trajectory_follower_node/controller_node_exe_smoke_test.xunit.xml'
<end of output>
Test time =   0.37 sec
----------------------------------------------------------
Test Failed.
"controller_node_exe_smoke_test" end time: Apr 04 13:11 CEST
"controller_node_exe_smoke_test" time elapsed: 00:00:00
----------------------------------------------------------
```

and I had to add these parameters to parameter files in `trajectory_follower_node/CMakeLists.txt`.

```yaml
/**:
  ros__parameters:
    lateral_controller_mode: "pure_pursuit" # "mpc"
    longitudinal_controller_mode: "pid"

    # steer offset
    steering_offset:
[... cut ...]
"controller_node_exe_smoke_test" end time: Apr 04 13:03 CEST
"controller_node_exe_smoke_test" time elapsed: 00:00:10
----------------------------------------------------------
```

This info log is printed every 5 seconds, so timing is not the issue.
For me it seems that the problem might be with `pure_pursuit_lateral_controller`, as `"mpc"` controller passes the test after adding necessary parameters.
I don't know what may be wrong with `pure_pursuit_lateral_controller` nor have any solutions, just wanted to share my findings.

TauTheLepton, 2023-04-05:
I checked and when launching autoware with `lateral_controller_mode` set to `pure_pursuit` it always dies at the end with exit code -6.

```log
[ERROR] [component_container-33]: process has died [pid 1673832, exit code -6, cmd '/opt/ros/humble/lib/rclcpp_components/component_container --ros-args -r __node:=control_container -r __ns:=/control -p use_sim_time:=False -p wheel_radius:=0.383 -p wheel_width:=0.235 -p wheel_base:=2.79 -p wheel_tread:=1.64 -p front_overhang:=1.0 -p rear_overhang:=1.1 -p left_overhang:=0.128 -p right_overhang:=0.128 -p vehicle_height:=2.5 -p max_steer_angle:=0.7'].
```

There should be opened a separate issue for this, as it is not an issue with smoke test, but `PurePursuitLateralController`.

TauTheLepton, 2023-09-04:
I am closing this issue as the smoke test fix PR has been merged and the `pure_pursuit` issue has been resolved and closed.

closed 2023-09-04 by TauTheLepton