autowarefoundation/autoware_universe #2695: Some launch files are missing parameters

VRichardJP, 2023-01-19:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Example:

Parameter `ndt_scan_matcher_param_path` is not defined for `pose_estimator.launch.xml`.

https://github.com/autowarefoundation/autoware.universe/blob/b38fa6a211614b74c5d70b149811fc372fa9bb7e/launch/tier4_localization_launch/launch/pose_estimator/pose_estimator.launch.xml#L1-L15

Included here:

https://github.com/autowarefoundation/autoware.universe/blob/b38fa6a211614b74c5d70b149811fc372fa9bb7e/launch/tier4_localization_launch/launch/localization.launch.xml#L27-L30

Trying to launch the node will result in an error:
```
$ ros2 launch tier4_localization_launch pose_estimator.launch.xml 
[INFO] [launch]: All log files can be found below /home/sig/.ros/log/2023-01-19-07-06-28-772358-vr-desktop-190338
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): launch configuration 'ndt_scan_matcher_param_path' does not exist
```

### Expected behavior

All params defined

### Actual behavior

undefined parameter

### Steps to reproduce

N/A

kminoda, 2023-01-19:
Thank you for the report! :+1: 

First of all, launch files in `tier4_xx_launch` is not expected to be launched independently, but to be launched via `autoware_launch`. So, that is a proper behavior.

If you want to launch `ndt_scan_matcher` independently, please use `ndt_scan_matcher.launch.xml`. If you want to launch localization module, please use `tier4_localization_component.launch.xml` in `autoware_launch`, or launch `autoware.launch.xml` with `perception:=false control:=false planning:=false ...` arguments.

VRichardJP, 2023-01-19:
Sorry, I don't understand why it would make a difference to use `autoware.launch.xml`.

On this line, the argument `ndt_scan_matcher_param_path` is used:
https://github.com/autowarefoundation/autoware.universe/blob/b38fa6a211614b74c5d70b149811fc372fa9bb7e/launch/tier4_localization_launch/launch/pose_estimator/pose_estimator.launch.xml#L12

Yet, it is not defined anywhere in the file (something like `<arg name="ndt_scan_matcher_param_path"/>`).

kminoda, 2023-01-19:
@VRichardJP It is defined here: https://github.com/autowarefoundation/autoware.universe/blob/87641bff81a70b5b0df587a25903d800cc146163/launch/tier4_localization_launch/launch/localization.launch.xml#L7

kminoda, 2023-01-19:
> Is this a new ROS2 feature? Is it documented anywhere?

Sorry, I don't know the specifics about this implementation. I also don't know whether it's a new feature from ROS 2 or from humble.

VRichardJP, 2023-01-20:
I see. I have looked at the ROS2 documentation, but I have not found any reference for such behavior.

Even though implicit parameter inheritance is a thing, may I suggest against using this kind of feature? I must not be the only one to think it makes launch files difficult to read.

For instance, `localization.launch.xml` currently looks like this:
```xml
 <arg name="ndt_scan_matcher_param_path"/>
 <group> 
   <push-ros-namespace namespace="pose_estimator"/> 
   <include file="$(find-pkg-share tier4_localization_launch)/launch/pose_estimator/pose_estimator.launch.xml"/> 
 </group> 
```

And `pose_estimator.launch.xml`:

```xml
<?xml version="1.0"?>
<launch>
  <group>
    <include file="$(find-pkg-share ndt_scan_matcher)/launch/ndt_scan_matcher.launch.xml">
      <!-- ... -->
      <arg name="param_file" value="$(var ndt_scan_matcher_param_path)"/>
    </include>
  </group>
</launch>
```

When I see `ndt_scan_matcher_param_path` in `localization.launch.xml`, I have no idea what this parameter is doing here. There are 5 other files included by the launch file. So I need to open each one by one to finally find it is being used by `pose_estimator.launch.xml`. Not only that, but considering the feature is not widely used in Autoware, it might be tempting to remove the `ndt_scan_matcher_param_path` as it may look unused...

I know ROS xml launch are quickly verbose, but at least this looks clear to me:
```xml
 <!-- localization.launch.xml -->
 <arg name="ndt_scan_matcher_param_path"/>
 <group> 
   <push-ros-namespace namespace="pose_estimator"/> 
   <include file="$(find-pkg-share tier4_localization_launch)/launch/pose_estimator/pose_estimator.launch.xml">
     <!-- explicit input -->
     <arg name="ndt_scan_matcher_param_path" value="$(var ndt_scan_matcher_param_path)"/>
   </include> 
 </group>
```

Similarly, I find `pose_estimator.launch.xml` clearer if required input arguments are explicited:
```xml
<?xml version="1.0"?>
<launch>
  <!-- explicit input -->
  <arg name="ndt_scan_matcher_param_path"/>
  <group>
    <include file="$(find-pkg-share ndt_scan_matcher)/launch/ndt_scan_matcher.launch.xml">
      <!-- ... -->
      <arg name="param_file" value="$(var ndt_scan_matcher_param_path)"/>
    </include>
  </group>
</launch>
```

VRichardJP, 2023-01-20:
Actually I found this: http://design.ros2.org/articles/roslaunch_xml.html

> `<arg>` Tag
The <arg> tag allows for launch file configuration via the command-line or when including it via an <include> tag. Arguments are launch configuration variables just like the ones defined by <let> tags. Arguments are limited to the scope of their definition and thus have to be explictly passed to included files if any.

This states arguments should be passed explicitly to included files...

kminoda, 2023-01-20:
Thank you for finding that. I also understand that providing the parameter path implicitly is a bit tricky, so I think it is worth considering adding the parameter argument for each file.

One obvious disadvantage is that it will increase the number of files we have to update when adding a new parameter file in `autoware_launch`.

(Note that the parameter configuration updates are in progress, and such improvement proposals are very welcome)

kenji-miyake, 2023-01-20:
@VRichardJP Thank you for asking it.
Please refer to this comment: https://github.com/ros2/launch/issues/620#issuecomment-1163534494

And here, `pose_estimator.launch.xml` is an internal launch.
I mean, TIER IV regard it a part of `localization.launch.xml`. Therefore, internal launch files don't have args.

Considering the maintenance costs, we probably won't add args to internal launch files of `tier4_*_launch` for a while.
However, since `tier4_*_launch` packages are just examples by TIER IV, you can propose other ones if you need.

isamu-takagi, 2023-01-20:
The arg tag in the include tag has no effect if arg tag in the child file has a default value. So in this case, It's easier to maintain if the include tag doesn't have arg tag because it don't need to copy the arg name. For readability, I think it's enough to check the arg tag in the child file.

```xml
<!-- parent.xml.launch -->
<arg name="foo" default="parent_foo"/>
<arg name="bar" default="parent_bar"/>
<include file="child.launch.xml">
  <arg name="foo" value="$(var foo)"/>
</include>

<!-- child.xml.launch -->
<arg name="foo" default="child_foo"/>
<arg name="bar" default="child_bar"/>
<log message="$(var foo)"/>  <!-- foo is parent_foo -->
<log message="$(var bar)"/>  <!-- bar is parent_bar -->
```

VRichardJP, 2023-01-20:
Sorry, I don't understand why using `default` is making anything different. Actually in your example, I must not be the only one to think that `$(var bar)` becoming `parent_bar` is rather unexpected behavior, although I understand why it happens.

I know it is painful to write 4 times the same parameter across 2 different files. I wish ROS2 xml files had a better syntax too. I understand the motivation behind your way of writing the launch files. As long as it is not abusive (like one parameter passed deep down many includes), and as long as the behavior is clear to everyone I am fine it.

isamu-takagi, 2023-01-20:
@VRichardJP If an arg tag in  include tag is specified, it may be misunderstood that the behavior will change if it is not specified. However, as I wrote earlier, it doesn't change even if it is not specified. My idea is to unify not writing arg tag in include tag to make the reader aware that arg is forwarded.

VRichardJP, 2023-01-20:
@isamu-takagi I totally agree. Inconsistency causes confusion.

To improve readability, what about just adding a few comments for the parameters that are implicitly forwarded? Even something as simple as:

```xml
  <!-- Parameter files -->
  <!-- Implicitly forwarded to children includes -->
  <arg name="crop_box_filter_measurement_range_param_path"/>
  <arg name="voxel_grid_downsample_filter_param_path"/>
  <arg name="random_downsample_filter_param_path"/>
  <arg name="ndt_scan_matcher_param_path"/>
  <arg name="localization_error_monitor_param_path"/>
```
or
```xml
  <!-- From parent launch context -->
  <arg name="param_file" value="$(var ndt_scan_matcher_param_path)"/> 
```

What do you think?

kasperornmeck, 2023-01-20:
@VRichardJP @kminoda @kenji-miyake @isamu-takagi 
I've read through the discussion in this issue and restructuring the ROS launch files is something which Open AD Kit WG will be addressing in _DevOps Dojo: ROS Launch Configuration_. I haven't setup the dojo yet. Would you like me to create the GitHub issue so we can start capturing how we want to reshape the launch files?

kasperornmeck, 2023-01-24:
@BonoloAWF 
Thanks for sharing the GitHub issue for _ROS Node Configuration,_ that dojo addresses the parameters which configure a ROS node. My understanding is that the discussion in this thread is focused on the launch files, which is of course related, but complex enough to have its own dojo.

BonoloAWF, 2023-01-24:
> @BonoloAWF Thanks for sharing the GitHub issue for _ROS Node Configuration,_ that dojo addresses the parameters which configure a ROS node. My understanding is that the discussion in this thread is focused on the launch files, which is of course related, but complex enough to have its own dojo.

Thanks - given the broader scope, I've removed it from Open AD Kit Project (it's still marked as related).