autowarefoundation/autoware_universe #2703: Change vitrual traffic signal interface (V2X/V2I)

isamu-takagi, 2023-01-20:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Currently, communication with infrastructure such as shutter is handled by `InfrastructureCommand.msg` and `VirtualTrafficLightState.msg`, but sync issues and requests for cancellation support have been reported. So I propose a new interface to resolve them.

### Purpose

Resolve sync issues and support cancellation.

### Possible approaches

Think of a controlled area by a infrastructure as a shared resource. The vehicle acquires and releases locks of that resources. Also, when communicating directly with the infrastructure without using the Internet, it is assumed that the communication area is limited, so lock acquisition and release must be completed within this area.

[image]

### Definition of done

- [ ] Add new interface.
- [ ] Add new element of lanelet2 map
- [ ] Add planning module for new interface
- [ ] Replace planning module
- [ ] Remove old interface
- [ ] Remove old element of lanelet2 map

isamu-takagi, 2023-11-15:
The following ideas were commented on in the API WG.
https://github.com/orgs/autowarefoundation/discussions/3994

 - command: information about time
 - status: recommended alternative route

jerry73204, 2023-11-17:
It looks like the areas are fully exclusive. Does it handle the case of ramps where two car flows merge?

isamu-takagi, 2023-11-17:
> It looks like the areas are fully exclusive. Does it handle the case of ramps where two car flows merge?

I think it is applicable. It can be considered an intersection with two entrances and one exit.