autowarefoundation/autoware_universe #2755: Issue with building ROS2 python packages (related to setuptools)

lchojnack, 2023-01-26:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

As mentioned in https://github.com/orgs/autowarefoundation/discussions/3215, when changing the version of `setuptools` to 66 or above, then the error `pkg_resources.extern.packaging.version.InvalidVersion: Invalid version: '0.23ubuntu1'` occurs and the build fails. 

As written in [this github issue](https://github.com/pypa/setuptools/issues/3772) **the change in setuptools is intended (related to [PEP 440 – Version Identification and Dependency Specification](https://peps.python.org/pep-0440/) ) and it is not a bug.** 

From my research, the changes in `setuptools` not only influence Autoware, but particularly building python ROS2 packages ([Developing a ROS 2 package — ROS 2 Documentation: Humble  documentation](https://docs.ros.org/en/humble/How-To-Guides/Developing-a-ROS-2-Package.html#python-packages) ).  As I understand, the issue occurs when building a python ROS2 package on Debian/Ubuntu system because the version is e.g. `0.23ubuntu1` but should be e.g. `0.23+ubuntu1` (according to PEP 440).

### Expected behavior

ROS2 python packages are building without `pkg_resources.extern.packaging.version.InvalidVersion: Invalid version: '0.23ubuntu1'`  error

### Actual behavior

Building ROS2 python packages fail

### Steps to reproduce

1. Change the `systemtools` version to 66 or above e.g. `pip install -U setuptools==66.0.0 `
2. Build the `simulator_compatibility_test package`  `colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release --packages-up-to simulator_compatibility_test` → the `pkg_resources.extern.packaging.version.InvalidVersion: Invalid version: '1.1build1'` appears 
3. Change the `systemtools` version to 65.7.0 (last working) e.g. `pip install -U setuptools==65.7.0`
4. Build the simulator_compatibility_test package → the package builds without issue

### Versions

Ubuntu 22.04, ROS2 Humble, Python 3.10.6

### Additional context

https://github.com/pypa/setuptools/issues/3772
https://github.com/orgs/autowarefoundation/discussions/3215

mitsudome-r, 2023-01-31:
This issue was discussed in ASWG. Here are some notes:
- Are there any packages in Autoware that require version 66?
  - Not sure
- Are there any discussion on-going in ROS 2 community?
  - Haven’t found one yet

Possible approach:
- For now, we can keep using setuptool 65. 
- First we should find someone to find out what exact change is needed to migrate to version 66.

piotr-zyskowski-rai, 2023-01-31:
@mitsudome-r 
I did some experiments on my computer.

I had setuptools version 59 before the upgrade, and Autoware was built correctly
```
$ pip show setuptools
Name: setuptools
Version: 59.6.0
```

After calling `pip install setuptools --upgrade`, the version became 67
```
$ pip show setuptools
Name: setuptools
Version: 67.0.0
```
which broke my Autoware build. It would need to be checked which setuptools are installed on a fresh machine. 

If it installs version 67 by default, it might be rather essential to fix this issue or at least notify users in the documentation that it might break the build and that the solution is to downgrade the setuptools.

xmfcx, 2023-02-07:
Right now the workaround works (by using the older version). We'll keep it open for a proper fix to come.

xmfcx, 2023-04-17:
I used the latest version but I get build errors for many message packages like:
```
autoware_adapi_v1_msgs autoware_adapi_version_msgs autoware_auto_control_msgs autoware_auto_geometry_msgs autoware_auto_mapping_msgs autoware_auto_system_msgs autoware_common_msgs autoware_control_msgs autoware_localization_msgs autoware_map_msgs autoware_perception_msgs autoware_sensing_msgs morai_msgs tier4_debug_msgs tier4_external_api_msgs tier4_localization_msgs tier4_planning_msgs tier4_rtc_msgs tier4_simulation_msgs tier4_v2x_msgs velodyne_msgs
```

To reproduce it, I have done `pip install --upgrade setuptools` or `pip install -U setuptools==67.6.1` (same thing for now)

```
mfc@mfc-SG13:~/projects/autoware$ pip show setuptools
Name: setuptools
Version: 67.6.1
```

And build `autoware_common_msgs` to see.

<details>
  <summary>Click to expand</summary>

```
mfc@mfc-SG13:~/projects/autoware$ colcon build --symlink-install  --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_EXPORT_COMPILE_COMMANDS=1  --packages-select autoware_common_msgs
Starting >>> autoware_common_msgs
--- stderr: autoware_common_msgs                             
Traceback (most recent call last):
  File "/home/mfc/projects/autoware/build/autoware_common_msgs/ament_cmake_python/autoware_common_msgs/setup.py", line 4, in <module>
    setup(
  File "/home/mfc/.local/lib/python3.10/site-packages/setuptools/__init__.py", line 107, in setup
    _install_setup_requires(attrs)
  File "/home/mfc/.local/lib/python3.10/site-packages/setuptools/__init__.py", line 75, in _install_setup_requires
    dist = MinimalDistribution(attrs)
  File "/home/mfc/.local/lib/python3.10/site-packages/setuptools/__init__.py", line 57, in __init__
    super().__init__(filtered)
  File "/home/mfc/.local/lib/python3.10/site-packages/setuptools/dist.py", line 486, in __init__
    for ep in metadata.entry_points(group='distutils.setup_keywords'):
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 1009, in entry_points
    return SelectableGroups.load(eps).select(**params)
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 459, in load
    ordered = sorted(eps, key=by_group)
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 1006, in <genexpr>
    eps = itertools.chain.from_iterable(
  File "/usr/lib/python3.10/importlib/metadata/_itertools.py", line 16, in unique_everseen
    k = key(element)
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 941, in _normalized_name
    return self._name_from_stem(stem) or super()._normalized_name
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 622, in _normalized_name
    return Prepared.normalize(self.name)
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 871, in normalize
    return re.sub(r"[-_.]+", "-", name).lower().replace('-', '_')
  File "/usr/lib/python3.10/re.py", line 209, in sub
    return _compile(pattern, flags).sub(repl, string, count)
TypeError: expected string or bytes-like object
gmake[2]: *** [CMakeFiles/ament_cmake_python_build_autoware_common_msgs_egg.dir/build.make:70: CMakeFiles/ament_cmake_python_build_autoware_common_msgs_egg] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:447: CMakeFiles/ament_cmake_python_build_autoware_common_msgs_egg.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< autoware_common_msgs [0.33s, exited with code 2]

Summary: 0 packages finished [0.78s]
  1 package failed: autoware_common_msgs
  1 package had stderr output: autoware_common_msgs
```
</details>

Then I tried with previoudly working version `65.7.0` with:
`pip install -U setuptools==65.7.0`

<details>
  <summary>Click to expand</summary>

```
[... 32 lines cut ...]
gmake[1]: *** [CMakeFiles/Makefile2:447: CMakeFiles/ament_cmake_python_build_autoware_common_msgs_egg.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< autoware_common_msgs [0.30s, exited with code 2]
                                
Summary: 0 packages finished [0.75s]
  1 package failed: autoware_common_msgs
  1 package had stderr output: autoware_common_msgs
```

</details>

Which gave the same error.

Then I tried with an older version `59.6.0` and it worked: `pip install -U setuptools==59.6.0`

```
mfc@mfc-SG13:~/projects/autoware$ colcon build --symlink-install  --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_EXPORT_COMPILE_COMMANDS=1  --packages-select autoware_common_msgs
Starting >>> autoware_common_msgs
Finished <<< autoware_common_msgs [0.56s]                     

Summary: 1 package finished [0.98s]
```

I don't know why this happened for me but wanted to share.

TauTheLepton, 2023-04-18:
@xmfcx On both my existing and fresh autoware workspaces it all works

```log
$ colcon build --symlink-install  --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_EXPORT_COMPILE_COMMANDS=1  --packages-select autoware_common_msgs
Starting >>> autoware_common_msgs
Finished <<< autoware_common_msgs [0.45s]                     

Summary: 1 package finished [0.87s]
```

I have the newest `setuptools` version possible inbstalled with `pip install --upgrade setuptools`

```log
$ pip show setuptools
Name: setuptools
Version: 67.6.1
```

Does your error occur on fresh workspace as well?

xmfcx, 2023-04-26:
@TauTheLepton maybe something was wrong with my OS setup.

I've reinstalled Ubuntu 22.04 from scratch and everything works fine.

On my previous installation I tried attempted many things but didn't work. I will try to use more docker-based development process from now.

Since https://github.com/autowarefoundation/autoware-documentation/pull/346 is merged, I think we can close this issue.

closed 2023-04-26 by xmfcx