autowarefoundation/autoware_universe #2841: fix: typo and expired link

taikitanaka3, 2023-02-08:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

There are many expired links and typos for example
https://github.com/autowarefoundation/autoware.universe/pull/2216

### Purpose

To eliminate typos and expired link to pass CI test

### Possible approaches

Fix readme and codes.

### Definition of done

pre-commit optional and spell check
https://github.com/autowarefoundation/autoware.universe/actions/workflows/pre-commit-optional.yaml
https://github.com/autowarefoundation/autoware.universe/actions/workflows/spell-check-differential.yaml

HansRobo, 2023-02-08:
I started to fix links in #2842!

HansRobo, 2023-02-08:
I found that `pre-commit-optional` is failed when no link errors are occured.
https://github.com/autowarefoundation/autoware.universe/pull/2842#issuecomment-1422356593

So, we need to debug and fix in another PR.

mitsudome-r, 2023-03-14:
@HansRobo @taikitanaka3 Do you still have other action items for this issue?

taikitanaka3, 2023-03-14:
@HansRobo 
I think this is done. how about you.

closed 2023-03-26 by taikitanaka3