autowarefoundation/autoware_universe #2860: feat Improve visualization of the perception module

lexavtanke, 2023-02-10:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Create an additional output from the autoware perception so the detected objects will look like following:
- Pedestrians:
  - Yellow point cloud and elliptic ground marker
- Cars:
  - Blue point cloud and rectangular ground marker
- Cyclists:
  - Green point cloud and rectangular ground marker
 
Example from Waymo interface: [image] 

### Purpose

Main goal in this task is to achieve better performance and elegance for the visualization of the perception stack.

### Possible approaches

Create node which will subscribe to topic /perception/obstacle_segmentation/pointcloud and  publish colored pointcloud according to topic  /perception/object_recognition/objects.

For the polygons colors and forms can be changed in autoware_auto_perception_plugin as it was done in the pr https://github.com/autowarefoundation/autoware.universe/pull/2780

### Definition of done

User is able to visualize colored pointcloud and polygons according to recognized classes.

yukkysaito, 2023-02-17:
I think it is better to implement as rviz plugin than to create a new node. Adding a node as a visualization node would complicate the node diagram and launch.

lexavtanke, 2023-02-23:
I have some thoughts about realization of the rviz plugin for detected (and etc) objects pointclouds visualization. 

It can be done by modification existed autoware_auto_perception_rviz_plugin and there are two ways:

- by adding ability to publish pointclouds to autoware plugin and vizualizing them by standard pointcloud rviz plugin
- by adding ability to visialize pointclouds to autoware plugin 

Or it can be done by creating additional plugin which will visualize objects pointclouds.

So there is difference in configuration experience if some one need to disable or enable pointclouds and type of polygon visualization he will need to navigate to different tabs in separate plugins or just click to check box in one plugin.

I hope to get some feedback from authors of the original autoware_auto_perception_rviz_plugin from Tier IV and Apex.AI.

Here is picture how it will look like from prototype node.
[image]

lexavtanke, 2023-03-09:
Here are some preliminary results.  
I made two version because it was more easy to start with separate node for this and then integrate it in to rviz_plugin. 

[This video ](https://user-images.githubusercontent.com/37497658/223960369-f4817087-7c91-4e8c-8cc2-7dc3421b21bb.webm) show comparison between debug output of detected_object_validator (purple) and my node for publishing detected objects pointclouds (blue).
[repo with node](https://github.com/lexavtanke/autoware.universe/tree/feat/add_perception_objects_pointcloud_publisher) 

[This video](https://user-images.githubusercontent.com/37497658/223957995-6b07debe-4df8-48e2-9707-bbf8ffd04597.webm) show comparison between debug output of detected_object_validator (purple) and autoware_auto_perception_rviz_plugin with with integrated feature (blue).
[autoware fork](https://github.com/lexavtanke/autoware.universe/tree/feat/add_perception_objects_pointcloud_publisher)

It seem to me that they are works kinda similar to each other and both of them are slower than debug output. I think it may be because validator use sync between topics and my code don't. I didn't find the way to make it in rviz_plugin. 

I would really appreciate any clue or advice how to speed up my code to delay between pointclouds and markers.