autowarefoundation/autoware_universe #2865: "lidar_centerpoint" gives poor pedestrian results after model change

kaancolak, 2023-02-10:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

 After re-train of "lidar_centerpoint" model, it gives many false negatives for pedestrians and mixing vehicle-pedestrian labels. 

is it possible some retraining parameters were missed while adding the model?

Revert to the previously trained version of the model and it looks better. 

Re-trained current model : 

[image]

Previously model:

[image]

 Probably relevant PR: https://github.com/autowarefoundation/autoware.universe/pull/2553/files

 A bag file will be provided.

### Expected behavior

 Correctly detect road objects.

### Actual behavior

 Gives false negatives for pedestrians, and labels mix each other. 

### Steps to reproduce

1. Concat enate point cloud topics in the given bag file
2. Run the lidar_centerpoint model 

### Possible causes

 - Missing configuration or training issues

knzo25, 2023-02-13:
Hi, this is due to a change in the database used for training (which had to be done for licensing reasons).

kaancolak, 2023-02-13:
I see, thank you for the explanation.

> Hi, this is due to a change in the database used for training (which had to be done for licensing reasons).

I see, thank you for the explanation. I am closing the issue.

closed 2023-02-13 by kaancolak