autowarefoundation/autoware_universe #2867: Visualize 3D objects and road information onto the 2D image

kaancolak, 2023-02-10:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Visualize useful autoware topics onto the 2D image.

Example topics:

- 3D Bounding boxes of objects 
- Trajectory of the vehicle
- Road boundaries 

### Purpose

Enhance debug options and increase the visualization capacity of autoware. 

### Possible approaches

Create module will subscribe to relevant topics and image/camera_info messages. Transform relevant information to the image plane and visualize on in the 2D image. 

### Definition of done

-  [ ] Module implemented 
-  [ ] Documentation prepared

miursh, 2023-05-09:
Is [rviz camera](https://wiki.ros.org/rviz/DisplayTypes/Camera) tool not enough for your usecase?
It shows like the pic below
[image]

xmfcx, 2023-06-14:
@miursh does it not lag behind? Can you also share a video recording of perception boxes projected onto images?

palas21, 2024-07-10:
I’ve implemented a solution. You can see it in [this video](https://youtu.be/-74QKgD7FDc).

palas21, 2024-07-16:
[Here](https://youtu.be/DjkyyZBKmeM?si=cldPjPjlNEuR8e7A) you can see the result of the same bag with the rviz camera plugin.