autowarefoundation/autoware_universe #2906: Update occupancy grid map frame to gain longer range visibility in the intersection

YoshiRi, 2023-02-17:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

# Problem statement

Currently, we are using `base_link` frame to generate occupancy grid map.
The base_link is typically set at the center of the rear wheel, behind the vehicle rather than the LiDAR or driver. (See following Figure)

Therefore, the field of view of the occupancy grid map generated in the `base_link` is narrower than that of the driver or the sensor's viewpoint, for example, when turning right at an intersection.

[image]

### Purpose

We need to change grid map estimation frame to achieve

- **wider range visibility**
- principled correctness

### Possible approaches

I think there are two possible approaches:

| name | figure | note |
|--|--|--|
|current|[image]|current setting|
|plan A|[image]|generate occupancy grid map in other frame|
|plan B|[image]|More faithful method to sensor visibility|

planA should be easier to implement.   
planB will accurately represent the visible range of the sensor.

Since we often use only top LiDAR to sense further objects, I think using planA and set its frame to top lidar sensor will be enough.

### Definition of done

[TBD]

Should be confirmed in scenarios involving right turns at intersections.

soblin, 2023-02-17:
In planning modules, the main customers for occupancy grid map are:
- occlusion_spot
- intersection
- pullover/pullout
modules.

This proposal maybe especially useful for intersection module.

VRichardJP, 2023-02-20:
Although it might be more tedious to implement, plan B looks like the only valid option to me: if a vehicle has multiple sensors at several places, it is because visibility is not the same everywhere. For small vehicles such as cars, the top lidar is indeed the one with the widest/farthest FOV, but when it comes to larger vehicles (minibus, shuttle, bus, truck, etc), no sensor can clearly see everything around.

miursh, 2023-02-21:
I suppose plan B is more suitable for expressing sensor FoV. 
Although, it would be quite complicated to implement and I believe there are several design points.
e.g. is isn't it difficult to determine whether out of sensor FoV or free space by only using limited FoV sensor, which both appears no return points.

taikitanaka3, 2023-02-21:
@soblin 
I think generate grid map only at driver frame is also ok. I don't know module which requires grid map at baselink frame.

YoshiRi, 2023-02-23:
Now I agree with VRichardJP and miurush that the plan B should be the final solution.
But you know, we need a lot of changes and there are also concern about computational load.

So I just implemented the plan A for the instant solution. 
@soblin could you check if this PR improves our planning scenarios?

idorobotics, 2023-10-05:
@YoshiRi what is the current status for this issue?

YoshiRi, 2023-10-05:
@idorobotics 
Sorry, I am Pending this matter now due to other prioritized tasks. 
The latest status is in [DevelopmentAboutOccupancyGridMapFusion.pdf](https://github.com/autowarefoundation/autoware.universe/files/12821145/DevelopmentAboutOccupancyGridMapFusion.pdf)

## remaining tasks
OGM fusion will be available after merging the following two PRs.

- [OGM fusion node PR](https://github.com/autowarefoundation/autoware.universe/pull/3107) 
- [launcher PR](https://github.com/autowarefoundation/autoware.universe/issues/3045)

YoshiRi, 2024-05-31:
All features are merged and successfully tested.
Also see related discussion: https://github.com/orgs/autowarefoundation/discussions/4158#discussioncomment-8664198.

closed 2024-05-31 by YoshiRi