autowarefoundation/autoware_universe #2915: Remove duplicate codes for conversion from lat/lon to xyz

mitsudome-r, 2023-02-20:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

In Autoware, there are some packages that does the conversion from lat/lon to xyz.
I believe they should be using the same backend library to avoid conversion errors between different nodes.

### Purpose

Reduce duplicate codes in different packages.

### Possible approaches

List all packages that does lat/lon -> xyz conversion and refactor as a single library.

Some packages that I am aware of:
- https://github.com/MapIV/llh_converter/tree/main/include/llh_converter (https://github.com/autowarefoundation/autoware/pull/3261)
- https://github.com/autowarefoundation/autoware.universe/tree/main/sensing/gnss_poser/include/gnss_poser
- https://github.com/autowarefoundation/autoware.universe/tree/main/sensing/geo_pos_conv
- https://github.com/autowarefoundation/autoware_common/blob/main/tmp/lanelet2_extension/lib/mgrs_projector.cpp

### Definition of done

TBD

mitsudome-r, 2023-02-20:
From this [discussion](https://github.com/autowarefoundation/autoware/pull/3261#discussion_r1111573054) we have decided to wait for this task until we have stable release of Autoware so that we can do test of the packages with stable code.

kminoda, 2023-11-08:
In Autoware it has been unified into https://github.com/autowarefoundation/autoware.universe/tree/main/common/geography_utils