autowarefoundation/autoware_universe #2922: Motion planning generates zigzagging path in planning simulator

VRichardJP, 2023-02-21:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I am testing Autoware planning simulator with a bus (from https://github.com/autowarefoundation/itri_bus_sensor_kit and https://github.com/autowarefoundation/cbus_vehicle_launch). I have created a simple lanelet map with large lanes (5.0m) and simple curves.

When I run the simulator with the bus, the vehicle sometimes brakes strongly as the planner generates zigzagging path right in front of the vehicle. 

For example at 15 second in this video:

[output.webm](https://user-images.githubusercontent.com/18645627/220256493-a6234fb1-8fd8-40d4-a5fb-a1d1f9f156b7.webm)

The problem does not occur with the sample vehicle, so it might be related to the size of the vehicle. In my example, the lane is twice as large as the bus, so I think there should be enough room for Autoware to moneuver the vehicle without such kind of zigzag.

I had a similar problem a long time ago (on a real car this time): https://github.com/autowarefoundation/autoware.universe/issues/1958

Any idea?

### Expected behavior

No zigzag

### Actual behavior

zigzagging path + strong brake

### Steps to reproduce

Run planning simulator with a big vehicle?

takayuki5168, 2023-02-21:
@VRichardJP Could you try `fitting_uniform_circle` for `vehicle_circles.method` in obstacle_avoidance_planner?
https://github.com/autowarefoundation/autoware.universe/pull/2646

VRichardJP, 2023-02-21:
@takayuki5168 thank you for the suggestion.

I still get path zigzags:

[image]

takayuki5168, 2023-02-27:
@VRichardJP Could you wait for a week from now?

I will merge this PR (some updates for obstacle_avoidance_planner) tomorrow and write a doc with another PR.
https://github.com/autowarefoundation/autoware.universe/pull/2796

VRichardJP, 2023-02-27:
No problem!

closed 2023-03-09 by VRichardJP