autowarefoundation/autoware_universe #2925: `ObstacleAvoidancePlanner` sometimes crash in `SplineInterpolation::getSplineInterpolatedDiffValues` because of NaN filled trajectories

VRichardJP, 2023-02-21:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

While playing with the planning simulator, I caught this crash in GDB:

```
Thread 12 "component_conta" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffe57fa640 (LWP 452830)]
__pthread_kill_implementation (no_tid=0, signo=6, threadid=140737043736128) at ./nptl/pthread_kill.c:44
44	./nptl/pthread_kill.c: No such file or directory.
(gdb) bt
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=140737043736128) at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=140737043736128) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=140737043736128, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007ffff7925476 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007ffff790b7f3 in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007ffff7bcdbbe in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff7bd924c in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff7bd92b7 in std::terminate() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#8  0x00007ffff7bd9518 in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6
#9  0x00007fffe47980fe in interpolation_utils::validateKeys (base_keys=..., query_keys=...) at /home/sig/autoware/install/interpolation/include/interpolation/interpolation_utils.hpp:79
#10 0x00007ffff402e1e5 in SplineInterpolation::getSplineInterpolatedDiffValues (this=this@entry=0x7fffe57f7340, query_keys=std::vector of length 0, capacity 0) at /home/sig/autoware/src/universe/autoware.universe/common/interpolation/src/spline_interpolation.cpp:248
#11 0x00007ffff40312ef in interpolation::slerp2dFromXY (base_keys=std::vector of length 99, capacity 128 = {...}, base_x_values=std::vector of length 99, capacity 99 = {...}, base_y_values=std::vector of length 99, capacity 99 = {...}, query_keys=std::vector of length 0, capacity 0) at /home/sig/autoware/src/universe/autoware.universe/common/interpolation/src/spline_interpolation_points_2d.cpp:79
#12 0x00007fffe43ae36e in interpolation_utils::interpolate2DTrajectoryPoints (base_x=std::vector of length 99, capacity 99 = {...}, base_y=std::vector of length 99, capacity 99 = {...}, resolution=0.10000000000000001) at /home/sig/autoware/src/universe/autoware.universe/planning/obstacle_avoidance_planner/src/utils/utils.cpp:393
#13 0x00007fffe43af034 in interpolation_utils::getInterpolatedTrajectoryPoints (points=std::vector of length 99, capacity 99 = {...}, delta_arc_length=0.10000000000000001) at /usr/include/c++/11/array:56
#14 0x00007fffe43d608f in ObstacleAvoidancePlanner::generateFineTrajectoryPoints (this=0x7fffbc010fc0, path_points=std::vector of length 352, capacity 352 = {...}, traj_points=std::vector of length 99, capacity 99 = {...}) at /home/sig/autoware/src/universe/autoware.universe/planning/obstacle_avoidance_planner/src/node.cpp:1539
#15 0x00007fffe43dfa75 in ObstacleAvoidancePlanner::generatePostProcessedTrajectory (this=0x7fffbc010fc0, path_points=std::vector of length 352, capacity 352 = {...}, optimized_traj_points=..., planner_data=...) at /home/sig/autoware/src/universe/autoware.universe/planning/obstacle_avoidance_planner/src/node.cpp:1437
#16 0x00007fffe43e0c27 in ObstacleAvoidancePlanner::generateTrajectory (this=0x7fffbc010fc0, planner_data=...) at /home/sig/autoware/src/universe/autoware.universe/planning/obstacle_avoidance_planner/src/node.cpp:972
#17 0x00007fffe43e25fc in ObstacleAvoidancePlanner::onPath (this=0x7fffbc010fc0, path_ptr=std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> >> (use count 4, weak count 0) = {...}) at /home/sig/autoware/src/universe/autoware.universe/planning/obstacle_avoidance_planner/src/node.cpp:931
#18 0x00007fffe44000f7 in std::__invoke_impl<void, void (ObstacleAvoidancePlanner::*&)(std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > >), ObstacleAvoidancePlanner*&, std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > > > (__f=<optimized out>, __t=<optimized out>, __f=<optimized out>, __t=<optimized out>) at /usr/include/c++/11/bits/invoke.h:74
#19 std::__invoke<void (ObstacleAvoidancePlanner::*&)(std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > >), ObstacleAvoidancePlanner*&, std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > > > (__fn=<optimized out>) at /usr/include/c++/11/bits/invoke.h:96
#20 std::_Bind<void (ObstacleAvoidancePlanner::*(ObstacleAvoidancePlanner*, std::_Placeholder<1>))(std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > >)>::__call<void, std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > >&&, 0ul, 1ul>(std::tuple<std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > >&&>&&, std::_Index_tuple<0ul, 1ul>) (__args=..., this=<optimized out>) at /usr/include/c++/11/functional:420
#21 std::_Bind<void (ObstacleAvoidancePlanner::*(ObstacleAvoidancePlanner*, std::_Placeholder<1>))(std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > >)>::operator()<std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > >, void>(std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > >&&) (this=<optimized out>) at /usr/include/c++/11/functional:503
#22 std::__invoke_impl<void, std::_Bind<void (ObstacleAvoidancePlanner::*(ObstacleAvoidancePlanner*, std::_Placeholder<1>))(std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > >)>&, std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > > >(std::__invoke_other, std::_Bind<void (ObstacleAvoidancePlanner::*(ObstacleAvoidancePlanner*, std::_Placeholder<1>))(std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > >)>&, std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > >&&) (__f=...) at /usr/include/c++/11/bits/invoke.h:61
#23 std::__invoke_r<void, std::_Bind<void (ObstacleAvoidancePlanner::*(ObstacleAvoidancePlanner*, std::_Placeholder<1>))(std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > >)>&, std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > > >(std::_Bind<void (ObstacleAvoidancePlanner::*(ObstacleAvoidancePlanner*, std::_Placeholder<1>))(std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > >)>&, std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > >&&) (__fn=...) at /usr/include/c++/11/bits/invoke.h:111
#24 std::_Function_handler<void (std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > >), std::_Bind<void (ObstacleAvoidancePlanner::*(ObstacleAvoidancePlanner*, std::_Placeholder<1>))(std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > >)> >::_M_invoke(std::_Any_data const&, std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > >&&) (__functor=..., __args#0=...) at /usr/include/c++/11/bits/std_function.h:290
#25 0x00007fffe43fe9a2 in std::function<void (std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > >)>::operator()(std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > >) const (__args#0=std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> >> (empty) = {...}, this=<optimized out>) at /usr/include/c++/11/bits/std_function.h:590
#26 rclcpp::AnySubscriptionCallback<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> >, std::allocator<void> >::dispatch(std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > >, rclcpp::MessageInfo const&)::{lambda(auto:1&&)#1}::operator()<std::function<void (std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > >)>&>(std::function<void (std::shared_ptr<autoware_auto_planning_msgs::msg::Path_<std::allocator<void> > >)>&) const (__closure=<optimized out>, __closure=<optimized out>, callback=...) at /opt/ros/humble/include/rclcpp/rclcpp/any_subscription_callback.hpp:556
[... cut ...] 8000000000000), y = -nan(0x8000000000000), z = 0}, orientation = {x = -nan(0x8000000000000), y = -nan(0x8000000000000),
```

So `generateOptimizedTrajectory(planner_data)` must be where the NaNs come from. But I cannot find the cause.

It seems to be the exact same issue I had long time ago: https://github.com/autowarefoundation/autoware.universe/issues/1956

### Expected behavior

No crash

### Actual behavior

Crash

### Steps to reproduce

Run a simulation for quite a long time (>10 minutes?)

VRichardJP, 2023-02-24:
Managed reproduce and track further.

- in `ObstacleAvoidancePlanner::generateOptimizedTrajectory`: `auto optimal_trajs = optimizeTrajectory(planner_data)` is filled with NaNs
- in `ObstacleAvoidancePlanner::optimizeTrajectory`: `eb_traj` data looks fine, but `mpt_trajs` is plagued with NaNs past the few first points. `debug_data_` also contains quite a lot of NaNs.
[mpt_trajs.txt](https://github.com/autowarefoundation/autoware.universe/files/10822271/mpt_trajs.txt)
[debug_data.txt](https://github.com/autowarefoundation/autoware.universe/files/10822274/debug_data.txt)
- in `MPTOptimizer::getModelPredictiveTrajectory`: `full_ref_points` has NaNs already from index 10:
[full_ref_points.txt](https://github.com/autowarefoundation/autoware.universe/files/10822269/full_ref_points.txt)
```
(gdb) p full_ref_points[10]
$62 = {p = {x = 68340.349355984203, y = 87556.120593821382, z = 0}, k = -0.013618963751321597, v = 0, yaw = 0.13447614778344213, s = 103.00034775595475, alpha = -0.018055671208563676, bounds = {lower_bound = -1.334256644760081, upper_bound = 1.2359930104880295, lower_collision_type = CollisionType::NO_COLLISION, upper_collision_type = CollisionType::NO_COLLISION}, near_objects = false, fix_kinematic_state = {<boost::optional_detail::optional_base<Eigen::Matrix<double, 2, 1, 0, 2, 1> >> = {<boost::optional_detail::optional_tag> = {<No data fields>}, m_initialized = true, m_storage = {dummy_ = {data = "`\367\377\377\377\377\377\377`\367\377\377\377\377\377\377", aligner_ = {<No data fields>}}}}, <No data fields>}, plan_from_ego = true, optimized_kinematic_state = {<Eigen::PlainObjectBase<Eigen::Matrix<double, 2, 1, 0, 2, 1> >> = {<Eigen::MatrixBase<Eigen::Matrix<double, 2, 1, 0, 2, 1> >> = {<Eigen::DenseBase<Eigen::Matrix<double, 2, 1, 0, 2, 1> >> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 2, 1, 0, 2, 1>, 3>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 2, 1, 0, 2, 1>, 1>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 2, 1, 0, 2, 1>, 0>> = {<Eigen::EigenBase<Eigen::Matrix<double, 2, 1, 0, 2, 1> >> = {<No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, m_storage = {m_data = {array = {-nan(0xffffffffff760), -nan(0xffffffffff760)}}}}, <No data fields>}, optimized_input = -nan(0xfffffffffffff), beta = std::vector of length 2, capacity 2 = {{<boost::optional_detail::tc_optional_base<double>> = {<boost::optional_detail::optional_tag> = {<No data fields>}, m_initialized = true, m_storage = -0.033652800855534099}, <No data fields>}, {<boost::optional_detail::tc_optional_base<double>> = {<boost::optional_detail::optional_tag> = {<No data fields>}, m_initialized = true, m_storage = 0.027222244326961437}, <No data fields>}}, vehicle_bounds = std::vector of length 2, capacity 2 = {{lower_bound = -1.2823146600490181, upper_bound = 1.2887599624182762, lower_collision_type = CollisionType::NO_COLLISION, upper_collision_type = CollisionType::NO_COLLISION}, {lower_bound = -1.286394608775006, upper_bound = 1.2832801305609018, lower_collision_type = CollisionType::NO_COLLISION, upper_collision_type = CollisionType::NO_COLLISION}}, vehicle_bounds_poses = std::vector of length 6, capacity 6 = {{position = {x = 68338.377956195429, y = 87555.785982500194, z = 0}, orientation = {x = 0, y = 0, z = 0.083966274399585844, w = 0.99646859697807511}}, {position = {x = 68344.988822598709, y = 87556.621077250617, z = 0}, orientation = {x = 0, y = 0, z = 0.053703979785262194, w = 0.9985569000088198}}, {position = {x = 68338.37795561926, y = 87555.785985564318, z = 0}, orientation = {x = 0, y = 0, z = 0.083965497661904434, w = 0.99646866242867305}}, {position = {x = 68344.988903609308, y = 87556.620268475846, z = 0}, orientation = {x = 0, y = 0, z = 0.053617014340098697, w = 0.99856157335101459}}, {position = {x = 68338.37795561926, y = 87555.785985564333, z = 0}, orientation = {x = 0, y = 0, z = 0.083965497658517296, w = 0.99646866242895848}}, {position = {x = 68344.988918233212, y = 87556.620121877102, z = 0}, orientation = {x = 0, y = 0, z = 0.053601251579584688, w = 0.9985624195958418}}}}
```

At this moment I realized `prev_optimal_trajs_ptr_` in `ObstacleAvoidancePlanner::optimizeTrajectory` has NaNs too. But I don't know if its the cause or a the consequence. Indeed, by the time the program crashes and is taken over by gdb, the `prev_optimal_trajs_ptr_` has been updated already. So as I explore the program state, it may not be exactly the same as how things were before the crash. 

So maybe it's wrong, but I tried reset the `prev_optimal_trajs_ptr_` and tried to call `ObstacleAvoidancePlanner::optimizeTrajectory` again
```
(gdb) call prev_optimal_trajs_ptr_->model_predictive_trajectory.clear()
(gdb) call prev_optimal_trajs_ptr_->mpt_ref_points.clear()
```
But then, back in `MPTOptimizer::getModelPredictiveTrajectory` does not produce any more NaNs. Anyway, at this point I totally lost the state of the program. I guess I have no choice but to add some random assert in the code so I can catch the NaNs as early as possible. That's for a next time.

In the meantime, if anyone is versed in the `ObstacleAvoidancePlanner` and dare to look at the log data, it would be much appreciated!

VRichardJP, 2023-02-25:
Down the rabbit hole again. This time I have added a few asserts before `prev_optimal_trajs_ptr_` is updated so I can keep the program state unaltered when it crashes.

This time, `full_ref_points` is totally fine. All data looks fine down to here:

https://github.com/autowarefoundation/autoware.universe/blob/bd0058142ee9266a4c31b73b24172a7097fbc58a/planning/obstacle_avoidance_planner/src/mpt_optimizer.cpp#L296-L297

Where I suddenly get NaNs only:
```
(gdb) p optimized_control_variables.get().size()
$59 = 90
(gdb) p *optimized_control_variables.get().data()@90
$60 = {-nan(0xfffffffffffff) <repeats 90 times>}
```

I also get this error message when the function `executeOptimization` is called:
```
ERROR in osqp_update_lower_bound: upper bound must be greater than or equal to lower bound
```
However I get this one fairly often so it might not be directly related.

Although it is difficult to check everything, the input of the function seemed ok to me (at least no NaN). A few examples:
```
(gdb) p mpt_matrix.Bex.size()
$50 = 16020
(gdb) p *mpt_matrix.Bex.data()@100
$51 = {1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0}
(gdb) p *mpt_matrix.Bex.data()@1000
$52 = {1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 0, 1, 0.50000257173695672, 1, 1.0000034642413596, 1, 1.5000054081307184, 1, 2.0000139428925507, 1, 2.5000152804727662, 1, 3.000021309151208, 1, 3.5000259654839567, 1, 4.0000315638297081, 1, 4.500037724050685, 1, 5.0000446928801665, 1...}
(gdb) p *mpt_matrix.Bex.data()@16020
value requires 128160 bytes, which is more than max-value-size
(gdb) p mpt_matrix.Wex.size()
$53 = 178
(gdb) p *mpt_matrix.Wex.data()@178
$54 = {0, 0, 0, 0.017548233583924384, 0.0087741324538379278, 0.035903063067845728, 0.026725733779343036, 0.055067608044679181, 0.054260007790601972, 0.075040988772028819, 0.091780602549958318, 0.09589394460780834, 0.13972815296761903, 0.11765775156279717, 0.19855757660265938, 0.1404204166941391, 0.26876857107177216, 0.16424070835378637, 0.3508899370077222, 0.18920289957917824, 0.44549270532005592, 0.21538972215560287, 0.55318915218737752, 0.24291996108287903, 0.67465108557201403, 0.27190726913607727, 0.81060706878537769, 0.30248662901855194, 0.96185324737589217, 0.33479013784391437, 1.1292518170858838, 0.36894692815579155, 1.3137297497878737, 0.40509928695989733, 1.5162837929255615, 0.44333602692113677, 1.7379593755244682, 0.4835547148715918, 1.9797411354941548, 0.52546844937934822, 2.2424844164912274, 0.56861228217958348, 2.5267977738685468, 0.61229045931620818, 2.832950336936408, 0.65570661007791509, 3.1608113398848343, 0.69825540524751817, 3.5099465272802162, 0.7394892807304897, 3.8796988629297782, 0.77912851210627332, 4.2692707210784304, 0.81708769152567973, 4.677822014137897, 0.85341319617485789, 5.1045357776541511, 0.88814734227874215, 5.5486163246597267, 0.92134609056122829, 6.0092962938121861, 0.9530662129502725, 6.4858358254733606, 0.98336317382274374, 6.9775239715607213, 1.0122903871298308, 7.4836751692104198, 1.0398938924863852, 8.0036277029285099, 1.0662209094789727, 8.5367433683804954, 1.0913290207769299, 9.0824128175062704, 1.1152822938663745, 9.6400586429758874, 1.1381500161447098, 10.209138092764125, 1.1600033000185128, 10.789143821623517, 1.1809067188044726, 11.379600899674656, 1.20091030116916, 11.980059468917315, 1.2200480935989122, 12.590086684050082, 1.238336540168915, 13.209257924112977, 1.2557808800082013, 13.837151103248322, 1.272379647703032, 14.473343387839726, 1.2881285449416702, 15.117409801600719, 1.3030243811340845, 15.768923884916632, 1.3170678912062559, 16.427459537629545, 1.3302624978163577, 17.09259232660084, 1.3426137760640984, 17.763900574848495, 1.3541289932545102, 18.440966220964121, 1.3648165792880123, 19.123375438357407, 1.3746857033306092, 19.810719056608026, 1.3837458798886353, 20.502592645172847, 1.39200666069882, 21.198596518208618, 1.3994773205680355, 21.898335609782649, 1.4061666411658453, 22.601419240042823, 1.412082730645791, 23.307460805605231, 1.41723284433078, 24.016077361848154, 1.4216233488462382, 24.726889126839065, 1.4252595639050101, 25.439518961858088, 1.4281455684737199, 26.153591767738487, 1.4302858183761729, 26.868734676206092, 1.4316781317527152, 27.584573729445655, 1.4323367318909694, 28.300742192065478, 1.4323138825330586, 29.016899065712028, 1.431693381951247, 29.732745875588929, 1.4305892768948338, 30.448040532774677, 1.4291600460088645, 31.162620595789136, 1.4275625325240342, 31.876401897472192, 1.4259093274243579, 32.589356588591926, 1.4242825564481849, 33.301497894602491, 1.4227319896460016, 34.012863913246356, 1.4212683433782012, 34.723498106324925, 1.4198865061633856, 35.433441377466032, 1.4185867156396756, 36.142734750533542, 1.4173676357660683, 36.851418581692009, 1.4162281040265308, 37.559532645 [... cut ...] .4151669269470841, 38.267116118956245, 1.4141828313666043, 38.974207543090408, 1.4132745481089375, 39.680844823835287, 1.412440800629297, 40.387065229352793, 1.4116803313260449, 41.092905399221223, 1.4109919152988799, 41.798401360316177, 1.4103743744531358, 42.503588550233665, 1.4098265893409514, 43.208501846844236, 1.4093475089949405, 43.91317560256558, 1.4089361584865954, 44.617643683032391, 1.4085916441728985}

```

Unfortunately, I broke the gdb stack after that. So that's for a next time...

VRichardJP, 2023-02-26:
The first NaNs appear here:

https://github.com/autowarefoundation/autoware.universe/blob/bd0058142ee9266a4c31b73b24172a7097fbc58a/planning/obstacle_avoidance_planner/src/mpt_optimizer.cpp#L842

```
(gdb) p result
$18 = std::tuple containing = {[1] = std::vector of length 179, capacity 179 = {-nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff)}, [2] = std::vector of length 628, capacity 628 = {-nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan(0xfffffffffffff), -nan [... cut ...]  1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30, 1e+30...}
```

I could not get further down in the osqp code yet.

satoshi-ota, 2023-02-26:
@VRichardJP Thanks for your report, but I'm not well versed in obstacle avoidance module :pray: 
So, @takayuki5168 could you review this issue ?

VRichardJP, 2023-02-26:
Unfortunately, I cannot get deeper than:

https://github.com/autowarefoundation/autoware.universe/blob/bd0058142ee9266a4c31b73b24172a7097fbc58a/common/osqp_interface/src/osqp_interface.cpp#L367

Before the data does not seem to contain NaNs. After it does. What happens in between is a black box.
[Others people have had issues with NaNs in OSQP](https://github.com/osqp/osqp/issues?q=nan) but I can't see any that fit the current use case. 

The OSQP solver maths is beyond my understanding, maybe the input data is absolutely wrong but I can't see.

Anyway, I have wandered around a bit with GDB. Trying to call `osqp_solver_ptr_->optimize()` manually and see what is going on in OSQP `m_work` input/output. If any one is interested: [gdb.txt](https://github.com/autowarefoundation/autoware.universe/files/10833670/gdb.txt)

Regarding the original crash itself, I guess it should be safe to assume osqp solver may return NaNs, and always check the solution does not contain any. I don't know if `status_polish == -1` is a sign there is a problem, or if we should just simply check the output values.

VRichardJP, 2023-02-27:
If anyone is interested into reproducing the bug. Here is my current setup:

- add exception when a NaN is found in `OSQPInterface::solve`, and start obstacle avoidance planner in gdb:
```diff
diff --git a/common/osqp_interface/src/osqp_interface.cpp b/common/osqp_interface/src/osqp_interface.cpp
index 972a586b16..9b08de25e9 100644
--- a/common/osqp_interface/src/osqp_interface.cpp
+++ b/common/osqp_interface/src/osqp_interface.cpp
@@ -24,6 +24,7 @@
 #include <string>
 #include <tuple>
 #include <vector>
+#include <stdexcept>
 
 namespace autoware
 {
@@ -385,6 +386,12 @@ OSQPInterface::solve()
 
   m_latest_work_info = *(m_work->info);
 
+  if ((!sol_primal.empty() && isnan(sol_primal.front())) 
+    || (!sol_lagrange_multiplier.empty() && isnan(sol_lagrange_multiplier.front())))
+  {
+    throw std::runtime_error("NaN!");
+  }
+
   return result;
 }
 
diff --git a/launch/tier4_planning_launch/launch/scenario_planning/lane_driving/motion_planning/motion_planning.launch.py b/launch/tier4_planning_launch/launch/scenario_planning/lane_driving/motion_planning/motion_planning.launch.py
index de8104af44..f847d92d1f 100644
--- a/launch/tier4_planning_launch/launch/scenario_planning/lane_driving/motion_planning/motion_planning.launch.py
+++ b/launch/tier4_planning_launch/launch/scenario_planning/lane_driving/motion_planning/motion_planning.launch.py
@@ -212,6 +212,7 @@ def launch_setup(context, *args, **kwargs):
             obstacle_avoidance_planner_component,
             obstacle_velocity_limiter_component,
         ],
+        prefix="gnome-terminal -- gdb -ex=r --args"
     )
 
     obstacle_stop_planner_loader = LoadComposableNodes(
```

- compile packages you want to investigate with `-DCMAKE_BUILD_TYPE=Debug`. For example:
```bash
$ colcon build --symlink-install --cmake-args ' -DCMAKE_BUILD_TYPE=Debug' --packages-select obstacle_avoidance_planner osqp_interface
```

I am using a bus-like vehicle, based on [this one](https://github.com/autowarefoundation/cbus_vehicle_launch). I am not sure whether the exact dimensions matter. I have used these parameters:
- `sample_bus_description/config/vehicle_info.param.yaml`
```yaml
/**:
  ros__parameters:
    # approx. bus size
    wheel_radius: 0.487 # The radius of the wheel, primarily used for dead reckoning.
    wheel_width: 0.226 #0.235 # The lateral width of a wheel tire, primarily used for dead reckoning.
    wheel_base: 4.0 # between front wheel center and rear wheel center
    wheel_tread: 2.10 #1.64 # between left wheel center and right wheel center
    front_overhang: 2.0 #1.0 # between front wheel center and vehicle front
    rear_overhang: 2.0 #1.1 # between rear wheel center and vehicle rear
    left_overhang: 0.150 #0.128 # between left wheel center and vehicle left
    right_overhang: 0.150 #0.128 # between right wheel center and vehicle right
    vehicle_height: 3.2
    # 0.5 = 30 degrees
    max_steer_angle: 0.50 #0.70 # [rad]
```
- `sample_bus_description/urdf/vehicle.xacro` (so that the 3D model in Rviz matches the vehicle info)
```xml
<?xml version="1.0"?>
<robot xmlns:xacro="http://ros.org/wiki/xacro">
[... 61 lines cut ...]
- run the planning simulator:
```bash
$ source install/setup.bash
$ ros2 launch autoware_launch planning_simulator.launch.xml map_path:=<MAP_PATH> vehicle_model:=sample_bus sensor_model:=sample_bus_sensor_kit rviz:=false
```
- run rviz2:
```
$ source install/setup.bash
$ ros2 run rviz2 rviz2 -d src/launcher/autoware_launch/autoware_launch/rviz/autoware.rviz 
```

- set initial/goal position there:
[image]

- set velocity limit to 10km/h. I am not sure about why, but the bug seems to occur more often when speed limit is set low.

- In a last terminal, run the pose reset script:
```bash
$ source install/setup.bash
$ ./initialpose.py
```

- Finally, set operation mode to AUTO to start the simulation

Normally, the vehicle is not fast enough to reach the goal pose within the 45 seconds, so the simulation will keep running endlessly until something crashes. If nothing has crashed after ~20 minutes, I usually change the goal position a bit, as it seems to have an influence.

VRichardJP, 2023-03-09:
Similarly to https://github.com/autowarefoundation/autoware.universe/issues/2922, I cannot reproduce the crash anymore. 
I don't know if it has been fixed or if it is just related to the zigzag issue. Anyway, I think my comment https://github.com/autowarefoundation/autoware.universe/issues/2925#issuecomment-1445346844 still holds: it would be safer to assume osqp may generate NaNs, and ignore the generated trajectory if any NaN is detected.

mitsudome-r, 2023-03-14:
I will close this issue since further discussion should be handled in https://github.com/autowarefoundation/autoware.universe/issues/3040

closed 2023-03-14 by mitsudome-r