autowarefoundation/autoware_universe #2961: Rviz sometimes hangs for ever on the splash screen

VRichardJP, 2023-02-28:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Sometimes Rviz just get stuck on the splash screen for ever. When this happens, the spash screen image also looks buggy:
[image]

```bash
$ ros2 run rviz2 rviz2 -d src/launcher/autoware_launch/autoware_launch/rviz/autoware.rviz 
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-sig'
[INFO 1677562384.079134045] [rviz2]: Stereo is NOT SUPPORTED (operator()() at ./src/main.cpp:62)
[INFO 1677562384.079196102] [rviz2]: OpenGl version: 4.5 (GLSL 4.5) (operator()() at ./src/main.cpp:62)
```

If I don't add `-d src/launcher/autoware_launch/autoware_launch/rviz/autoware.rviz ` to load Autoware's configuration, Rviz never has any problem.

### Expected behavior

Rviz always work

### Actual behavior

Sometimes never show up

### Steps to reproduce

```bash
$ ros2 run rviz2 rviz2 -d src/launcher/autoware_launch/autoware_launch/rviz/autoware.rviz
```

VRichardJP, 2023-03-10:
I have not faced the issue for a few days.
Could be related to a recent update, or rebuilding autoware, or rebuilding the docker image, or it's just my bug triggering curse is gone.

closed 2023-03-10 by VRichardJP

YuqiHuai, 2024-06-11:
I was able to launch RVIZ a few hours ago but now it keeps crashing.

```
ros2 run rviz2 rviz2 -d ~/autoware/src/launcher/autoware_launch/autoware_launch/rviz/autoware.rviz
[INFO] [1718088670.431022203] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1718088670.431107114] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [1718088670.474140852] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1718088670.813666349] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1718088670.816537480] [rviz2]: Stereo is NOT SUPPORTED
[ros2run]: Segmentation fault
```