autowarefoundation/autoware_universe #2966: dead link in autoware_auto_geometry README

satoshi-ota, 2023-03-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Fow now, pre-commit optional outputs dead link error.

[image]

### Expected behavior

pre-commit optional doesn't output dead link error in `autoware_auto_geometry`.

### Actual behavior

pre-commit optional outputs dead link error in `autoware_auto_geometry`.

### Steps to reproduce

1. run pre-commit optional

NOTE: This error seems to be happend in github CI/CD only. (This error doesn't be reproduce in local environment.)

satoshi-ota, 2023-03-01:
Hi @esteve :hand: 
Thanks for your fix in https://github.com/autowarefoundation/autoware.universe/pull/743, but unfortunately, pre-commit optional outputs dead link error, so could you fix it again?

esteve, 2023-03-14:
@satoshi-ota can you point me to a GitHub CI run where this shows? Thanks for the screenshot, but unfortunately it does not have enough context and I'd prefer a link to a CI job. I checked the link locally and it resolves fine, I don't know what's wrong here.

esteve, 2023-03-14:
@satoshi-ota I don't know much about that paper, I only updated the link in #743 The official link to that paper is the one we already have in Autoware.universe, so I don't know of any other source, if you find another link that works, please update #3077 

I suspect that the link blocks access from GitHub, @kenji-miyake is there a way to avoid a link from being checked in the pre-commit script? Thanks.

satoshi-ota, 2023-03-15:
Hi @esteve Thanks for your work.

> can you point me to a GitHub CI run where this shows?

Sorry for being unkind :pray: The CI error can be seen in any opened PRs. 
For example, your can see the error in https://github.com/autowarefoundation/autoware.universe/pull/3079. Please click the `Details` button, and the error is going to be shown.

[image]

satoshi-ota, 2023-03-15:
> I don't know much about that paper, I only updated the link in https://github.com/autowarefoundation/autoware.universe/pull/743 The official link to that paper is the one we already have in [...]

I got it. Thanks, too.
I'll try to look for any other links, but I'm going to remove this link at the worst case.

esteve, 2023-05-15:
@satoshi-ota did you update/remove the link to the paper? Can this PR be closed? Thanks.

satoshi-ota, 2023-05-15:
Hi @esteve 
Sorry, I fogot to remove the link. I remove it today in your PR https://github.com/autowarefoundation/autoware.universe/pull/3077 :pray: 
And, I'll fix the other dead links.

PR #3077 (2023-05-19): docs(autoware_auto_geometry): fix link to paper
## Description

Fixes #2966 

## Pre-review checklist for the PR author

The PR author **must** check the checkboxes below when creating the PR.

- [x] I've confirmed the [contribution guidelines].
- [x] The PR follows the [pull request guidelines].

## In-review checklist for the PR reviewers

The PR reviewers **must** check the checkboxes below before approval.

- [x] The PR follows the [pull request guidelines].

## Post-review checklist for the PR author

The PR author **must** check the checkboxes below before merging.

- [ ] There are no op [... cut]
closed 2023-05-19 by satoshi-ota