autowarefoundation/autoware_universe #2970: `ring_outlier_filter` output is sometimes empty

VRichardJP, 2023-03-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Following the [sample sensor kit](https://github.com/autowarefoundation/sample_sensor_kit_launch/blob/06330e6b07449a3609ced295c0a9ce6a1805ef99/common_sensor_launch/launch/velodyne_node_container.launch.py#L154) I have enabled the ring outlier filter on our car.

After that the `ndt_scan_matcher` started to crash on startup on receiving empty input cloud:
[image]

I found `ring_outlier_filter` is the node responsible for this. On the picture below, I printed on the left the size of `ring_outlier_filter` output cloud (`$ ros2 topic echo /sensing/lidar/top/outlier_filtered/pointcloud --field width`) and on the right the size of its input cloud (`$ ros2 topic echo /sensing/lidar/top/rectified/pointcloud_ex --field width`):
[image]

As you can see, the output is sometimes empty. 

I could also confirm with Rviz the rectified cloud looks totally normal while the the outlier cloud sometimes "blinks".

Besides the output being empty, I think it is also annoying `ndt_scan_matcher` crashes because of that.  A small guard could be added to `NDTScanMatcher::callback_sensor_points`.

### Expected behavior

No empty output

### Actual behavior

Sometimes empty output

### Steps to reproduce

use the outlier filter?

kminoda, 2023-03-01:
I also observed the similar issue (where the number of points becomes zero occasionally) in the latest awf/autoware with sample rosbag provided in the Autoware tutorial. 

@sykwer Do you think the above phenomena is related to this PR? https://github.com/autowarefoundation/autoware.universe/pull/2908/
(Anyway, we need to have some sort of error-catch algorithm in ndt_scan_matcher to accept 0 point inputs)

YamatoAndo, 2023-03-01:
Thank you for your report!

I also reproduced the problem of the number of point clouds in `/sensing/lidar/top/outlier_filtered/pointcloud` becomes zero.

But `ndt_scan_matcher` doesn't crash in my environment (ROS2 galactic).
According to my sources, it seems that the zero division happens in the PCL installed with ROS2 humble.
It seems to be fixed in the latest PCL. https://github.com/PointCloudLibrary/pcl/pull/5113

VRichardJP, 2023-03-01:
@YamatoAndo I see. I am not sure how I installed PCL on the car PC, but I think I used autoware setup script. So I guess the PCL people normally have installed on there machine will have this bug, no?

kminoda, 2023-03-02:
Reverting the above PR https://github.com/autowarefoundation/autoware.universe/pull/2908 solved the problem. Currently @sykwer is trying to investigate and fix the issue for us.

sykwer, 2023-03-03:
Fixed https://github.com/autowarefoundation/autoware.universe/pull/2992

closed 2023-03-03 by sykwer