autowarefoundation/autoware_universe #2991: Autoware parks like an asshole on parking lot

VRichardJP, 2023-03-03:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

In this example, I have set the goal point perpendicular to the initial position (in red), so a maneuver is required. Autoware starts its maneuver but kind of stops halfway, as it considers the goal has been reached as soon as the XY matches:
[image]

I guess this is because the goal heading error tolerance is too high. For lane driving, it is not a problem as we can't maneuver, but when the vehicle is on a parking lot, Autoware should park precisely.

I guess this should be fixed for cargo ODD.

### Expected behavior

Autoware parks precisely on parking lot

### Actual behavior

heading error is too big on goal.

### Steps to reproduce

Run simulator with a parking lot and ask autoware to park somewhere.

yukkysaito, 2023-03-04:
Thanks for the report.
The parking planner has not been changed for some time since it was created as a prototype.
The free space planner doesn't generate stable trajectory, so when it stops at a certain place, we create the lanelet.

It may be better to review the design from the beginning if necessary this time with cargo ODD or bus ODD.

NorahXiong, 2023-03-16:
@VRichardJP Which planning algorithm did you choose? A-star or RRT?

VRichardJP, 2023-03-16:
@NorahXiong A* in this example

NorahXiong, 2023-03-16:
I did a similar test with the default settings and I got a pretty good result. 
@VRichardJP Have you repeated the test? Did it stop halfway everytime? 

[image]

VRichardJP, 2023-03-16:
In my case, I added a few obstacles to make the parking maneuver less trivial. You can see the 2 buses left and right of the parking spot in my screenshot. I only tested a couple times with this setup.

VRichardJP, 2023-03-22:
I am using the same map you are showing.

Starting position:
[image]
After struggling with the maneuver, going back and forth many times just a few meters, autoware drove very quickly close to the target goal and got stuck:
[image]

Another again. Start position:
[image]
After goal is reached:
[image]

As @yukkysaito said, it seems that Autoware considered the goal has been reached if it is stopped around the goal, even if the maneuver is not finished. This occurs typically when Autoware moves back and forth around the goal to adjust its orientation

NorahXiong, 2023-04-14:
Hi, @VRichardJP 
I updated my environment and reproduced your issue. I reduced the "vehicle_shape_margin_m" value to 0.2(m) and the parking procedure succeeded.
I checked the "vehicle_shape_margin_m" parameter changelog and I found no one modifying the default value recently. The problem may come from the perception module. Maybe algorithms and logic changed in the perception module and it results in unstable/incorrect distance measurement.
Anyway, hope my information could help you!

closed 2023-04-20 by VRichardJP