autowarefoundation/autoware_universe #3034: Address an issue that pcl::transformPointCloud dies when receives pointcloud with zero points

kminoda, 2023-03-09:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

pcl::transformPointCloud, which is used in several packages in Autoware, dies when receiving pointcloud with zero points. This causes a error during runtime.

### Purpose

To avoid Autoware failing when zero length pointcloud is published

### Possible approaches

1. Implement alternative function to pcl::transformPointCloud (e.g. in tier4_autoware_utils)
2. Install the latest PCL (as a matter of fact, the bug has been addressed here: https://github.com/PointCloudLibrary/pcl/pull/5113)
3. Get rid of PCL

I think 1 is the best approach as a temporary solution.

### Definition of done

Autoware does not die because of pcl::transformPointCloud even when the pointcloud with zero width is published.

mitsudome-r, 2023-03-14:
@kminoda Are you planning to work on this task? or do you help from AWF members?

kminoda, 2023-03-14:
@mitsudome-r No plan for now. Would appreciate if we can get a help from AWF members!

kaancolak, 2023-03-14:
@mitsudome-r @kminoda,  I would like to work on this issue, assigned myself.

h-ohta, 2023-05-19:
Now, some packages in autoware use `pcl_ros::transformPointCloud`.
e.g. https://github.com/autowarefoundation/autoware.universe/blob/faf4bd0286971e7ceaf52f1ff1ff8df3954348e4/planning/obstacle_velocity_limiter/src/pointcloud_utils.cpp#L47.
https://github.com/autowarefoundation/autoware.universe/blob/faf4bd0286971e7ceaf52f1ff1ff8df3954348e4/perception/ground_segmentation/src/ransac_ground_filter_nodelet.cpp#L177
Do we have to fix them too?

kminoda, 2023-05-19:
@h-ohta Thank you for the report. 
I think we need to check
- if pcl_ros::transformPointCloud has the same issue mentioned here
- why we are using pcl_ros::transformPointCloud instead of pcl::transformPointCloud

h-ohta, 2023-05-25:
@kminoda 
> if pcl_ros::transformPointCloud has the same issue mentioned here

In this repository, below functions are used.

1. https://github.com/strawlab/perception_pcl/blob/master/pcl_ros/src/pcl_ros/transforms.cpp#L104-L207
2. https://github.com/strawlab/perception_pcl/blob/master/pcl_ros/src/pcl_ros/transforms.cpp#L46-L81

Please refer to https://github.com/search?q=repo%3Aautowarefoundation%2Fautoware.universe%20pcl_ros%3A%3AtransformPointcloud&type=code.

As a result, when sensor points are empty,  `Input dataset has no X-Y-Z coordinates! Cannot convert to Eigen format.` occurs but a node is not killed.

vividf, 2024-07-23:
@kminoda 
Seems that the issue is fixed.
Can we close this one? Thanks!

amadeuszsz, 2024-08-05:
The issue seems to be resolved. If the problem still exists, please open this thread.

closed 2024-08-05 by amadeuszsz