autowarefoundation/autoware_universe #3045: Re-origanize launch file for probabilisitic_occupancy_grid_map packages

YoshiRi, 2023-03-10:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Try to reorganize launch file for occuapncy grid map generation.

### Purpose

Problem Statements

- There are duplicated launch files in [tier4_perception_launch](https://github.com/autowarefoundation/autoware.universe/tree/main/launch/tier4_perception_launch/launch/occupancy_grid_map) and [probabilisitic_occupancy_grid_map](https://github.com/autowarefoundation/autoware.universe/blob/main/perception/probabilistic_occupancy_grid_map/launch/) packages.
  - This is confusing for users which file to edit 
- poor readability because of launch.py style
  - It's better to rewrite launch file in xml file
  - or create organizer launch to access each launch.py files

### Possible approaches

- Create origanizer launch.xml in tier4_perception_launch instead of current launch.py
  - From the launch.xml file, call launch.py in `probabilistic_occupancy_grid_map` packages.
  - Switchable between potincloud/laserscan based methods

### Definition of done

Reproduce the same behavior

mitsudome-r, 2023-03-14:
@YoshiRi
@Zeysthingz is willing to help with this task. @Zeysthingz Could you describe your approach (or create a draft PR) and align with @YoshiRi about this task?

YoshiRi, 2023-03-14:
@mitsudome-r  @Zeysthingz 
OK. Where should we discuss? I'll just describe my load map here.

Zeysthingz, 2023-03-14:
> 
I did not examine the issue in detail yet but I think we can discuss on discord if it's fine.

YoshiRi, 2023-03-14:
# My goal

As described in https://github.com/autowarefoundation/autoware.universe/issues/2906, I think we need to create occupancy grid map in each sensor frame to correctly reflect the FOV of the sensor.

So, my goal image is like following figures:

|Current|Final goal|
|--|--|
|[image]|[image]|

## Sub Goals

To achieve the goal, I 

- <1> Fix grid map generation
  - a. need to generate same grid map for fusion but different scan frame
  - b. manage each sensor FOV correctly
  - c. launch refactoring
- <2> Add fusion node
  - a. asyncronous grid map fusion with [[Binary Bayes/ Linear Opinion Pool/ Dempster-Shafer]]

## Related PRs

- 1.a: Separate grid map origin and scan origin :
  -  https://github.com/autowarefoundation/autoware.universe/pull/2939
- 1.b: Filter the obstacle pointcloud by raw pointcloud to correctly reflect FOV: 
  - https://github.com/autowarefoundation/autoware.universe/pull/3054
- 1.c: Refactor launch.xml
  - https://github.com/autowarefoundation/autoware.universe/pull/3058
- 2.a: under going...

YoshiRi, 2023-03-14:
> > 
> 
> I did not examine the issue in detail yet but I think we can discuss on discord if it's fine.

@Zeysthingz Fine, but I am not familiar with discord (just used chat space), could you guide me?

Zeysthingz, 2023-03-14:
> 

Yes sure I can ,we usually use discord for company stuffs. zeys#0531 is me you can add me.

idorobotics, 2023-10-04:
> # My goal
> 
> As described in #2906, I think we need to create occupancy grid map in each sensor frame to correctly reflect the FOV of the sensor.
> 
> So, my goal image is like following figures:
> Current 	Final goal
> [image] 	[image]
> ## Sub Goals
> 
> To achieve the goal, I think there should be some sub goals to be achieved.
> 
>     * <1> Fix grid map generation
>       
>       * a. need to generate same grid map for fusion but different scan frame
>       * b. manage each sensor FOV correctly
>       * c. launch refactoring
> 
>     * <2> Add fusion node
>       
>       * a. asyncronous grid map fusion with [[Binary Bayes/ Linear Opinion Pool/ Dempster-Shafer]]
> 
> 
> ## Related PRs
> 
>     * 1.a: Separate grid map origin and scan origin :
>       
>       * [feat(probablistic_occupancy_grid_map): add alternative to gridmap origin frame #2939](https://github.com/autowarefoundation/autoware.universe/pull/2939)
> 
>     * 1.b: Filter the obstacle pointcloud by raw pointcloud to correctly reflect FOV:
>       
>       * [feat(probabilistic_occupancy_grid_map): add pointcloud filter in gridmap generation #3054](https://github.com/autowarefoundation/autoware.universe/pull/3054)
> 
>     * 1.c: Refactor launch.xml

Looks like all PRs are merged. Can this issue be closed or is there other work outstanding?

technolojin, 2024-04-05:
I think this issue is good to be closed.

closed 2024-04-05 by technolojin