autowarefoundation/autoware_universe #3065: rename `localization_accuracy`

YamatoAndo, 2023-03-13:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Rename  `localization_accuracy`.
https://github.com/autowarefoundation/autoware.universe/blob/5ca8e689bf431413c1072fd8d9b523536beed129/localization/localization_error_monitor/src/node.cpp#L55

### Purpose

It does not indicate `accuracy`, so change to the correct wording.
like `precision` or `error_ellpize_size`, etc.

### Possible approaches

- [ ] modify the localization_error_monitor node and the yaml-file.
- [ ] modify the system_error_monitor's yaml-file.
- [ ] modify the fault_injection's yaml-file
- [ ] (If there is anything else relevant)

### Definition of done

`localization_accuracy` has been corrected to the correct wording.

Motsu-san, 2023-03-13:
I prefer `localization_error_ellpize_size`.

Motsu-san, 2023-08-23:
test

Motsu-san, 2023-09-28:
Eventually, I rename it `localization_error_ellipse` on the pull request https://github.com/autowarefoundation/autoware.universe/pull/5178.

closed 2024-01-31 by Motsu-san