autowarefoundation/autoware_universe #3112: Crop box filter node returns error message "Not implemented logic"

ajay1606, 2023-03-17:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hello,

While testing universe main branch with custom data, ended with following error message from `pointclodu_preprocessor`/ `crop_box_filter_nodelet.cpp`, But could understand cause of error. 

`[sensing.lidar.top.crop_box_filter_self]: Not implemented logic (faster_filter()`

But in the inside the `faster_filter()`, there link provided to implement the code logic to fix this bug. 

Have anyone encountered this before ? Is any fix available ? Or will it possible to describe the reason for this error ?

Appreciate any response. 

### Expected behavior

Successful run of crop box filter

### Actual behavior

Crop box filtering fail

### Steps to reproduce

Launch Autoware
Run recorded bag file

### Versions

Ubuntu 22.04
Humble
Autoware Universe : main

Custom LiDAR: Robosense 80ch

mitsudome-r, 2023-03-17:
It seems like the error comes from a point having invalid values. 
It should still crop the data with available points, but if you want to remove the error message, you should make sure that there are no points with inf values.
https://github.com/autowarefoundation/autoware.universe/blob/104ebf73bb16ccf2da625ad59bce70656d4662a5/sensing/pointcloud_preprocessor/src/crop_box_filter/crop_box_filter_nodelet.cpp#L136

mitsudome-r, 2023-03-17:
It seems like the error comes from a point having invalid values. 
It should still crop the data with available points, but if you want to remove the error message, you should make sure that there are no points with inf values. (cc. @sykwer)
https://github.com/autowarefoundation/autoware.universe/blob/104ebf73bb16ccf2da625ad59bce70656d4662a5/sensing/pointcloud_preprocessor/src/crop_box_filter/crop_box_filter_nodelet.cpp#L136-L142

ajay1606, 2023-03-17:
@mitsudome-r Thank you so much for your quick response, i will check lidar driver once, which causing NAN values in point cloud.

VRichardJP, 2023-03-20:
@ajay1606 I don't think replacing NaNs by any value is appropriate.
The crop box filter purpose to filter out points outside or inside a box. However whether a point is inside or outside the box is undecidable if its xyz coordinates have NaNs.
Keeping NaNs around is bad, because NaNs break so many common assumptions on float numbers. For example, any comparison with NaNs returns `false`:
```c++
#include <iostream>
#include <cmath>

int main() {
    std::cout << (std::nan("") < 0.) << std::endl;
    std::cout << (std::nan("") == 0.) << std::endl;
    std::cout << (std::nan("") > 0.) << std::endl;
    return 0;
}
```
These all print "0".

For example, `point_is_inside` is always `false` if the point data has any NaNs:
https://github.com/autowarefoundation/autoware.universe/blob/104ebf73bb16ccf2da625ad59bce70656d4662a5/sensing/pointcloud_preprocessor/src/crop_box_filter/crop_box_filter_nodelet.cpp#L145-L147

So if we use a negative box filter (keep only outside points) on a cloud with NaNs, the NaN points would be copied...

To avoid any issue, I think the best would be to skip NaN points and just print a warning when it happens (maybe count the number of NaN points).

ajay1606, 2023-03-20:
@VRichardJP @mitsudome-r Thank you so much for your inputs, certainly crop box filtering working well, by skipping nan values points. Assuming this as ideal solution for reported bug, issue closed for this moment.

closed 2023-03-20 by ajay1606

VRichardJP, 2023-03-20:
@ajay1606 Actually I think the code should be fixed to always skip points with NaNs, currently NaN may leak through when `param_.negative` is `true`.

Something like:
```C++
    if (!std::isfinite(point[0]) || !std::isfinite(point[1]) || !std::isfinite(point[2])) {
      // RCLCPP_WARN(...);
      continue;
    }

    if (transform_info.need_transform) {
        point = transform_info.eigen_transform * point;
    }
```

ajay1606, 2023-03-20:
@VRichardJP I also modified code almost similar to avoid NaN values. Thank you so much

ajay1606, 2023-05-25:
@TZECHIN6 It would be very helpful if could share error message shown in the terminal.

VRichardJP, 2023-05-25:
@TZECHIN6 Sorry but could you create a new separate issue for this? Your problem is totally unrelated to the reported issue here.

TZECHIN6, 2023-05-26:
@ajay1606 there is no error message as what I have spotted, might be I missed. Details can be found on the link below.
@VRichardJP sorry for that. I created a new one in https://github.com/autowarefoundation/autoware.universe/issues/3827#issuecomment-1563820937

zymouse, 2023-06-09:
With robosense radar, the NAN point can be removed by setting inside the driver file:https://github.com/RoboSense-LiDAR/rs_driver/blob/ef1bdd45ee4df51e5acb57b64e49d8fed233039c/doc/intro/04_parameter_intro.md