autowarefoundation/autoware_universe #3135: Move CUDA-dependent packages to a separate repository

esteve, 2023-03-22:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

As discussed in https://github.com/autowarefoundation/autoware/issues/3222#issuecomment-1478085589, in order to be able to build Autoware.Universe with the available packages in the ROS index, we decided to move those packages that depend on CUDA to a separate repository

### Purpose

Be able to compile a subset of the Autoware.universe without depending on CUDA

### Possible approaches

Move the the packages that depend on CUDA to a separate repository

### Definition of done

There are no packages that depend on CUDA in Autoware.universe

miursh, 2023-03-23:
Is there any options not to separate CUDA dependent packages?
Because of reasons below, I would not want to separate just a part of perception packages
* If some of perception packages are placed in the different repository, it is very confusing for developers. For example, when we want to add some change, we might need to make 3-4 PRs to different repository, such as new perception repo, autoware.unvierse, autoware_launch, and perception utils (new repo too?) 
* Very difficult to understand especially for new users 
* There might be some dependency between autoware.universe and separated packages. If there is many dependency, github CI might not work well.

xmfcx, 2023-03-24:
## Name of the new cuda packages repository
What are your naming choices about this new repository?

I'd go for `cuda_autoware_pkgs`, what are your suggestions?

## Make naming consistent
Also after moving the packages, we should make the naming conventions uniform too.

And for that my proposal is:
- `cuda_trt_package_name`
- `cuda_package_name` (we don't have any cuda only packages yet, just convention)
- `cuda_tvm_package_name` (this is also preserved for future, we will have a wide discussion on what to do with tvm packages in general soon)

## Handling of TVM packages
Also we should move `tvm` packages to `tvm_autoware_pkgs` and they will contain `tvm` packages for non-cuda targets.

This topic will also be discussed by @ambroise-arm @angry-crab @mitsudome-r @esteve @kenji-miyake and me.

kenji-miyake, 2023-04-17:
I'm sorry for joining late.

To give you some additional information, splitting repositories helps us reduce CI time. There is no need to test all packages with CUDA.
https://github.com/autowarefoundation/autoware.universe/blob/541122f6d214db36b1358c6ba02be2f8d8f7b089/.github/workflows/build-and-test-differential.yaml#L15-L17

But I understand splitting repositories may increase maintenance costs and troubles.

I think it's okay not to split repositories right now, we should do the following at least:
- As @xmfcx says, make package names consistent, which helps us ignore packages in CI.
  - Related: https://github.com/jspricke/ros-deb-builder-action/issues/19
- Plan a migration timeline to be CUDA-independent for hardware-agnostic, which I believe is one of the goals of Autoware.

I'll discuss how to proceed with this also in TIER IV.

esteve, 2023-04-18:
@kenji-miyake another reason for splitting the CUDA-related packages to another repository is because when/if we submit Autoware to the ROS buildfarm that Open Robotics maintains (https://build.ros.org/), all the packages in the repository are submitted and they need to be distributable and not covered by a proprietary license. 

However, following @mitsudome-r 's rationale of only submitting Autoware.core and not all of Autoware.universe to the ROS buildfarm, we can either leave CUDA packages in universe, or move them in a separate repository, but we can't migrate them to Autoware.core.

kenji-miyake, 2023-04-19:
> However, following @mitsudome-r 's rationale of only submitting Autoware.core and not all of Autoware.universe to the ROS buildfarm, we can either leave CUDA packages in universe, or move them in a separate repository, but we can't migrate them to Autoware.core.

Yes, I think so, too.

kenji-miyake, 2023-04-24:
As a result of TIER IV's internal discussion, we'd like to propose the following actions:
- To resolve the build issues in Debian package release, we'll rename existing packages, and don't move packages to another repository yet.
  - Having a unified naming convention can help us ignore specific packages in CI.
  - Moving packages to another repository has significant impacts, so we need step-by-step improvements.

For that, 
- Define naming conventions.
  - For CUDA/TensorRT packages, use `_trt`.
  - For TVM packages, use `_tvm`.
- Rename existing packages to follow the defined conventions.
  - In addition, maybe old-style packages should be fixed.
    - For example, https://github.com/autowarefoundation/autoware.universe/tree/aba954f30124155afe15ba4f216076101a125265/perception/lidar_apollo_segmentation_tvm_nodes can be merged into https://github.com/autowarefoundation/autoware.universe/tree/aba954f30124155afe15ba4f216076101a125265/perception/lidar_apollo_segmentation_tvm.
- Write the naming convention rules in the contribution guidelines.

---

@xmfcx @esteve cc @mitsudome-r What do you think about this?
If it looks good, we'd appreciate it if @esteve -san would work on these tasks as part of the Debian package issue.

Regarding the long-term milestones, I think as follows:
- We should continue discussing how we should utilize TVM.
- Each one should try to gain a deeper understanding around TVM through learning/experiments.

xmfcx, 2023-04-24:
I agree that we can start by renaming.

> - For CUDA/TensorRT packages, use `_trt`.
> - For TVM packages, use `_tvm`.

I'd prefer prefix over suffix. But I'm OK either way.

Also merging of `_nodes` double-packages will simplify the maintenance in general.

esteve, 2023-05-15:
> I agree that we can start by renaming.
> 
> > * For CUDA/TensorRT packages, use `_trt`.
> > * For TVM packages, use `_tvm`.
> 
> I'd prefer prefix over suffix. But I'm OK either way.

I also agree with using suffixes, makes it clear that they are an extension to the base packages or that they are based on other packages.

> Also merging of `_nodes` double-packages will simplify the maintenance in general.

I'm afraid I don't follow you here, can you elaborate? Thanks.

xmfcx, 2023-05-15:
> I'm afraid I don't follow you here, can you elaborate? Thanks.

For example,

- https://github.com/autowarefoundation/autoware.universe/tree/main/perception/lidar_apollo_segmentation_tvm
- https://github.com/autowarefoundation/autoware.universe/tree/main/perception/lidar_apollo_segmentation_tvm_nodes

could be merged into a single package.

esteve, 2023-05-15:
I can't clearly remember what was the rationale for having `_nodes` packages, I think it was inherited from Autoware.Auto because Apex.AI wanted the core functionality separate from the node code because it was incompatible with Apex.OS, but I might be making this up, I don't remember.

Anyway, I agree with you, it'd simplify our tree, especially now that we're enforcing all nodes to be components, so they all have uniform instantiation.

esteve, 2025-01-21:
@xmfcx @idorobotics @mitsudome-r sorry for reviving a stale issue, but it's not clear to me whether we reached an agreement here. I'm working on adding support for building CUDA-enabled packages in our buildfarm and it'd be good to sort this topic out. Thanks.

xmfcx, 2025-01-21:
How to structure the separated CUDA packages. The reason to separate CUDA packages is because CUDA is hardware specific.

For both Universe and Core, 2 new repositories:
- `autoware/src/universe/autoware.universe_cuda`
- `autoware/src/core/autoware.core_cuda` (if necessary, in the future)

**Additional:**
- CUDA dependent packages should fail if there is no CUDA.
- We should update the installation instructions as well for separated packages.

**Pros:**
- Easier to submit Universe to build farm
- Easier to maintain CI
- Easier to generate debian packages

**Cons:**
- More work to maintain with Universe
  - Multiple PRs

@YoshiRi @miursh @youtalk @oguzkaganozt

esteve, 2025-01-22:
@xmfcx let me know when the repository is created and I'll move the CUDA-dependent packages to it

esteve, 2025-01-22:
These are all the packages (and their downstream dependencies) that need to be moved:

<tt>
  <table>
    <tr>
      <th>autoware_image_projection_based_fusion</th>
    </tr>
    <tr>
      <td>autoware_image_projection_based_fusion</td>
      <td>perception/autoware_image_projection_based_fusion</td>
    </tr>
    <tr>
      <td>tier4_perception_launch</td>
      <td>launch/tier4_perception_launch</td>
    </tr>
    <tr>
      <th>autoware_lidar_centerpoint</th>
    </tr>
    <tr>
      <td>autoware_image_projection_based_fusion</td>
      <td>perception/autoware_image_projection_based_fusion</td>
    </tr>
    <tr>
      <td>autoware_lidar_centerpoint</td>
      <td>perception/autoware_lidar_centerpoint</td>
    </tr>
    <tr>
      <td>tier4_perception_launch</td>
      <td>launch/tier4_perception_launch</td>
    </tr>
    <tr>
      <th>autoware_lidar_transfusion</th>
    </tr>
    <tr>
      <td>autoware_lidar_transfusion</td>
      <td>perception/autoware_lidar_transfusion</td>
    </tr>
    <tr>
      <th>autoware_shape_estimation</th>
    </tr>
    <tr>
      <td>autoware_detection_by_tracker</td>
      <td>perception/autoware_detection_by_tracker</td>
    </tr>
    <tr>
      <td>autoware_shape_estimation</td>
      <td>perception/autoware_shape_estimation</td>
    </tr>
    <tr>
      <td>tier4_perception_launch</td>
      <td>launch/tier4_perception_launch</td>
    </tr>
    <tr>
      <th>autoware_tensorrt_classifier</th>
    </tr>
    <tr>
      <td>autoware_tensorrt_classifier</td>
      <td>perception/autoware_tensorrt_classifier</td>
    </tr>
    <tr>
      <td>autoware_traffic_light_classifier</td>
      <td>perception/autoware_traffic_light_classifier</td>
    </tr>
    <tr>
      <td>tier4_perception_launch</td>
      <td>launch/tier4_perception_launch</td>
    </tr>
    <tr>
      <th>autoware_tensorrt_common</th>
    </tr>
    <tr>
      <td>autoware_bytetrack</td>
      <td>perception/autoware_bytetrack</td>
    </tr>
    <tr>
      <td>autoware_detection_by_tracker</td>
      <td>perception/autoware_detection_by_tracker</td>
    </tr>
    <tr>
      <td>autoware_image_projection_based_fusion</td>
      <td>perception/autoware_image_projection_based_fusion</td>
    </tr>
    <tr>
      <td>autoware_lidar_apollo_instance_segmentation</td>
      <td>perception/autoware_lidar_apollo_instance_segmentation</td>
    </tr>
    <tr>
      <td>autoware_lidar_centerpoint</td>
      <td>perception/autoware_lidar_centerpoint</td>
    </tr>
    <tr>
      <td>autoware_lidar_transfusion</td>
      <td>perception/autoware_lidar_transfusion</td>
    </tr>
    <tr>
      <td>autoware_shape_estimation</td>
      <td>perception/autoware_shape_estimation</td>
    </tr>
    <tr>
      <td>autoware_tensorrt_classifier</td>
      <td>perception/autoware_tensorrt_classifier</td>
    </tr>
    <tr>
      <td>autoware_tensorrt_common</td>
      <td>perception/autoware_tensorrt_common</td>
    </tr>
    <tr>
      <td>autoware_tensorrt_yolox</td>
      <td>perception/autoware_tensorrt_yolox</td>
    </tr>
    <tr>
      <td>autoware_traffic_light_classifier</td>
      <td>perception/autoware_traffic_light_classifier</td>
    </tr>
    <tr>
      <td>autoware_traffic_light_fine_detector</td>
      <td>perception/autoware_traffic_light_fine_detector</td>
    </tr>
    <tr>
      <td>tier4_perception_launch</td>
      <td>launch/tier4_perception_launch</td>
    </tr>
    <tr>
      <th>autoware_tensorrt_yolox</th>
    </tr>
    <tr>
      <td>autoware_tensorrt_yolox</td>
      <td>perception/autoware_tensorrt_yolox</td>
    </tr>
    <tr>
      <td>autoware_traffic_light_fine_detector</td>
      <td>perception/autoware_traffic_light_fine_detector</td>
    </tr>
    <tr>
      <td>tier4_perception_launch</td>
      <td>launch/tier4_perception_launch</td>
    </tr>
    <tr>
      <th>autoware_traffic_light_classifier</th>
    </tr>
    <tr>
      <td>autoware_traffic_light_classifier</td>
      <td>perception/autoware_traffic_light_classifier</td>
    </tr>
    <tr>
      <td>tier4_perception_launch</td>
      <td>launch/tier4_perception_launch</td>
    </tr>
    <tr>
      <th>autoware_traffic_light_fine_detector</th>
    </tr>
    <tr>
      <td>autoware_traffic_light_fine_detector</td>
      <td>perception/autoware_traffic_light_fine_detector</td>
    </tr>
    <tr>
      <td>tier4_perception_launch</td>
      <td>launch/tier4_perception_launch</td>
    </tr>
    <tr>
      <th>autoware_cuda_utils</th>
    </tr>
    <tr>
      <td>autoware_bytetrack</td>
      <td>perception/autoware_bytetrack</td>
    </tr>
    <tr>
      <td>autoware_cuda_utils</td>
      <td>sensing/autoware_cuda_utils</td>
    </tr>
    <tr>
      <td>autoware_detection_by_tracker</td>
      <td>perception/autoware_detection_by_tracker</td>
    </tr>
    <tr>
      <td>autoware_lidar_apollo_instance_segmentation</td>
      <td>perception/autoware_lidar_apollo_instance_segmentation</td>
    </tr>
    <tr>
      <td>autoware_shape_estimation</td>
      <td>perception/autoware_shape_estimation</td>
    </tr>
    <tr>
      <td>autoware_tensorrt_classifier</td>
      <td>perception/autoware_tensorrt_classifier</td>
    </tr>
    <tr>
      <td>autoware_tensorrt_yolox</td>
      <td>perception/autoware_tensorrt_yolox</td>
    </tr>
    <tr>
      <td>autoware_traffic_light_classifier</td>
      <td>perception/autoware_traffic_light_classifier</td>
    </tr>
    <tr>
      <td>autoware_traffic_light_fine_detector</td>
      <td>perception/autoware_traffic_light_fine_detector</td>
    </tr>
    <tr>
      <td>tier4_perception_launch</td>
      <td>launch/tier4_perception_launch</td>
    </tr>
  </table>
</tt>

xmfcx, 2025-01-22:
Thanks @esteve , here is the single level version of your table:

<table>
  <tr>
    <th>#</th>
    <th>Package Name</th>
    <th>Relative Path</th>
  </tr>
  <tr>
    <td>1</td>
    <td>autoware_bytetrack</td>
    <td>perception/autoware_bytetrack</td>
  </tr>
  <tr>
    <td>2</td>
    <td>autoware_cuda_utils</td>
    <td>sensing/autoware_cuda_utils</td>
  </tr>
  <tr>
    <td>3</td>
    <td>autoware_detection_by_tracker</td>
    <td>perception/autoware_detection_by_tracker</td>
  </tr>
  <tr>
    <td>4</td>
    <td>autoware_image_projection_based_fusion</td>
    <td>perception/autoware_image_projection_based_fusion</td>
  </tr>
  <tr>
    <td>5</td>
    <td>autoware_lidar_apollo_instance_segmentation</td>
    <td>perception/autoware_lidar_apollo_instance_segmentation</td>
  </tr>
  <tr>
    <td>6</td>
    <td>autoware_lidar_centerpoint</td>
    <td>perception/autoware_lidar_centerpoint</td>
  </tr>
  <tr>
    <td>7</td>
    <td>autoware_lidar_transfusion</td>
    <td>perception/autoware_lidar_transfusion</td>
  </tr>
  <tr>
    <td>8</td>
    <td>autoware_shape_estimation</td>
    <td>perception/autoware_shape_estimation</td>
  </tr>
  <tr>
    <td>9</td>
    <td>autoware_tensorrt_classifier</td>
    <td>perception/autoware_tensorrt_classifier</td>
  </tr>
  <tr>
    <td>10</td>
    <td>autoware_tensorrt_common</td>
    <td>perception/autoware_tensorrt_common</td>
  </tr>
  <tr>
    <td>11</td>
    <td>autoware_tensorrt_yolox</td>
    <td>perception/autoware_tensorrt_yolox</td>
  </tr>
  <tr>
    <td>12</td>
    <td>autoware_traffic_light_classifier</td>
    <td>perception/autoware_traffic_light_classifier</td>
  </tr>
  <tr>
    <td>13</td>
    <td>autoware_traffic_light_fine_detector</td>
    <td>perception/autoware_traffic_light_fine_detector</td>
  </tr>
  <tr>
    <td>14</td>
    <td>tier4_perception_launch</td>
    <td>launch/tier4_perception_launch</td>
  </tr>
</table>

esteve, 2025-01-22:
@xmfcx thank. Reading the table, it seems clearer to me that `autoware_cuda_utils` would better be moved to `perception/`, none all of its dependent packages are in `sensing/`

esteve, 2025-01-22:
@xmfcx the packages and their dependencies I posted only considered one level of recursive dependencies. Here's a table with all the recursive dependencies (e.g. `autoware_launch` which depends on `tier4_perception_launch` wasn't listed), this table should have all the recursive dependencies:

<table>
<tr><td>autoware_bytetrack</td><td>src/universe/autoware.universe/perception/autoware_bytetrack</td></tr>
<tr><td>autoware_cuda_utils</td><td>src/universe/autoware.universe/sensing/autoware_cuda_utils</td></tr>
<tr><td>autoware_detection_by_tracker</td><td>src/universe/autoware.universe/perception/autoware_detection_by_tracker</td></tr>
<tr><td>autoware_image_projection_based_fusion</td><td>src/universe/autoware.universe/perception/autoware_image_projection_based_fusion</td></tr>
<tr><td>autoware_launch</td><td>src/launcher/autoware_launch/autoware_launch</td></tr>
<tr><td>autoware_lidar_apollo_instance_segmentation</td><td>src/universe/autoware.universe/perception/autoware_lidar_apollo_instance_segmentation</td></tr>
<tr><td>autoware_lidar_centerpoint</td><td>src/universe/autoware.universe/perception/autoware_lidar_centerpoint</td></tr>
<tr><td>autoware_lidar_transfusion</td><td>src/universe/autoware.universe/perception/autoware_lidar_transfusion</td></tr>
<tr><td>autoware_shape_estimation</td><td>src/universe/autoware.universe/perception/autoware_shape_estimation</td></tr>
<tr><td>autoware_tensorrt_classifier</td><td>src/universe/autoware.universe/perception/autoware_tensorrt_classifier</td></tr>
<tr><td>autoware_tensorrt_common</td><td>src/universe/autoware.universe/perception/autoware_tensorrt_common</td></tr>
<tr><td>autoware_tensorrt_yolox</td><td>src/universe/autoware.universe/perception/autoware_tensorrt_yolox</td></tr>
<tr><td>autoware_traffic_light_classifier</td><td>src/universe/autoware.universe/perception/autoware_traffic_light_classifier</td></tr>
<tr><td>autoware_traffic_light_fine_detector</td><td>src/universe/autoware.universe/perception/autoware_traffic_light_fine_detector</td></tr>
<tr><td>tier4_perception_launch</td><td>src/universe/autoware.universe/launch/tier4_perception_launch</td></tr>
</table>

In any case, for a first step, we would only move the CUDA-dependent packages in `universe` to another repo. We can discuss later how we can deal with packages that reside in other repositories but depend on CUDA in some way.

knzo25, 2025-02-03:
@esteve 
I have several PRs that introduce CUDA to sensing. Planning it also considering using TensorRT AFAIK

xmfcx, 2025-02-03:
In addition to my [comment above](https://github.com/autowarefoundation/autoware.universe/issues/3135#issuecomment-2605029526),

## Option 0: Do nothing

- Right now, if machine doesn't have CUDA installed, the CUDA dependent packages will build an empty package. (misleading behavior)
- We don't have a clear way to separate CUDA packages to handle them with CI or release them to the build farm.

## Option 1: Move CUDA packages to a separate repository

- **Repo name:** autoware_universe_cuda
- New repo means, we need to update the `autoware.repos` and `autoware-nightly.repos` files to not have them. And provide them with something like `autoware-with-cuda.repos` and `autoware-nightly-with-cuda` repos. This would make things way too confusing and hard to maintain.
- We need to worry about a change in Universe breaking the build for CUDA repository.
  - This is because the new repo will depend on Universe. The packages from universe that CUDA packages currently depend:
    - autoware_component_interface_specs_universe
    - autoware_component_interface_utils
    - autoware_euclidean_cluster
    - autoware_image_transport_decompressor
    - autoware_interpolation
    - autoware_object_recognition_utils
    - autoware_pcl_extensions
    - autoware_pointcloud_preprocessor
    - autoware_pyplot
    - autoware_test_utils
    - autoware_universe_utils
    - perception_utils
- For any changes in the packages above, we need to test if CUDA packages still compile. (Additional CI work if moved to new repo)
- Many changes will be more painful to introduce because we will have to create multiple PRs for Universe and CUDA repo.

## Option 2: Make CUDA hard dependency for Autoware

If we decide to keep CUDA packages in the same repository, and as @knzo25 said, if we will make more and more packages depend on CUDA, maybe we can just declare CUDA as a hard dependency for the repo as a whole too. As in, we will say that Autoware depends on CUDA, this wouldn't be too weird since many AI projects depend on it anyway.

This would mean, let's just stop trying to make things work in a non-CUDA environment for Autoware as a whole, there is no deployment of an autonomous vehicle (that I'm aware of) that doesn't already have an NVIDIA hardware anyways. And Autoware won't drive autonomously without CUDA too, all these perception packages are very critical for Autonomous driving.

## Option 3: Keep things same, introduce meta-dependency-package

**Goal:**
- Make it easy to filter out CUDA packages for CI or building purposes.

We could also just keep things the same, and somehow make a meta-package as such:
- **Package name:** `autoware_cuda_dependency_meta`
- The package only has a basic `CMakeLists` that checks for CUDA dependency.
- All of the Autoware packages that depend on CUDA has to depend on this meta-package.
- This way, if we want to filter for CUDA packages for CI purposes, we can call `colcon list --packages-above autoware_cuda_dependency_meta --names-only`
- Right now, we can also call `colcon list --packages-above autoware_tensorrt_common autoware_cuda_utils --names-only` for a similar effect.

And here is the "above" dependency graph for all CUDA dependent packages in Autoware:

`colcon graph --packages-above autoware_cuda_utils autoware_tensorrt_common --dot | dot -Tsvg -Grankdir=LR -Gnodesep=0.2 -Granksep=0.3 -Gconcentrate=true -o graph.svg`

[image]

## My choice

Personally I would go with either **Option 2 or 3** because moving a repo out would be very painful and keeping things same makes life harder for CI, Build farm, Debian file generation and also causes confusion.

**Option 2** might come out too harsh, as this would mean we won't have ROS build farm as a general goal for the Universe. But at the same time, it also has some appeal and would make life simpler if we just stop worrying about CUDA/noCUDA thing once and for all and just go full CUDA.

**Option 3** I think is the safest, easiest to implement and least obtrusive one. Also would solve all of our problems.

esteve, 2025-02-03:
@xmfcx thanks for summarizing the different options.

For option 1:

> We need to worry about a change in Universe breaking the build for CUDA repository.

I wonder how often this happens. And if it does, it might be a sign that packages are too tightly coupled. It shouldn't happen on a regular basis and packages should aim at being stable in their interfaces by now. One way to mitigate some issues like this would be to move the entirety of perception to its own repository, with or without CUDA. The other packages in `common` (like `*_utils`) shouldn't break their interface often, otherwise we're doing something wrong in our development process.

For option 2:

> Personally I would go with either Option 2 or 3 because moving a repo out would be very painful and keeping things same makes life harder for CI, Build farm, Debian file generation and also causes confusion.

Option 2 would actually make it very difficult to release Debian packages. I'd have to try, but it might make it even impossible. It'd be possible if we used the CUDA package from Ubuntu (rosdistro has a key for nvidia-cuda), but since we're using the latest package from nVidia, we can't rely on rosdistro (ithe package also has a different name).

For option 3:

> We could also just keep things the same, and somehow make a meta-package as such:

I actually implemented something like that in https://github.com/esteve/autoware.universe/tree/add-cuda-deps, adding an thin package called "cuda" and make the CUDA-dependant packages depend on t, but I didn't fully explore it. I can give it another try and see if it works.

mitsudome-r, 2025-02-03:
Thanks @xmfcx and @esteve 
> I actually implemented something like that in https://github.com/esteve/autoware.universe/tree/add-cuda-deps, adding an thin package called "cuda" and make the CUDA-dependant packages depend on t, but I didn't fully explore it. I can give it another try and see if it works.

If we can do this, then I think it would be the least disruptive modification for the current users.

yukkysaito, 2025-02-04:
Thanks 👍 
Personally, I choose Option 3.

Option 0 is already the current approach that has been taken.
Option 1 is, in my opinion, out of the question.

Option 2 is a good option. However, I remember that in previous discussions, Autoware has aimed to avoid being vendor-locked to NVIDIA. For example, we decided not to officially depend on TensorRT and instead transition to the open-source TVM.

While in practice, not using NVIDIA is hardly an option at the moment, the question is whether AWF should consider supporting alternatives to NVIDIA in the future. I believe this is something that should be further discussed within AWF's WG.

Option 3 seems to be the most manageable and the best choice.

esteve, 2025-02-04:
@yukkysaito 

> Option 1 is, in my opinion, out of the question.

Could you elaborate on that? Thanks.

esteve, 2025-02-04:
@xmfcx another approach, which could be done in addition to option 3, is to move the CUDA-dependant packages to a subfolder of its own (e.g. `cuda`) and recreate the folder structure there (e.g. `cuda/perception`, `cuda/sensing`, etc.) This would have the benefit of simplifying how to build and test a non-CUDA environment by just excluding all the package from the `cuda` folder.

xmfcx, 2025-02-04:
Right now it is possible to build non-cuda packages with:
`colcon build --packages-ignore $(colcon list --packages-above autoware_tensorrt_common autoware_cuda_utils --names-only | tr '\n' ' ')`

I'd still prefer the meta-package approach to keep the folder structure simpler.

esteve, 2025-02-04:
@xmfcx I know, but the same way we moved the visualization plugins to a separate folder, it just made the Docker files easier to read.

xmfcx, 2025-02-04:
Upon the discussions here and in [the Software WG](https://github.com/orgs/autowarefoundation/discussions/5728), we'll go with the Option 3 and create a `autoware_cuda_dependency_meta` package and make first-hand CUDA dependent packages to depend on this meta-package.

Closing this issue for now.

closed 2025-02-04 by xmfcx

esteve, 2025-02-04:
@xmfcx since I already started the work, I'd like to get the task of creating a meta package assigned to me. Thanks.

youtalk, 2025-02-04:
AWF, like ARM being a platinum sponsor, must aim for independence from specific hardware vendors. Therefore, major packages should avoid strong dependencies on CUDA.
Thus, Autoware Core will first aim to be vendor-free. Additionally, vendor-dependent packages should be separated into different repositories from the perspectives of version control and isolation.

esteve, 2025-02-05:
@youtalk 

> Additionally, vendor-dependent packages should be separated into different repositories from the perspectives of version control and isolation.

I completely agree. However, we reached a compromise in the ASWG meeting that this decision could be revisited in the future, I suspect there will be issues if we attempt to submit packages from Autoware.universe to the ROS Buildfarm if there are packages that depend on CUDA in the same repository.

Also, it might be worth revisiting TVM (or something similar), since it'd help us decouple some of the code from CUDA and other vendor-specific APIs.