autowarefoundation/autoware_universe #3137: Disable downloading ONNX models by default

esteve, 2023-03-22:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Some packages download ONNX models by default, which is not necessary for compiling them. Additionally, during the process for generating Debian packages, access to the network is restricted, so building these packages fail (see https://github.com/autowarefoundation/autoware/issues/3222#issuecomment-1478085589)

The packages that download ONNX models are:

 - `image_projection_based_fusion`
 - `lidar_centerpoint`
 - `tensorrt_yolo`
 - `tensorrt_yolox`
 - `traffic_light_classifier`
 - `traffic_light_ssd_fine_detector`

### Purpose

Be able to build the packages without requiring to download the ONNX models

### Possible approaches

Change the download logic so that ONNX models are not downloaded by default

### Definition of done

Packages can be built without downloading the ONNX models

yukkysaito, 2023-03-23:
Thank you for your proposal.
If DOWNLOAD_ARTIFACTS is default false, won't the rosbag replay simulator stop working?

esteve, 2023-03-23:
@yukkysaito thanks for your comment. I've only submitted changes to the packages that download ONNX models. Could you link to the packages that download `rosbag` files? Thanks.

In any case, the reason for not downloading external files by default is because the processes we use for generating Debian packages restrict access to the network (as is the Debian policy for building packages), so we can't download files in `CMakeLists.txt`. I have some questions:

 - Is there another way to distribute the `rosbag` files?
 - Perhaps a separate package or adding instructions for users?
 - How big are the files?
 - Maybe we could use something like `git-lfs` to store them in Git?

Thanks.

yukkysaito, 2023-03-23:
@esteve sorry, I may have written it badly and caused a misunderstanding.
I am not talking about rosbag, but rosbag replay simulator. :pray: 
https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/

rosbag replay simulator uses DNN such as centerpoint, so it is necessary to download the onnx file.
If the default is set to false, the onnx file will not be downloaded and the rosbag replay simulator in the tutorial will not work.

I would preferably download it when setting up the environment in ansible, so that no internet connection is needed at build time.

esteve, 2023-03-23:
@yukkysaito ah, sorry, I misunderstood. Thanks, I agree it'd be much easier if downloading the ONNX models is done outside the build process of `CMakeLists.txt`. I can remove the logic from `CMakeLists.txt` and we can update the `ansible` scripts. I'm not familiar with `ansible`, but I can try.

xmfcx, 2023-03-24:
I agree, the links are in the `CMakeLists.txt` file and most are in https://web.auto domain name right now.

Moving them to ansible and also adding the manual steps in the readme of the [role](https://github.com/autowarefoundation/autoware/tree/main/ansible/roles/) should be enough.

And yes like @esteve suggested we should remove the remains of the downloads from the `CMakeLists.txt` files.

esteve, 2023-03-24:
Following this discussion I've closed https://github.com/autowarefoundation/autoware_common/pull/164 and will rework the rest of the PRs to remove the download logic. I'll also update the documentation and the ansible scripts.

yukkysaito, 2023-03-25:
There are a few things to consider for downloading with ansible.

- Where to put the downloads?
- How do DNN pkgs (centerpoint, etc.) refer to the downloaded file from launch?

As an example,to refer to the downloaded file in launch file by ansible, we can create a specific package, put the weight file there, and refer to it from launch.
Do you have any other suggestions?

xmfcx, 2023-03-28:
@yukkysaito

Right now Autoware has `map_path` as an external directory necessary for the runtime:
```bash
ros2 launch autoware_launch logging_simulator.launch.xml \
map_path:=$HOME/autoware_map/sample-map-rosbag \
vehicle_model:=sample_vehicle \
sensor_model:=sample_sensor_kit
```

What about adding a `neural_net_models` path too?

Then this path could be passed to the necessary nodes.

esteve, 2023-03-28:
> There are a few things to consider for downloading with ansible.
>
>     * Where to put the downloads?
>
>     * How do DNN pkgs (centerpoint, etc.) refer to the downloaded file from launch?
>
>
> As an [...]

Sorry, I'm afraid I'm not following here. Will the package store the weight file? And if not using `ansible`, how will the package get populated with the model files? Could you explain further? Thanks.

esteve, 2023-03-28:
@xmfcx

> What about adding a `ml_models_path` (machine learning) path too?
> 
> Then this path could be passed to the necessary nodes.

I think this is a good idea, it'll standardize how to access the models across all nodes. What path by default should we tell users to download files to? With `ansible` we can decide for them, but I'm thinking of when Autoware is installed manually. 

Also, when/if we generate Debian packages for these packages, should the models be installed system-wide or in a user's folder?

yukkysaito, 2023-03-30:
>What about adding a ml_models_path (machine learning) path too?
Then this path could be passed to the necessary nodes.

I think the map path is necessary because it specifies the environment in which to run, but ml_model_path seems to be a parameter of convenience inside Autoware.
So I would like to make the user don't care it, is there a good way to do this?

xmfcx, 2023-09-05:
Related: https://github.com/orgs/autowarefoundation/discussions/3649

esteve, 2023-11-10:
Closing this ticket now that all the linked PRs are merged. Thanks everyone, especially @lexavtanke

closed 2023-11-10 by esteve