autowarefoundation/autoware_universe #3199: `cropPoints` functions should not delete points within range?

VRichardJP, 2023-03-29:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

As pointed out here: https://github.com/autowarefoundation/autoware.universe/pull/3176, I am wondering if the `cropPoints` function should really delete the first point beyond crop distance. 

Here is an example for `cropForwardPoints`:
[image]

In the diagram, there are 3 arrows:
- in red, the distance the caller asked to be cropped
- in blue, the actual cropped distance if we delete points beyond crop distance (current behavior)
- in green, the actual cropped distance if we want to guarantee the forward length of the output path is at least longer than cropped distance.

In the current implementation, the output path is always shorter than crop distance. But is it what we want? I think the function is used to remove points that are too far from a pose, while making sure we keep some necessary distance forward/backward.

For example, if the distance is shorter than the points step, the current implementation delets all the points beyond `target_seg_idx`:
[image]

### Expected behavior

Make sure the cropped path is at least longer than crop distance

### Actual behavior

cropped path is shorted than crop distance. 

### Steps to reproduce

N/A

takayuki5168, 2023-03-29:
@VRichardJP
Thank you for the issue.

The reason for my implementation like the output path is always shorter than the designated distance, is I wanna make the backward path as short as possible around 5 meters.

This is because of the following complicated reasons.
- If the backward path is quite shorter than 5 meters, the precision of curvature smoothing in mpc_lateral_controller will be worse and the steering is jerky on the physical vehicle.
- If the backward path is long enough,[ the path cut function for overlapping route](https://github.com/autowarefoundation/autoware.universe/blob/1d23e178cb09ece32d9d69998ce1f8e860e9dd1f/planning/behavior_path_planner/src/utilities.cpp#L1005) keeps cutting in a case where the route's loop length is very short shown in the video.

[Screencast from 2023年03月28日 23時22分05秒.webm](https://user-images.githubusercontent.com/20228327/228436662-2b6c8ff9-f889-43e4-bb0d-4fda1b4e6112.webm)

takayuki5168, 2023-03-29:
IMO, in order not to make the output path's forward/backward lengths undefined (depending on the original resampling interval), the best solution for cropPoints is doing the resample containing the two end points, the point `backward_length` backward from the designated position and the point `forward_length` forward from the designated position.

What do you think?

VRichardJP, 2023-03-29:
Interpolating the end points is a cleaner solution indeed.
I am wondering if it would not make MPC solution more stable as well. Currently the path tail "jumps" from one point to the next. If the crop distance is exact, would it improve the planner stability?

VRichardJP, 2023-03-29:
I meant:
- With the current crop, the backward distance is non linear (it "jumps") = MPC solution could "jump" as well?
- With the interporlated crop, the backward distance would be fixed = MPC solution more stable?

takayuki5168, 2023-03-30:
That's right.