autowarefoundation/autoware_universe #3208: autoware.launch.xml doesn't set initial pose without gnss

annb3, 2023-03-29:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I am running autoware autouniverse with my personal vehicle and sensor. The planning launcher works perfectly, however, when I try to run the global launcher, independently if I use my vehicle and its sensor configuration or the sample_vehicle with sample_sensor_kit, when I set the pose from rviz2 or by topic the vehicle doesn't go in the initial position and the log seems not to have receive the initial position. I don't know why. May I wrong something? 

### Expected behavior

I expected that the initial pose could be set or using GNSS or by topic or rviz2 independently and moves the vehicle in that position of the chosen map with the repository vehicle.

### Actual behavior

The initial pose isn't set using the rviz2 initializer and also in the log file doesn't seemd it has received the command, totally strange!

### Steps to reproduce

clone the autoware auto universe repository using docker updating it at the last version pushed using ros2humble.
compile the package and inside the docker environment run:

`ros2 launch autoware_launch autoware.launch.xml map_path:=$HOME/autoware_map/sample-map-planning vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit`

### Versions

- OS: ubuntu 20.04LTS
- ROS_version: ROS2 humble
- docker environment 
- autowareautouniverse: installed not using pre-compiled package but in the docker env

### Possible causes

I don't know, seems that without the gnss position it is unable to set the initial pose from rviz2 button

### Additional context

NA

isamu-takagi, 2023-03-29:
What do you get when you filter the terminal log by `grep /api/localization/initialize`? If the following log appears, the vehicle stop condition is not satisfied.

```
[component_container_mt-33] [INFO]: server call: /api/localization/initialize
[component_container_mt-33] [INFO]: server exit: /api/localization/initialize
[service_log_checker-2] [ERROR]: /api/localization/initialize: status code 1 'The vehicle is not stopped.' (/default_ad_api/node/localization)
[service_log_checker-2] [ERROR]: /api/localization/initialize: status code 1 'The vehicle is not stopped.' (/default_ad_api/helpers/initial_pose_adaptor)
```
Please check if the `/sensing/vehicle_velocity_converter/twist_with_covariance` topic is output. Or change [here](https://github.com/autowarefoundation/autoware_launch/blob/211dfb4bce4d79b48bbc3b22d7cb838e7404da93/autoware_launch/config/localization/pose_initializer.param.yaml#L6) to false to disable the stop check. Comment out [here](https://github.com/autowarefoundation/autoware.universe/blob/07b7837d42309376dc55bf576e9bcba1497c1799/launch/tier4_localization_launch/launch/util/util.launch.xml#L23-L25) if the log is hard to see due to automatic initialization.

annb3, 2023-03-29:
[image]
this is what happens if i comment [here](https://github.com/autowarefoundation/autoware.universe/blob/07b7837d42309376dc55bf576e9bcba1497c1799/launch/tier4_localization_launch/launch/util/util.launch.xml#L23-L25)

annb3, 2023-03-29:
Differently not changing anything, when somethimes happens that the `grep /api/localization/initialize` doesn't gives error, the `/sensing/vehicle_velocity_converter/twist_with_covariance` is the following:
[image]

annb3, 2023-03-30:
But it is not my vehicle, it is the vehicle that I have downloaded with the entire repository:
map_path: sample-map-planning 
vehicle_model: sample_vehicle
sensor_model: sample_sensor_kit
everything directly downloaded from autowareautouniverse repository

isamu-takagi, 2023-03-30:
`planning_simulator.launch.xml` is launching `autoware.launch.xml` with parameters changed for simulation. Launching `autoware.launch.xml` directly will not work without connecting to sensors and vehicles.

annb3, 2023-03-31:
Ah, ok, so now everything it is clear.
Just the last doubt: is it not possible to use the sensing part with simulation, so we can use it once connected the real vehicle and sensors or can I use it independently?

Just to understand if it is possible to test additional functionalities in simulation or I have to do it in real-life.
Thanks @isamu-takagi

VRichardJP, 2023-04-03:
You could drive manually with your vehcile while running sensing and vehicle modules from `autoware.launch.xml`, and recording the messages from `/sensing/*` and `/vehicle/*` topics (not everything, just the raw sensor output and vehicle feedback). You could then replay the rosbag with `logging_simulator.launch.xml` and see how Autoware would behave.

annb3, 2023-04-03:
Ok, clear.
So thanks a lot for the support, extremely efficient.
@VRichardJP @isamu-takagi

closed 2023-04-03 by annb3