autowarefoundation/autoware_universe #3215: On abusive use of `reinterpret_cast` UB

VRichardJP, 2023-03-30:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

As pointed [here](https://github.com/autowarefoundation/autoware.universe/pull/3014#discussion_r1128758988), there are places in Autoware where `reinterpret_cast` is abusively used to "convert" data types.

[There are endless talks over the internet](https://stackoverflow.com/questions/573294/when-to-use-reinterpret-cast/) about how misleading `reinterpret_cast` is. Basically, dereferencing a reinterpreted pointer `p` with `*reinterpret_cast<T*>(p)` is only valid if `p` originally points to data of type `T` (e.g. `p = new T()`). If `p` originally points to any other type, even if "it looks" the same, is straight UB.

For example, copying `PointXYIZ` from `uint8_t[]` is UB:

https://github.com/autowarefoundation/autoware.universe/blob/cb103ec9c8149fffcf2a665ca37b69fe465e0a61/sensing/pointcloud_preprocessor/src/outlier_filter/ring_outlier_filter_nodelet.cpp#L123

Similarly, even though `PointXYZIRADRT` and `PointXYZI` "start the same", it is UB to use `reinterpret_cast` to copy between the 2 structs. For example:
```C++
PointXYZIRADRT p1{};
PointXYZI p2 = *reinterpret_cast<PointXYZI*>(&p1); // UB
```

It may work now... until the compiler decides otherwise.

### Purpose

Remove all UB `reinterpret_cast` in Autoware code base.

### Possible approaches

grep!

### Definition of done

No more `reinterpret_cast` UB.

VRichardJP, 2023-04-01:
# Taxonomy of `reinterpret_cast` usage in autoware.universe

## 1. Passing pointer to an API

For example:

```C++
sockaddr_in client;
int new_sock = accept(sock, reinterpret_cast<sockaddr *>(&client), &len);
```

Valid. Exactly what `reinterpret_cast` has been designed for.
Note that it is recommended to use `static_cast` in the case it is only necessary to cast from/to `void*`.

## 2. Reading/Writing an object byte representation

For example:

```C++
file.write(reinterpret_cast<const char *>(plan_->data()), plan_->size());
```

Valid. It is possible to cast into `char*`, `unsigned char*`, `std::byte*` and their `const` variants.
Writing the object bytes is legal, but not portable (e.g. because endianess, data type size, etc).

## 3. Reading/Writing from/into a buffer as if is was an object

For example:

```C++
reinterpret_cast<float32_t *>(v.data())[i] = dis(gen)
```

This is UB and dangerous. I know everyone does that and "it works" because most compilers are kind enough, but it's a tic-toc bomb you leave behind.
First, why is that not valid C++ code? C++ has [many complex rules regarding what makes a valid "object"](https://en.cppreference.com/w/cpp/language/object) and using `reinterpret_cast<T*>` on a buffer is not one of them. `reinterpret_cast` is basically a way to say to the compiler "trust me, someone else has created a valid object here before".

In practice however, this UB is known to work perfectly fine under the following conditions:

1. `T` is trivially copiable (such as `int`, `float`, `struct { int, char }`)
2. the buffer meet the alignment requirement of `T` (e.g. `float` must be aligned on 4 bytes)
3. you are not messing around with aliasing (because you violate strict aliasing rule)

Condition 1 is not a problem in autoware code base, as I have ever only seen `reinterpret_cast` with simple types and structs.
Condition 3 is normally not a problem in this context. To make the compiler break the program because of strict aliasing, you [really have to want it](https://gist.github.com/shafik/848ae25ee209f698763cffee272a58f8#preliminary-examples).
Condition 2 is the important one. If the buffer is not aligned properly, the compiler may definitely break the program. Luckily (or unluckily depending how you see it) modern x86 and ARM64 can access badly aligned data (just slightly slower). So the program may still run fine when optimization level is low. However if you start to push the compiler for more advanced optimizations (e.g. `-msse`, `-march=native`), then your program will either compute rubbish and simply fail. A good illustration [here](http://pzemtsov.github.io/2016/11/06/bug-story-alignment-on-x86.html).

If 1, 2 and 3 are met, the `reinterpret_cast` is officially an undefined behavior but works in practice. However, if you can't be 100% sure the data is properly aligned (for example if the buffer is created by another library), then [it is better to use `std::memcpy`](https://gist.github.com/shafik/848ae25ee209f698763cffee272a58f8#type-punning-arrays). With `-O2` or `-O3` the `memcpy` has no performance impact over the unsafe `reinterpret_cast`.

## 4. Reading/Writing data from different types

This is just a worse version of 3. Even if the types looks the same, there is no reason not to copy the data properly, which can't break the program and will be optimized out by the compiler.

## Grep time

Here is a list of all `reinterpret_cast` use in universe, categorized by pattern (from 1 to 4). For type 3, I make the difference between A and B: A when the cast is "safe" (typically alignment is correct) and B when it smells:

```grep
perception/traffic_light_ssd_fine_detector/src/nodelet.cpp
240:        reinterpret_cast<const float *>(pixels.datastart), // Type 3A: cv::Mat data is aligned on 16-bytes
241:        reinterpret_cast<const float *>(pixels.dataend));  // Type 3A: cv::Mat data is aligned on 16-bytes

perception/traffic_light_ssd_fine_detector/lib/src/trt_ssd.cpp
164:  file.write(reinterpret_cast<const char *>(plan_->data()), plan_->size()); // Type 2

perception/lidar_apollo_instance_segmentation/src/detector.cpp
72:    outfile.write(reinterpret_cast<char *>(plan->data()), plan->size()); // Type 2

perception/lidar_apollo_instance_segmentation/lib/include/Utils.hpp
122:  *reinterpret_cast<T *>(buffer) = val;        // Type 3B: unaligned write -> should use memcpy instead
129:  val = *reinterpret_cast<const T *>(buffer);  // Type 3B: unaligned read -> should use memcpy instead
[... 105 lines cut ...]
154:        *reinterpret_cast<float *>(&output.data[output_size + z_offset]) = point[2]; // Type 3B: unknown data alignment

sensing/pointcloud_preprocessor/src/outlier_filter/ring_outlier_filter_nodelet.cpp
61:    input_ring_map[*reinterpret_cast<uint16_t *>(&input_ptr->data[idx + ring_offset])].push_back( // Type 3B: unknown data alignment
87:        *reinterpret_cast<float *>(&input_ptr->data[current_idx + azimuth_offset]);  // Type 3B: unknown data alignment
89:        *reinterpret_cast<float *>(&input_ptr->data[next_idx + azimuth_offset]); // Type 3B: unknown data alignment
94:        *reinterpret_cast<float *>(&input_ptr->data[current_idx + distance_offset]); // Type 3B: unknown data alignment
96:        *reinterpret_cast<float *>(&input_ptr->data[next_idx + distance_offset]); // Type 3B: unknown data alignment
[... cut ...] _t == unsigned char).
221:            &rgb, reinterpret_cast<const char *>(&input_->points[cp]) + rgba_index, sizeof(int)); // Type 2
272:            &rgb, reinterpret_cast<const char *>(&input_->points[cp]) + rgba_index, sizeof(int)); // Type 2
```

As explained, pattern 1 and 2 are valid, 3A is "UB but safe" so acceptable, however I think code with pattern 3B and 3 should be fixed. Even if the code "works" today, it may not be the same if new platforms, compiler or optimization flags are used.

ambroise-arm, 2023-04-19:
Very interesting, thanks for the report.

> With -O2 or -O3 the memcpy has no performance impact over the unsafe reinterpret_cast.

Do you have resources on this point? It is not obvious to me that it is true.

VRichardJP, 2023-04-19:
It is the common consensus I have found on the internet:
For example:
https://gist.github.com/shafik/848ae25ee209f698763cffee272a58f8#how-do-we-type-pun-correctly
> At a sufficient optimization level any decent modern compiler generates identical code to the previously mentioned reinterpret_cast method or union method for type punning. 

The article give [this example](https://gcc.godbolt.org/#g:!((g:!((g:!((h:codeEditor,i:(j:1,lang:c%2B%2B,source:'%23include+%3Ccstdint%3E%0A%23include+%3Ccstring%3E%0A%23include+%3Ccstdio%3E%0A%23include+%3Ccinttypes%3E%0A%0Astatic_assert(+sizeof(+double+)+%3D%3D+sizeof(+int64_t+),+%22%22+)+%3B%0A%0Avoid+func1(+double+d+)%0A%7B%0A++++std::int64_t+n%3B%0A++++std::memcpy(%26n,+%26d,+sizeof+d)%3B+//+OK%0A++++printf(+%22%25%22+PRId64+%22%5Cn%22,+n+)+%3B%0A%7D%0A%0Avoid+func2(+double+d+)%0A%7B%0A++++union+u1%0A++++%7B%0A++++++std::int64_t+n%3B%0A++++++double+d+%3B%0A++++%7D+%3B%0A++%0A++++u1+u+%3B%0A++++u.d+%3D+d+%3B%0A++++printf(+%22%25%22+PRId64+%22%5Cn%22,+u.n+)+%3B%0A%7D%0A%0Avoid+func3(+double+d+)%0A%7B%0A++++std::int64_t+n+%3B%0A++++n+%3D+*reinterpret_cast%3Cstd::int64_t+*%3E(%26d)+%3B%0A++++%0A++++printf(+%22%25%22+PRId64+%22%5Cn%22,+n+)+%3B%0A%7D%0A%0Aint+main()%0A%7B%0A++double+d+%3D+0.1%3B%0A++%0A++func1(+d+)+%3B%0A%7D'),l:'5',n:'0',o:'C%2B%2B+source+%231',t:'0')),k:50,l:'4',n:'0',o:'',s:0,t:'0'),(g:!((h:compiler,i:(compiler:clang_trunk,filters:(b:'0',binary:'1',commentOnly:'0',demangle:'0',directives:'0',execute:'1',intel:'1',trim:'1'),lang:c%2B%2B,libs:!(),options:'-std%3Dc%2B%2B17+-O2',source:1),l:'5',n:'0',o:'x86-64+clang+(trunk)+(Editor+%231,+Compiler+%231)+C%2B%2B',t:'0')),k:50,l:'4',n:'0',o:'',s:0,t:'0')),l:'2',n:'0',o:'',t:'0')),version:4)

There is article give a more advanced use case: http://pzemtsov.github.io/2016/11/06/bug-story-alignment-on-x86.html

> These days, however, it doesn’t have to be implemented as a procedure call at all; the compiler can treat it as an intrinsic, and replace with a memory-to-register load. The GCC definitely does. It compiles the following code into the code similar to the original SSE one, except it uses movdqu instead of movdqa. This instruction allows misaligned data; its performance varies. On some processors it is much slower than movdqa even when the data is actually aligned. On others it runs at virtually the same speed.

I don't think we are in need of performance to the point we have to care about such tiny detail (program correctness is way more important).

ambroise-arm, 2023-04-19:
Thanks!

> I don't think we are in need of performance to the point we have to care about such platform specific details. Program correctness is way more important than micro performance improvement.

Absolutely, I wasn't trying to argue the rational for the change.

closed 2023-06-05 by VRichardJP