autowarefoundation/autoware_universe #3221: Add pull over as MRM behavior

mkuri, 2023-03-30:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Currently, the only choices for the MRM (Minimum Risk Maneuver) behavior are emergency stop and comfortable stop.
In order to stop the vehicle so as not to impede traffic when the autonomous driving system fails, we will add pull over as MRM behavior.

### Purpose

- Add pull over as MRM behavior

### Possible approaches

- Create MRM pull over operater node
- The node sends the target pull over position to the mission planner and the existing pull over module in the planning component will pull over to the target position appropriately.

[image]

### Definition of done

- Able to pull over as MRM behavior

kosuke55, 2024-02-01:
[image]

kosuke55, 2024-02-01:
PR: https://github.com/autowarefoundation/autoware.universe/pull/5634